MarlinFirmware/Marlin #17229: [BUG] (Toolchange + ubl) Z axis not resumed perfectly (on singlenozzle)

studiodyne, 2020-03-21:
### Bug Description

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

thinkyhead, 2020-03-31:
The `TEMPORARY_BED_LEVELING_STATE(false)` line is equivalent to this:
```
const bool tbls = planner.leveling_active;
set_bed_leveling_enabled(false);
```
 …and then at the point where `tool_change` exits it causes this to be done:
```cpp
set_bed_leveling_enabled(tbls);
```
When leveling is enabled, under most systems the Z position is immediately corrected. With U.B.L. it might require an extra step (such as a zero-length move) to prime the Z offset when leveling is enabled in the same spot.

To narrow down the issue we will need to add some logging and examine the details. In the meantime, you can try replacing the `TEMPORARY_BED_LEVELING_STATE` with the code described above to see if it makes any difference. If it does, this would indicate possible problem with `TEMPORARY_BED_LEVELING_STATE`.

It would also help to see if this same issue occurs with ABL-Bilinear.

If the problem does not go away then we have to examine the way Z correction is applied when turning UBL on and off.

studiodyne, 2020-03-31:
Don't forget , i've a SINGLENOZZLE printer , but i have no idea if the bug exists on other configuration

closed 2020-04-25 by thinkyhead

McUtty, 2020-05-17:
Hallo,

I thing i have the same problem.
After toolchange the Z Axes don't come down.
If i print a part with every tool his own layer, there is no problem. In gcode is a Zxxx command after toolchange.
If i print a part with more then one tool a layer, at fierst time, there is no Zxx in gcode, only Xxx, Yxx Exx, but in second layer. Then the X-Axes comes down. There is a Zxx command for the next layer.
If i manualy change the tool, it don't comes down.