MarlinFirmware/Marlin #17230: [BUG] (BABYSTEP + OFFSET_RANGE >9 ) -->Bad floating position ->float42_52 is wrong

studiodyne, 2020-03-21:
With
BABYSTEP_ZPROBE_OFFSET  & OFFSET_RANGE <9

Babystep display have bad floating point position 
float42-52 display type is wrong

Kaween-prog, 2020-03-25:
Can confirm:

Hardware : SKR1.3 + TMC2209 steppers 
Delta Printer.

Firmware 2.0.5.1 
BABYSTEP_ZPROBE_OFFSET   :  value set to -16.20
Expected result : Z offset showing -16.20
Result : Z offset indeed showing -16.20   (correct)

Firmware 2.0.5.2 
BABYSTEP_ZPROBE_OFFSET   :  value set to -16.20
Expected result : Z offset showing -16.20
Result : Z offset indeed showing value 10x smaller (decimal point shifted) -1.620  (wrong)

Initialising eeprom, restoring failsafe +  store  : situation does not change.

Taomyn, 2020-03-26:
I'm seeing the same thing between 2.0.5.1 and 2.0.5.2 and does seem to be a display issue as the value is stored correctly it's just shown as a number 10x smaller. Doesn't seem to affect anything as far as I can tell.

studiodyne, 2020-03-26:
In menu.h , it's a just a /100 scale
```
#if HAS_BED_PROBE
  #if Z_PROBE_OFFSET_RANGE_MIN >= -9 && Z_PROBE_OFFSET_RANGE_MAX <= 9
    #define LCD_Z_OFFSET_TYPE float43    // Values from -9.000 to +9.000
  #else
    #define LCD_Z_OFFSET_TYPE float42_52 // Values from -99.99 to 99.99
  #endif
#endif
```
I suggest to change float42_52 by a float53 or anything without scaling

Roxy-3D, 2020-03-26:
I recently made a change to add an extra decimal place of precision to that number on the display.
And the space for that extra digit came from the left side of the decimal point.
It may be we need a new, higher precision edit menu type.   Maybe 6.3 ?    

Does reverting this change fix your display?  https://github.com/MarlinFirmware/Marlin/commit/9f85165101008a87f37cb935591522e45910e848

If so...   I think we better add a new edit menu type!

studiodyne, 2020-03-26:
the best is to create a new float42_52 displaying but with no scale
Create it 👍

Roxy-3D, 2020-03-26:
So what do you think this float42_52 is exactly?    Why is it some times 4 and some times 5 ?
And 2 decimal places is not enough for precision machines and settings.   We probably need a 3 decimal place version of what ever we decide we need.

ropch, 2020-03-27:
Is this behavior due to a typo in the multiplier within the new ftostr53sign function?

The following change corrected the display on my machine:
[image]

Roxy-3D, 2020-03-27:
@ropch   Is that change still working for you?   Is it well enough tested that we can confidently change that line in the repository?   (I'll zap it if you know it fixes the problem.)

ropch, 2020-03-27:
Yes, I have the change in place and it is working for me.

I noticed the problem right after moving from 2.0.5.1 to 2.0.5.2, upon setting my z-offset.

Although I'm new to Marlin, looking over commits between 2.0.5.1 and 2.0.5.2, I saw that my configuration utilized the new ftostr53sign function.  The multiplier in that function appeared to be incorrect, and changing it (to * 10000) fixed the problem for me.

My fixed build has been working correctly since a few hours after 2.0.5.2 was tagged.  So, it gets the "It works on my machine" seal of approval from me.  ;)

studiodyne, 2020-03-28:
fix founded
thks

Roxy-3D, 2020-03-28:
OK...   I updated the number to string library with the suggested change.

closed 2020-04-03 by studiodyne