MarlinFirmware/Marlin #17231: [BUG] Stuttering printing curves

Mosher23, 2020-03-21:
### Bug Description
I have tried to verify this bug by printing flat circle. It fails.
Instead of printing circle as curve it is being printed as short lines and extruder producing too much noise. 

### My Configurations
Custom Ender 3
Skr 1.4 turbo TMC5160
Dual Z
Marlin 2.0.5

### Steps to Reproduce

1. Use my Configuration files
2. prepare circle in Fusion360 
3. Export it to Cura
4. Slice it
5. Send it to Octoprint
6. Observe Failed Print

**Expected behavior:** 
I expect curve to be printed as curve

**Actual behavior:** 
Curve is being printed as stuttered line

[Configuration Files.zip](https://github.com/MarlinFirmware/Marlin/files/4363079/Configuration.Files.zip)

ellensp, 2020-03-21:
can you also include the model and gcode?

Mosher23, 2020-03-21:
[image]
and here is gcode and stl

[CE3_test.gcode.txt](https://github.com/MarlinFirmware/Marlin/files/4363104/CE3_test.gcode.txt)

[test.stl.txt](https://github.com/MarlinFirmware/Marlin/files/4363106/test.stl.txt)

remove .txt extension if you want to use check .stl

ellensp, 2020-03-21:
How is octoprint connected to your skr1.4 USB? serial? something else?

Mosher23, 2020-03-21:
Octoprint installed on RPi4 as Octopi, wired to my board via 3-Wire Serial

ellensp, 2020-03-21:
When you start doing detailed curves you can't use serial, its just to slow....  you cant keep the command buffer full.  But check the serial baud rate, make it as fast as you can set it.  
The SKR 1.4 support full usb 2 speeds, it is a shame to cripple it and use serial.

Mosher23, 2020-03-21:
USB is a serial, so i don‘t understand how just different interface can effect print quality.

ellensp, 2020-03-21:
USB is serial, but it is a virtual UART, it does not have baud rates, its transmits data as fast as it can (up to 480 megabits per second, compared with 0.25 megabits per second over a hardware serial UART)

ellensp, 2020-03-21:
Imagine you have a sports car, you have fuel in the tank and and suitable racing engine. But the two are connected by a 0.25mm pipe...  The engine just get starved of fuel.

Mosher23, 2020-03-21:
Ok! Thank you for your response, i will try that, if i find a suitable cable

Mosher23, 2020-03-21:
do i need define somewhere in Marlin USB as interface?

Marlor, 2020-03-21:
@Mosher23 
Do you use cheap dupont wires to connect the SKR to the Raspberry?
See if your Octoprint log has disconnects or long waits.
Had the same problem, fixed it by using thicker wires. 

@ellensp 
Is this new? The last time i changed the baud of the USB interface it made a huge difference, at least in combination with octoprint

bjoernwedi, 2020-03-21:
Same Herr: but If i disconnect the serial tft cable the marlin  Mode and Print works  Fine. Increased the marlin serial buffer, this made it better, but did not solve the issue.

Mosher23, 2020-03-21:
> Same Herr: but If i disconnect the serial tft cable the marlin Mode and Print works Fine. Increased the marlin serial buffer, this made it better, but did not solve the issue.
do you use USB or serial?

bjoernwedi, 2020-03-21:
Seems the display does not read from serial in marlin mode. By this print gets blocked once serial buffer is full and marlin waits a while to flush.
This makes the blobs and stutte, i think.

bjoernwedi, 2020-03-21:
Used onboard (printer board) sd and tft in marlin Mode.
If tft serial cable is Connected: stuttering. If i Remote it: perfect print.

bjoernwedi, 2020-03-21:
This Schall have the same effect wehen using serial usb

ellensp, 2020-03-21:
> do i need define somewhere in Marlin USB as interface?

you need a serial port set to -1

p3p, 2020-03-21:
@Marlor
Baud rate only applies with the bridge chip usb connections used on the 8 bit boards, (or any 32bit board that use the ftdi like chips) on boards with the LPC176x mcu that use native usb (no uart bridge) the serial connection runs at native usb speeds. clocked in at just over 1MByte/s during testing, so using the hardware UART connections for gcode is considerably slower and lack the hardware flow control / lossless transmission. 

Other settings such as the buffers, and the host computers speed can have a effect on quality as well.

ellensp, 2020-03-21:
> @ellensp
> Is this new? The last time i changed the baud of the USB interface it made a huge difference, at least in combination with octoprint

Define new...    
Most 8 bit board don't have this feature, the USB is connected to a usb/serial chip and these have a baud rate. Some 32 bit board also use a usb/serial chip and are also limited to the baud rate.
But some 32bit boards (skr 1.3, skr 1.4 for eg) the usb is directly connected to the controller cpu and ignore the baud rate.

ellensp, 2020-03-21:
Please don't get this confused with other boards. eg SKR_MINI_V1_1, SKR_MINI_E3_V1_2, BTT_SKR_E3_DIP etc. these are very different controllers with very different processors.

Mosher23, 2020-03-21:
Problem solved! I connected Octopi via USB cable and all of a sudden print is perfect!

Thank you all for advices and help!
Cheers

closed 2020-03-21 by Mosher23