MarlinFirmware/Marlin #17233: [BUG] Homing G28 Z with triple Z Axis result in Error:Printer halted. kill() called!

D33jay, 2020-03-21:
Bug Description

Homing G28 Z prints error message _Error:Printer halted. kill() called!_ after homing seems to be "mechanically" successful.

### My Configurations

Bigtree Skr Pro 1.1
Marlin 2.0.5
3 z-axis with 3 endstops

[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/4363089/Configs.zip)

### Steps to Reproduce

1. Use 3 Z-Axis. Connect Endstops to ZMin. YMax, ZMax
2. Home Z with G28 Z
3. See that everything runs smoothly
4. LCD prints _Error:Printer halted. kill() called!_ 

**Expected behavior:**
Homing Z finishes without emergency stop

**Actual behavior:** 
Printer homes Z and throws emergency stop after homing seems to be successful

#### Additional Information

Commenting out VALIDATE_HOMING_ENDSTOPS fixes the issue, but that can't be the proper fix.
In the validate_homing_move() (which will be called if VALIDATE_HOMING_ENDSTOPS is active) it checks the trigger_state which returns the hit_state of the endstop. In the endstops.h the hit_state decleration has a comment:
// Use X_MIN, Y_MIN, Z_MIN and Z_MIN_PROBE as BIT index

Could it be the problem that the BIT index of X_MAX, Y_MAX, Z_MAX leads to a undefined hit_state and that's why validating fails?

D33jay, 2020-04-03:
I tested a little bit. 2 z axis are working fine, So I really think this is a firmware issue. 

Any suggestions?

D33jay, 2020-04-11:
Some further informations:

2 z axis works great in any combination. Homing Z with a BLTouch leads to the same issue as described above.

boelle, 2020-06-20:
@D33jay is the issue still  there?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle

tgirard, 2020-07-26:
I'm getting the exact same issue with BigTreeTech v1.2 and 2209 TMC Drivers Marlin 2.0.5.3
The Z axis will move up for a little ways and then the Kill Signal is sent.
I'm actually using the sensor less homing and it works fine as long as I move the Z axis up far enough that there's just minimal travel to the physical end where the Axis Butts up. I added both my config files here to hopefully help
[Config Files.zip](https://github.com/MarlinFirmware/Marlin/files/4977945/Config.Files.zip)

tgirard, 2020-07-26:
I tried Commenting out VALIDATE_HOMING_ENDSTOPS  as D33jay  suggested and it does fix the issue. I'm not sure if I actually need VALIDATE_HOMING_ENDSTOPS  because I'm using the Sensorless homing , to be honest, I don't understand what the VALIDATE_HOMING_ENDSTOPS  function totally does so I'm just guessing

sjasonsmith, 2020-07-26:
Are you testing with the latest bugfix-2.0.x branch? If so and the description is accurate we can re-open this. If your behavior differs then you could create a new issue.

tgirard, 2020-07-26:
Thanks Jason, I am on the latest bugfix-2.0 branch though I'm a minor version behind 2.0.5.3 (Latest is 2.0.5.4)
I should be close enough?

sjasonsmith, 2020-07-26:
@tgirard the actual bugfix branch is much newer than either of those releases, although we are expecting a new release very soon. Possibly today. If you could test once that is available or using the nightly download (matches the current bugfix-2.0.x) branch that would be great.

tgirard, 2020-07-26:
@sjasonsmith Sounds good. 
I should just use the next version that shows up here right:
https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x

It may take a while to get my setup configured or can I simply use my old Config Files? (I'm assuming not)
Let me know and Thanks!!
-Tim

sjasonsmith, 2020-07-26:
@tgirard yep, that is what you need to tst with. That will become the next release, so testing with that should be good.
You will need to update your configurations. Finding a diff tool to help compare will make it easier.

dinasind, 2020-08-06:
I‘m getting the same issue with 4 z-axis on branch 2.0.x，which commit can fix the issue on branch bugfix-2.0.x？
could you tell me the commit-id？

iShavgula, 2020-08-19:
Hello, 
I just updated to the latest release and I'm getting the same error.

sjasonsmith, 2020-08-19:
For anybody who has responded to this since it was closed:

1. If the original description describes your issue well, please attach your 2.0.6 configs, and I will re-open this issue.

2. If the original description is similar, but not quite your issue, please report a new issue and fill out the template completely.

Examples of differences which would make this a different issue:

1. Quad-Z, instead of Triple-Z
2. Homing with probe or single endstop (instead of 3 endstops)

Triple and Quad Z configurations are uncommon. Even less common when multiple endstops are used. It may take some time to find a person really able to help solve your issues, or you may be asked to help debug code yourself.

iShavgula, 2020-08-26:
Hello,

My bad, turns out it was a hardware problem in my case.