MarlinFirmware/Marlin #17234: arlin\src\gcode\calibrate\G28.cpp:264:85: error: no match for call to '(GcodeSuite::G28()::<lambda(const char*, int16_t, int16_t)>) (const char [2], const int16_t&, int, int)'

sochy99, 2020-03-21:
Good day. does anyone know what a mistake it tells me? I've searched everywhere on the Internet and I found nothing. Thanks for the help. Marlin 2.0 to skr1.3 with tmc2208 Uart and Bltouch.

`Marlin\src\gcode\calibrate\G28.cpp: In static member function 'static void GcodeSuite::G28()':
Marlin\src\gcode\calibrate\G28.cpp:264:85: error: no match for call to '(GcodeSuite::G28()::<lambda(const char*, int16_t, int16_t)>) (const char [2], const int16_t&, int, int)'
  264 |       if (DEBUGGING(LEVELING)) debug_current("X", tmc_save_current_X, X_CURRENT_HOME);
      |                                                                                     ^
Marlin\src\gcode\calibrate\G28.cpp:264:85: note: candidate: 'void (*)(const char*, int16_t, int16_t)' {aka 'void (*)(const char*, short int, short int)'} <conversion>
Marlin\src\gcode\calibrate\G28.cpp:264:85: note:   candidate expects 4 arguments, 5 provided
Marlin\src\gcode\calibrate\G28.cpp:258:26: note: candidate: 'GcodeSuite::G28()::<lambda(const char*, int16_t, int16_t)>'
  258 |     auto debug_current = [](const char * const s, const int16_t a, const int16_t b){
      |                          ^
Marlin\src\gcode\calibrate\G28.cpp:258:26: note:   candidate expects 3 arguments, 4 provided
Compiling .pio\build\LPC1768\src\src\gcode\calibrate\G425.cpp.o
*** [.pio\build\LPC1768\src\src\gcode\calibrate\G28.cpp.o] Error 1`

ellensp, 2020-03-21:
Please close one or the other of these to identical issues.

sochy99, 2020-03-21:
sry

closed 2020-03-21 by sochy99