MarlinFirmware/Marlin #17236: [FR] Add LDO-RRLCD-A004 Screen

codiac2600, 2020-03-21:
In reference to:  #17167

### Description

When trying to enable the brightness control it doesn't work with the standard REPRAP 2004 display because it doesn't have a controller for this. LDO makes an LDO-RRLCD-A004 which is a 2004 display BUT it includes the brightness adjustment capability. Prusa has enabled this in their firmware as seen here: https://github.com/prusa3d/Prusa-Firmware/pull/2270

 I hope I've entered enough information to help.

### Feature Workflow

Go to the menu and change the LCDs brightness.

#### Additional Information

https://github.com/prusa3d/Prusa-Firmware/pull/2270

ellensp, 2020-03-21:
You did you create a new FR when one already existed?   Please close this issue and add any new information to the original request.

thisiskeithb, 2020-12-03:
Duplicate of #17167

closed 2020-12-03 by thisiskeithb