MarlinFirmware/Marlin #17242: [BUG] BLTouch Fails Probing Bed running Marlin 2.0.5

toophast2catch, 2020-03-22:
I have a Tevo Tornado running an MKS Gen L 1.0. Probing worked fine in previous releases, but I'm having problems getting it to probe correctly in Marlin 5.0. I enabled Debugging for Probing and I'll provide that output. 

What happens is, it will probe start to probe and fail at point 6,7,or 8. Now it is failing at point 1. Its a BLTOUCH v3.1. Below is the debug output and I will also provide a zip of my config files. 

Changing monitoring state from "Connecting" to "Operational"
Send: N0 M110 N0*125
Recv: ok
Send: N1 M115*39
Recv: FIRMWARE_NAME:Marlin 2.0.5 (GitHub) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Greg's Tornado EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Recv: Cap:SERIAL_XON_XOFF:0
Recv: Cap:BINARY_FILE_TRANSFER:0
Recv: Cap:EEPROM:1
Recv: Cap:VOLUMETRIC:1
Recv: Cap:AUTOREPORT_TEMP:1
Recv: Cap:PROGRESS:0
Recv: Cap:PRINT_JOB:1
Recv: Cap:AUTOLEVEL:1
Recv: Cap:Z_PROBE:1
Recv: Cap:LEVELING_DATA:1
Recv: Cap:BUILD_PERCENT:1
Recv: Cap:SOFTWARE_POWER:0
Recv: Cap:TOGGLE_LIGHTS:0
Recv: Cap:CASE_LIGHT_BRIGHTNESS:0
Recv: Cap:EMERGENCY_PARSER:0
Recv: Ca:PROMPT_SUPPORT:0
Recv: Cap:AUTOREPORT_SD_STATUS:0
Recv: Cap:THERMAL_PROTECTION:1
Recv: Cap:MOTION_MODES:0
Recv: Cap:CHAMBER_TEMPERATURE:0
Recv: ok
Send: M20
Recv: Begin file list
Recv: BED_LE~1.GCO 54030
Recv: End file list
Recv: ok
Send: M155 S2
Recv: ok
[...]
Send: M111 S32
Recv: echo:DEBUG:LEVELING
Recv: ok
[...]
Send: G28
Recv: >>> G28
Recv: Machine Type: Cartesian
Recv: Probe: BLTOUCH
Recv: Probe Offset X-27.00 Y13.00 Z-2.70 (Left-Back & Below Nozzle)
Recv: Auto Bed Leveling: BILINEAR (disabled)
Recv: 
Recv: Raise Z (before homing) to 5.00
Recv: >>> do_blocking_move_to X0.00 Y0.00 Z5.00
Recv: <<< do_blocking_move_to
Recv:   current_position= X0.00 Y0.00 Z5.00 : sync_plan_position
Recv: >>> do_blocking_move_to X-468.00 Y-412.50 Z5.00
Recv: <<< do_blocking_move_to
[...]
Recv: >>> homeaxis(X)
Recv: Home 1 Fast:
Recv: >>> do_homing_move(X, -468.00, [50.00])
Recv: <<< do_homing_move(X)
Recv: Move Away:
Recv: >>> do_homing_move(X, 5.00, [50.00])
Recv: <<< do_homing_move(X)
Recv: Home 2 Slow:
Recv: >>> do_homing_move(X, -10.00, 25.00)
Recv: echo:busy: processing
Printer seems to support the busy protocol, will adjust timeouts and set busy interval accordingly
Recv: <<< do_homing_move(X)
Recv: >>> set_axis_is_at_home(X)
Recv: Axis 88 home_offset = 12.00 position_shift = 0.00
[... 430 lines cut ...]
Recv: BLTouch Command :90
Recv: bltouch.stow_proc() end
Recv:   current_position= X52.00 Y12.00 Z5.31 : sync_plan_position
Recv:   current_position= X52.00 Y12.00 Z5.31 : <<< Probe::probe_down_to_z
Recv: FAST Probe fail!
Recv:   current_position= X52.00 Y12.00 Z5.31 : <<< run_z_probe
Recv:   current_position= X52.00 Y12.00 Z5.31 : Probe::set_deployed
Recv: deploy: 0
Recv: Probe::move_z(10.00)
Recv: >>> do_blocking_move_to X52.00 Y12.00 Z12.70
Recv: <<< do_blocking_move_to
Recv: echo:busy: processing
[...]
Recv: >>> do_blocking_move_to X52.00 Y12.00 Z12.70
Recv: <<< do_blocking_move_to
Recv: Error:Probing Failed
Changing monitoring state from "Operational" to "Error: Probing Failed"
Send: M112
Send: N2 M112*35
Send: N3 M104 T0 S0*34
Send: N4 M140 S0*97
Changing monitoring state from "Error: Probing Failed" to "Offline (Error: Probing Failed)"
Connection closed, closing down monitor

Any help is appreciated. Looks to me this is firmware bug.

toophast2catch, 2020-03-22:
Config Files attached. 
[TT Config.zip](https://github.com/MarlinFirmware/Marlin/files/4364688/TT.Config.zip)

static-max, 2020-03-23:
Same problem here after merging lasted changes, SKR 1.3 and branch bugfix-2.0.x from yesterday.
Went back to commit a5ad88585e0157bd1fb83ab9836aac69140844fb [a5ad885] and BLTouch works fine again.

Sonsi79, 2020-03-25:
Same Error here.
BLTouch works on bugfix.2.0.x fine. 2.0.4.4 and 2.0.5.x probe fails after first point :-(

Mosher23, 2020-03-25:
i have same issue. Probing fails at different  probing points.
I am on Marlin 2.0.5 using Bilinear auto leveling 4x4 with BLTouch

toophast2catch, 2020-03-28:
Any update on this issue?

lucableve, 2020-03-29:
I have same issue on SKR 1.4 and the printer when hotbed heated crash serial , I think it's due to the same problem

toophast2catch, 2020-03-29:
All I've come to notice if my bed leveling knobs are just the slightest bit loose (not snug) then this issue occurred. Whats funny is an earlier version of Marlin did not do this. I have two printers - An Ender 3 running and SKR Mini E3 1.2 and a Tevo Tornado running an MKS Gen L V1.0. At first, it only did this on the Tevo. I ended up resolving that by adjusting the bed leveling knobs and using a Torpedo level to level out the bed the best I could. 

Now my Ender 3 started doing it. The Ender always failed after probing the second time in the 3rd spot. I noticed my bed level knob was loose and snug it back up. Issue hasn't happened again. 

Did something change in Marlin in rergards to multiple probing? Is it comparing a variance from the first time it probes to the second and then fails if that $variance  > X (where X is some configurable value lower than configured variance)?

MVS8, 2020-03-29:
Hi, I had the same issue, BLTouch bilinear auto levelling, 7x7 grid, 4 multiple probing AND 1 extra probing. Yesterday I tried 4 times G29 and all failed at different moments. Today, after reflashing it with disabled extra probing (//#define EXTRA_PROBING    1), it worked!
I just post this in case it helps anyone but I cannot find a quick explanation for it.

phoenixfire425, 2020-03-30:
Having the same issue.  Using 2.0.5.2(release) also have tried latest 2.0.x bugfix (747a4bb)
BLTouch probe can be deployed via the control panel.  But it does not deploy when running G28
Using SKR 1.4 Turbo.  BLTouch is connected via the dedicated Probe and Servo pins.  (P0_10)  This was working on 2.0.4.4

lucableve, 2020-03-31:
Hi everyone, I confirm with version 2.0.4.4 the bltouch works and the printing does not stop with the heated bed.
It's a big bug in 2.0.5.x

I tried to make a comparison between the versions and I reported some fixes for the skr but it is too much code to analyze not having written it myself ....  and I don't have the time to patch.
We hope that the developers solve .... in the meantime I stay on 2.0.4.4

FaultyData, 2020-04-09:
> Having the same issue.  Using 2.0.5.2(release) also have tried latest 2.0.x bugfix (747a4bb)
> BLTouch probe can be deployed via the control panel.  But it does not deploy when running G28
> Using SKR [...]

SERVO0_PIN is P2_00 (dedicated servo pins)
P0_10 is Z_MIN_PIN

[Page 8](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/BTT%20SKR%20V1.4/BTT%20SKR%20V1.4%20Instruction%20Manual.pdf)

Kaimane, 2020-04-17:
> Now my Ender 3 started doing it. The Ender always failed after probing the second time in the 3rd spot.

The same behavior here on an Ender 3 with a SKR Mini E3 1.2 and bilinear ABL.
Running 2.0.5.3 the last week with two or three fails when leveling the hotbed. Then after restarting the printer probing works fine. But today it fails all the time in the 3rd probing spot. I flashed 2.0.4.4, restored and saved configuration but also no luck; it fails in the 3rd probing spot again. As my last chance I tried the latest bugfix-2.0.x ( [50fa87f](https://github.com/MarlinFirmware/Marlin/commit/50fa87f1cfad92ff88f7c14ecaa39e1c2b33639f) ) but still no luck. Probing fails in the third spot again.
I currently cannot use ABL no matter which version of marlin.
Are there any new insights in the meantime what could be causing the error?

patriot1889, 2020-04-17:
Does anyone have adpative_step_smoothing enabled?

I had very repeatable errors with adaptive step smoothing enabled when I had hybrid threshold disabled and a high z homing feedrate. It caused printer halts all the time.

I fixed it by enabling hybrid threshold, but then the probing accuracy was way off and created some pretty crazy meshes. Currently I’m running with adaptive_step_smoothing disabled and no problems at all!

Kaimane, 2020-04-17:
I configured 2.0.5.3 with ADAPTIVE_STEP_SMOOTHING enabled. Because of your comment I disabled it and built the firmware again. But nothing happened. Probing stops at the 3rd spot.
HOMING_FEEDRATE_Z is 3*60 (3mm/s) and HYBRID_THRESHOLD is enabled.

patriot1889, 2020-04-17:
> I configured 2.0.5.3 with ADAPTIVE_STEP_SMOOTHING enabled. Because of your comment I disabled it and built the firmware again. But nothing happened. Probing stops at the 3rd spot.
> HOMING_FEEDRATE_Z is 3*60 (3mm/s) and HYBRID_THRESHOLD is enabled.

Ah, worth a try.

Kaimane, 2020-04-17:
In the meantime I have tried several configurations based on 2.0.5.3. But now the probing seems to work again. The only thing I changed in my configuration is the parameter GRID_MAX_POINTS_X from 4 to 3.
With this configuration, I successfully leveled the hotbed about 10 times in a row without a single failure.

Einhorn-1802, 2020-04-18:
I am observing a similar issue. I have an MKS Robin Nano 1.2 and an origital BLTouch 3.1.
Randomly probing fails. In my case SLOW probing fails and I observed that the problem is linked to the probing speed. When I reduce the probing speed everything works fine. 

Digging a little bit into the firmware, I see the following behavior:
BLTouch should trigger at Z = 0.5 and the Z target in the movement is -0.5. The printer head reaches -0.5, but no trigger occured. When I lower the probing peed, e. g. 

// Feedrate (mm/m) for the "accurate" probe of each point
#define Z_PROBE_SPEED_SLOW (Z_PROBE_SPEED_FAST/6)  // In my case this is 1000 / 6

the problem disappears.

qwewer0, 2020-04-18:
I had this from 2.0, but I thought it is because of the noise 5V that the board makes.
I had the fast probing at (20 * 60), and I had a lot of failed probing attempts, but now with (3 * 60) it happens very rarely.

Ender-3, SKR Mini E3 v1.2, latest bugfix.

Mosher23, 2020-04-20:
Issue is still there.
5x5 Bilinear probing failing every time.
Marlin 2.0.5.3

Einhorn-1802, 2020-04-20:
> Issue is still there.
> 5x5 Bilinear probing failing every time.
> Marlin 2.0.5.3

Which offset between probe and nozzle do you have? M851
In configuration.h: Which #define Z_PROBE_LOW_POINT do you have?

I would like to compare that, with my latest settings.

Kaimane, 2020-04-22:
> In the meantime I have tried several configurations based on 2.0.5.3. But now the probing seems to work again. The only thing I changed in my configuration is the parameter GRID_MAX_POINTS_X from 4 [...]

After this good news and a successful leveling, the problem appears again. ABL does not work. It is now the case that G28 still works without problems, but with a G29 the errors appear again. After the axes have moved to the first leveling point, the BLTouch (in my case a clone from Geeetech, 3DTouch) deploy, the LED turns off, then it stows and the LED lights up. This happens three times until the process is canceled.

I tested the behavior with several stable 2.0.x.x firmwares, as well as with the latest [bugfix 2.0.x](https://github.com/MarlinFirmware/Marlin/commit/5a5be7e287183e633ee3235ee1bcd79a72a1a1f5).

Here is my configuration based on the latest Bugfix-2.0.x for my Ender-3, SKR Mini E3 v1.2, Geeetech 3DTouch. Is there any configuration error I made and overlooked?
[marlin-bugfix-2.0.x-config.zip](https://github.com/MarlinFirmware/Marlin/files/4516287/marlin-bugfix-2.0.x-config.zip)

sputnik378, 2020-06-02:
I have been battling this same issue all night. I have tried every suggestion listed on this page and it still fails randomly at some point in the probing process.

It was working, and I was fine tuning when it just suddenly started failing. It's been over a month since the last post, I'm curious if anyone has found a solution?

Ender 3, SKR 1.4 Turbo, 2208, BLTouch 3.1, latest bugfix 2.0.x

pmikulicz, 2020-06-07:
I have the same problem in robin nano 1.2. Marlin firmware 2.0.5.3. What is funny that on this version one week earlier everythig worked perfect.

ilium007, 2020-06-10:
Same issues here running 2.0.5.3

`FIRMWARE_NAME:Marlin 2.0.5.3 (GitHub) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:3D Printer EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff`

I have enabled `#define DEBUG_LEVELING_FEATURE` and re-compiled.

I have issued `M502` and `M500` and `M111 S247` but I get no debug output when BLTouch probing fails.

```
echo:M111 S247
echo:DEBUG:ECHO,INFO,ERRORS,COMMUNICATION
ok
echo:G29
echo:busy: processing
echo:busy: processing
Error:Probing Failed
X:10.00 Y:10.00 Z:10.00 E:0.00 Count X:3200 Y:3200 Z:3200
ok
```

Einhorn-1802, 2020-06-10:
After fixing other items on my printer I planned to work on a workaround to get rid of that issue. Before I started I modified my configuration settings to get an easier start for that fix. After that modification in configuration.h the issue was gone! Currently I am using the following settings: 

#define HOMING_FEEDRATE_Z  (4*60)
#define Z_PROBE_SPEED_FAST (HOMING_FEEDRATE_Z)
#define Z_PROBE_SPEED_SLOW (Z_PROBE_SPEED_FAST / 3)

and I have removed:

//#define MULTIPLE_PROBING 2
//#define EXTRA_PROBING    1

// for this setting you need an offset > 2 mm between probe and nozzle
#define Z_PROBE_LOW_POINT          -2 // Farthest distance below the trigger-point to go before stopping

Maybe someone is able to double-check these settings. At the moment it might be a speed issue or something releated to multiple or extra probing.

boelle, 2020-06-21:
@toophast2catch still an issue?

ilium007, 2020-06-21:
> @toophast2catch still an issue?

Yes, I had to go back to 2.0.4.4 and everything works, same hardware / config.

toophast2catch, 2020-06-21:
. Please ignore my last response. I thought I was responding to another case. 

Yes, this is still an issue for many as you can see above. 

I figured out a work around by leveling my bed with my bed leveling knobs, but this defeats the purpose of ABL. 

I've found this issue occurs when the bed is so out of level that the BLTOUCH seems to detect a Variance in the probe points and it fails probing. Maybe it's a new safety? But still odd. 

If I level my bed with my knobs, probing works just fine. But this defeats the purpose of ABL. My bed is so flat now, my z axis doesn't compensate for anything.

ilium007, 2020-06-21:
I don't have bed levelling knobs on my printer unfortunately. I have to shim it to close and then use UBL on 2.0.4.4 which works fine.

boelle, 2020-06-23:
Please test the [bugfix-2.0.x branch](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands.

jpflouret, 2020-06-26:
I have a similar problem with slow probe failing.

The sanity check in probe.cpp seems to fail if the BL touch is mounted low and offset.z is zero (as I often use initially to find the actual offset):
```
      // Probe downward slowly to find the bed
      if (probe_down_to_z(z_probe_low_point, MMM_TO_MMS(Z_PROBE_SPEED_SLOW))      // No probe trigger?
        || (sanity_check && current_position.z > -offset.z + _MAX(Z_CLEARANCE_MULTI_PROBE, 4) / 2)  // Probe triggered too high?
      ) {
        if (DEBUGGING(LEVELING)) {
          DEBUG_ECHOLNPGM("SLOW Probe fail!");
          DEBUG_POS("<<< run_z_probe", current_position);
        }
        return NAN;
      }
```

In my case, the trigger position is 2.09mm and offset.z is 0mm with Z_CLEARANCE_MULTI_PROBE default to 5.

The result of `-offset.z + _MAX(Z_CLEARANCE_MULTI_PROBE, 4) / 2` is 2 (because it is is all ints) and thus my current_position.z of 2.09mm causes the sanity check to fail. Hoever, even if done in float, I could mount my BL touch slightly lower causing the sanity check to still fail.

The author of this bug @toophast2catch had a failure on fast probing:
```
Recv: FAST Probe fail!
Recv: current_position= X52.00 Y12.00 Z5.31 : <<< run_z_probe
```

This is most likley due to the similar sanity check for the fast probe ` current_position.z > -offset.z + _MAX(Z_CLEARANCE_BETWEEN_PROBES, 4) / 2`. The trigger position of 5.31 will fail this test (assuming the default value of Z_CLEARANCE_BETWEEN_PROBES of 5).

This particular sanity check was added by @InsanityAutomation in commit 84b6e11bd53acbfde0ad82445e21d7bce1cc14f6. I have commented out the sanity check locally and my BLTouch works as expected.

jpflouret, 2020-06-26:
@boelle - branch bugfix-2.0.x has the same issue but reported as triggered early:

    Send: G29
    Recv:   current_position= X-5.00 Y-5.00 Z0.00 : >>> G29
    Recv: Machine Type: Cartesian
    Recv: Probe: BLTOUCH
    Recv: Probe Offset X-25.00 Y-30.00 Z0.00 (Left-Front & Same Z as Nozzle)
    Recv: Auto Bed Leveling: BILINEAR (disabled)
    Recv: 
    Recv:   current_position= X-5.00 Y-5.00 Z0.00 : Probe::set_deployed
    Recv: deploy: 1
    Recv: Probe::move_z(10.00)
    Recv: >>> do_blocking_move_to X-5.00 Y-5.00 Z10.00
    Recv: <<< do_blocking_move_to
    [...]
    Recv: echo:busy: processing
    Recv: BLTouch DEPLOY requested
    Recv: BLTouch Command :10
    Recv: bltouch.deploy_proc() end
    Recv: >>> do_blocking_move_to X-5.00 Y-5.00 Z10.00
    Recv: <<< do_blocking_move_to
    Recv: reset_bed_level
    Recv: >>> Probe::probe_at_point(20.00, 20.00, raise, 0, probe_relative)
    Recv:   current_position= X-5.00 Y-5.00 Z10.00 :
    Recv: >>> do_blocking_move_to X45.00 Y50.00 Z10.00
    Recv: <<< do_blocking_move_to
    [...]
    Recv: echo:busy: processing
    Recv:   current_position= X45.00 Y50.00 Z10.00 : Probe::set_deployed
    Recv: deploy: 1
    Recv:   current_position= X45.00 Y50.00 Z10.00 : >>> Probe::run_z_probe
    Recv:   current_position= X45.00 Y50.00 Z10.00 : >>> Probe::probe_down_to_z
    Recv: BLTouch DEPLOY requested
    Recv: BLTouch Command :10
    Recv: bltouch.deploy_proc() end
    Recv: >>> do_blocking_move_to X45.00 Y50.00 Z-2.00
    Recv: <<< do_blocking_move_to
    [...]
    Recv: echo:busy: processing
    [...]
    Recv: echo:busy: processing
    Recv: BLTouch STOW requested
    Recv: BLTouch Command :90
    Recv: bltouch.stow_proc() end
    Recv:   current_position= X45.00 Y50.00 Z2.09 : sync_plan_position
    Recv:   current_position= X45.00 Y50.00 Z2.09 : <<< Probe::probe_down_to_z
    Recv: SLOW Probe fail! - Triggered early.
    Recv:   current_position= X45.00 Y50.00 Z2.09 : <<< run_z_probe
    Recv:   current_position= X45.00 Y50.00 Z2.09 : Probe::set_deployed
    Recv: deploy: 0
    Recv: Probe::move_z(10.00)
    Recv: >>> do_blocking_move_to X45.00 Y50.00 Z10.00
    Recv: <<< do_blocking_move_to
    [...]
    Recv: echo:busy: processing
    [...]
    Recv: BLTouch STOW requested
    Recv: BLTouch Command :90
    Recv: bltouch.stow_proc() end
    Recv: >>> do_blocking_move_to X45.00 Y50.00 Z10.00
    Recv: <<< do_blocking_move_to
    Recv: Error:Probing Failed

The code has been changed to use a lambda but the clearance calculation remains the same:

```cpp
      if (try_to_probe(PSTR("SLOW"), z_probe_low_point, MMM_TO_MMS(Z_PROBE_SPEED_SLOW),
                       sanity_check, _MAX(Z_CLEARANCE_MULTI_PROBE, 4) / 2) ) return NAN;
```

Perhaps `_MAX(Z_CLEARANCE_MULTI_PROBE, 4) / 2` should simply be `Z_CLEARANCE_MULTI_PROBE`.

boelle, 2020-07-02:
@toophast2catch @jpflouret did #18435 fix the issue?

jpflouret, 2020-07-10:
It did for me.

CavinRauch, 2020-07-13:
This happened to me out of the blue. 
Turned out 2 of the wires were slightly damaged so homing in the centre of the bed worked ( G28 ).
But moving to the first point of G29 it would fail and the probe would deploy and stow around 2 - 3 times then throw the error.

The 2 wires on the BL Touch were the Black and White.

boelle, 2020-07-15:
@jpflouret oki will close close this one as fixed then

we can always reopen if its not the case

PR #18435 (2020-06-28): Fix probe clearance calculation (#17242)
### Description

This pull request fixes the clearance calculation for z probing with zero probe offset which often causes the probe to fail. The calculation of required clearance of half the configured distance requres a probe to be mounted at least twice as far as the clearance between probe distance. Additionally, the calculation is done in integer math which also requires the probe to be mounted even slightly higher than twice the clearance.

### Benefits

Increases the range of allowable z probe mounting distances for the z probe. Also allows probi [... cut]
closed 2020-07-15 by boelle