MarlinFirmware/Marlin #17243: [BUG] UBL probing not reaching reachable points

nfons, 2020-03-22:
### Bug Description

I previously ran a 6x6 Bi-linear ABL which probed all 36 points without issue. however using the same settings with a UBL yields sporadic/inconsistent probing.
 

### My Configurations

Marlin 2.0.5, Ender 3 Pro

Running a BLT3.1 with Bullseye mount @ -48,-10. 

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

Can do this with mesh inset 0,1,5,10 (#s i have tried)

- Create a 6x6 UBL mesh.
- Run G28 to home
- G29 P1 to start mesh
- G29 T to print after done

**Expected behavior:** [What you expect to happen]

On a 6x6, with `mesh_inset = 0` we would expect the mesh to be reachable all the way to the end of print bed (even if its a dumb move, since the nozzle is offset).

[image]

as you can see column 0, 4,5 is blank, as well as row 5 and row 0. 

Using a higher `mesh_inlet` I would expect the probable areas to be smaller..however it reaches more places with a higher inlet

5 inlet:

[image]

and finally a 10 inlet gets a much better probe points 31/36
[image]

According to Marlin documentation
mesh min/max should increase with inset decretion thus should probe more areas

```
#define MESH_MIN_X UBL_MESH_INSET
#define MESH_MAX_X (X_BED_SIZE - (UBL_MESH_INSET))
#define MESH_MIN_Y UBL_MESH_INSET
#define MESH_MAX_Y (Y_BED_SIZE - (UBL_MESH_INSET))
```

qwewer0, 2020-03-22:
Did you consider `MIN_PROBE_EDGE 10`?

Vertabreak, 2020-03-22:
min probe edge in combination with UBL results in skipping rows generally handled by G29 P3.

nfons, 2020-03-22:
Hi @qwewer0  @Vertabreak , then why would max x/y be lowered with smaller insets?.
logically you would expect even if min probe to effect mesh size, a larger inset would make min probe more pronounced?

qwewer0, 2020-03-22:
MESH_INSET determines the UBL area (mm^2) on the bed. If MESH_INSET is 0 then the 6 * 6 mesh points are spread across the whole bed area, but if the probe can't probe outside the given safe area (MIN_PROBE_EDGE) then there will be unprobed points, and the same applies to the X/Y_MIN_POS and X/Y_MAX_POS, as the printhead can't leave that area.

nfons, 2020-03-23:
@qwewer0 thanks for the clarification. but that still doesnt answer why there was missing probes when I set min_probe to 0. (see edit)

Vertabreak, 2020-03-23:
inset limits the probing area min probe edge skips points.

qwewer0, 2020-03-23:
@nfons 

> and the same applies to the X/Y_MIN_POS and X/Y_MAX_POS, as the printhead can't leave that area.

This ^ should answered your question.
If the hotend carriage can't go far enough to the right because it would be outside the X_MAX_POS, then there will be ubprobes points. In order to be able to probe that area, the printhead would need to move to the right by [X_MAX_POS - NOZZLE_TO_PROBE_OFFSET { X }], so if your NOZZLE_TO_PROBE_OFFSET { X } = -42 and your X_BED_SIZE = 220, then in order to be able to probe all the points, the nozzle would need to reach 262 mm in the X axis, which isn't always possible to the printer.

qwewer0, 2020-03-23:
@nfons I think this can be closed.

closed 2020-03-23 by nfons