MarlinFirmware/Marlin #17244: SD card unreadable

lacchinig, 2020-03-22:
Dears,

I have a Tango board with REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER LCD board connect. I have ever used the original 1.1.8 marlin firmware without problem. I moved to 2.x version but I cannot read the SD card.
First the LCD set to REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER doesn't work, I solved it by next suggestion :
In Marlin/src/lcd/dogm/ultralcd_DOGM.h
find U8GLIB_ST7920_128X64_RRD
and replace with U8GLIB_ST7920_128X64_4X_HAL

### Bug Description
The issue of SD card remains: 
if I remove the SD card the LCD check correctly the removing and viceversa the insert of CD card is correctly identified, but there were a media non present message. By serial monitor of Arduino the system said "SD Init Fail" and "Error:volume.init failed".

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
[Archivio.zip](https://github.com/MarlinFirmware/Marlin/files/4364826/Archivio.zip)

Powerprobot, 2020-03-22:
Take a fresh copy of Marlin 2.0-Bugfix. The issue is fixed in [#17112](https://github.com/MarlinFirmware/Marlin/pull/17112)

lacchinig, 2020-03-23:
I'm working with Marlin 2.0.5.1 which have just the SD_DETECT_STATE instead SD_DETECT_INVERTED.

After different tests the problem appear randomly. I have just tried to set SD_DETECT_STATE HIGH, without particular change. Now sometimes works sometimes not. I start to think it could be damage the physical connection...

boelle, 2020-06-20:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-22 by boelle