MarlinFirmware/Marlin #17247: Sensor-less homing problem

Talha909, 2020-03-22:
I have SKR 1.4 Turbo and TMC 2130 SPI. I used the Marlin 2.0.5.1. I configue the files and upload the firmware. I have problem with the sesnorless homing. Its not working. I did not installed the end stops. 
[image]

Geitde, 2020-03-23:
Are you sure you set the soldering pad on the step sticks right?

it is hard to say what is going on without seeing the config files you used.

boelle, 2020-06-20:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-20 by boelle