MarlinFirmware/Marlin #17249: I cant upload to LPC1768

JXMIEZ, 2020-03-22:
### Bug Description

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

ellensp, 2020-03-22:
020000 is old please get a current marlin.
Atom no longer supports platformio, please use vscode.
https://marlinfw.org/docs/basics/install_platformio_vscode.html

JXMIEZ, 2020-03-22:
what marlin should i use in vscode

JXMIEZ, 2020-03-22:
In vscode the same error keeps happening. I don't understand what error is occurring whenever i build it. even when i don't make any changes it still gives me errors

Powerprobot, 2020-03-22:
Check your microSD on the LPC1768 Board. I have the same issue for 5-6 days and i check the microSD and the Card are died.

JXMIEZ, 2020-03-22:
My card is good. this isn't the problem. I think this error message is something to do with the firmware
c:\users\fcast\documents\marlin-bugfix-2.0.x\marlin\src\hal\shared\marduino.h:36:10: fatal error: Arduino.h: No such file or directory

Powerprobot, 2020-03-22:
try this: use shorter file name. example C:\Project\marlin....   and take a fresh copy from Marlin Bugfix and Use VSCode with PlatformIO.

JXMIEZ, 2020-03-23:
I'm able to build to mega2560 but not LPC1768

JXMIEZ, 2020-03-23:
When i download to LPC1768 it gives me these random errors

JXMIEZ, 2020-03-23:
It wants me to download vcpkg

Powerprobot, 2020-03-23:
I think it is a windows problem, not a marlin problem. Try to make a new User in Windows an Install VSCode and PlatformIO new in this new User. (Or try to deinstall vscode and restart Windows with Shift key and re install)
Please close this Issue.

thinkyhead, 2020-03-23:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-03-23 by thinkyhead