MarlinFirmware/Marlin #17252: SKR PRO 1.1 + TMC5160, Homing is "working" but no other movement

mcmuff91, 2020-03-22:
Hello everyone,
atm I´am working on a big selfmade Delta-printer for Ceramics.
The setup:

- Skr Pro
- TMC 5160 V1.2 BTT (removed the CLK Pin), SPI
- Nema 23, 3 A, 1,8°
- 24 Volt Power supply
- RepRap Full Graphic LCD
- Mechanical Endstops

When I send G28 via Pronterface or via the RepRap-Controller the printer starts to home. But one after the other (x,y,z). There is no movement of all three Axis at the same time, so I have to move the Effector to the upper end of the printer. The axes are hitting the endstops and doing a rebump . But after this there is no movement to the safe zone (enabled DELTA_HOME_TO_SAFE_ZONE)
After homing the Printer is not responding to movement commands via Pronterface or the Controller.

Tried multiple Versions of Marlin ( the BTT-Version, 2.04, 2.05).
I have no idea how to fix the Problem.

[Marlin config 2.0.5.zip](https://github.com/MarlinFirmware/Marlin/files/4365679/Marlin.config.2.0.5.zip)

thinkyhead, 2020-03-24:
Remember to send `M502`, `M500` after re-flashing.

Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information so that we're not just taking stabs in the dark. Here is the boilerplate:

- Download Marlin `bugfix-2.0.x` to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

mcmuff91, 2020-03-24:
@thinkyhead 
Thank you for your answer.
I tried it witch the Marlin 2.0.5 version (latest). Right now I set the  Bumping divisor to 0, cause i read about an issue, but no changes.

[Hominglog.txt](https://github.com/MarlinFirmware/Marlin/files/4374194/Hominglog.txt)

closed 2020-03-31 by mcmuff91