MarlinFirmware/Marlin #17253: [BUG] the z probe do not compensate enought the bed level

rainito, 2020-03-22:
Hi to all dear friend. I have a serious problem with my printer. I have an flsun with double extruder, inductive sensor for the autolevel and the zise of the biuld plate of 300 x 300 with glass plate.
My problem is the probe leveling. I see the z stepper motor move to the right direction for compensate the bed level but don't move enouht.  
 I have the right side of the bed with the nozzle too close to the bed and the left side with the nozzle too far from the bed. I try linear and bilinear mode but the problem is the same. I have two z motor axis and i have leveled every motor to have the same distance of the nozzle in the left and right side.
I try with marlin 1.1.9x bugfix, marlin 2.0.x bugfix and an older version of marlin 1.1.4 but the problem always exist. I have an mks gen l like motherboard and for x-y and z  axis the tmc2208 driver and for the extruders the lv8729 drivers.Sorry for my english but i’m italian-

### Bug Description
The z probe do not compensate perfectly the level of the bed 

I will update mi configuration h and adv when i will go home after work. Thanks to all

### Steps to Reproduce

1. G28
G29
Z0 and with a paper below the nozzle try the resistance. In the center is perfect but going to the left of the bed, the nozzle go down but not enought, so the nozzle stay too far from the bed and vice versa for the right side (the nozzle go up but not enought so the nozzle is too far to the bed)
**Expected behavior:** 
The probe would compensate the level of the bed correctly.

rado79, 2020-03-22:
https://github.com/MarlinFirmware/Marlin/issues/15206

rainito, 2020-03-23:
Hi and thanks for the reply. I've read that post but my axis are perfect. I haven't hardware problem or i think this. I checked all amd i don't find anything. The probel can be only by hardware?

rainito, 2020-03-24:
[Archivio.zip](https://github.com/MarlinFirmware/Marlin/files/4372538/Archivio.zip)

thinkyhead, 2020-03-24:
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information so that we're not just taking stabs in the dark. Here is the boilerplate:

- Download Marlin `bugfix-2.0.x` to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Do some of the moves that revealed problems before. Take notes.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

thinkyhead, 2020-03-24:
Since your mesh doesn't seem to be working for you, test under very controlled conditions with simple moves. Use `M114 D` to check the current position for leveling being applied. Observe your Z motors to see that they are turning as you do XY moves. You can also clear and then edit the mesh and do tests with a single value set to (for example) -0.2 or +0.2 to see that the nozzle raises or lowers as expected when it goes to that point.

Please collect as much data as possible to demonstrate incorrect behavior, and this will help to locate the problem, if any.

rainito, 2020-03-25:
> Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information so that we're not just taking stabs in the dark. Here is the boilerplate: [...]

Hi and thanks to you too for the reply. Sorry for my late reply but i’m a nurse and in this period i’m very busy with the work and sorry for my English but i’m italian :)
Today i will try your first tip and then the second and will let you knoe the tests results. Thanks and stay tuned :)

rainito, 2020-03-25:
ok these are the reluts of the two attempts of your first tip.
in the extreme left of the bed i have a perfect distance for the first layer but just as soon i move the x axxis to the right the nozzle don't go up enought. i see the z axis motor moves to compensate but nothing...
asap i try your second tip and i will update the results.
[primo.pdf](https://github.com/MarlinFirmware/Marlin/files/4381516/primo.pdf)
[secondo.pdf](https://github.com/MarlinFirmware/Marlin/files/4381518/secondo.pdf)

rainito, 2020-03-27:
> Since your mesh doesn't seem to be working for you, test under very controlled conditions with simple moves. Use `M114 D` to check the current position for leveling being applied. Observe your Z [...]

ok,how you can see,the levelling is working as you can see by m114 d but in the bed the nozzle is too far and in the right too close
left side:
SENDING:M114 D
X:20.00 Y:180.00 Z:0.00 E:0.00 Count X:2000 Y:18000 Z:-74
Logical: X:20.000 Y:180.000 Z:0.000
Raw:     X:20.000 Y:180.000 Z:0.000
Leveled: X:20.000 Y:180.000 Z:-0.185
UnLevel: X:20.000 Y:180.000 Z:0.000
Stepper: X:2000 Y:18000 Z:-74 E:0
FromStp: X:20.000 Y:180.000 Z:-0.185 E:0.000
Diff:  X:0.000 Y:0.000 Z:-0.000 E:0.000

right side
SENDING:M114 D
X:270.00 Y:180.00 Z:0.00 E:0.00 Count X:27000 Y:18000 Z:60
Logical: X:270.000 Y:180.000 Z:0.000
Raw:     X:270.000 Y:180.000 Z:0.000
Leveled: X:270.000 Y:180.000 Z:0.151
UnLevel: X:270.000 Y:180.000 Z:0.000
Stepper: X:27000 Y:18000 Z:60 E:0
FromStp: X:270.000 Y:180.000 Z:0.150 E:0.000
Diff:  X:0.000 Y:0.000 Z:-0.001 E:0.000

i don't know what i need to do...can i change manually the mesh in every single problem points? if yes how can i do this?? is better the UBL sistem with my kind of problem? help

rainito, 2020-03-28:
someone can help me??

rainito, 2020-03-30:
Is it possible to invert the logic of the probe? it appears that the sensor regulates the height reversed. When it has to go down, it goes up and vice versa

rainito, 2020-04-01:
I read in some post that can be a resistor problem in my board. I have an mks gen l v 1.0 and an lj18a3-8-z/bx npn no inductive sensor mounted. Can i post the link of that post? Is another forum

thinkyhead, 2020-05-01:
For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-05-01 by thinkyhead