MarlinFirmware/Marlin #17254: [BUG] upload does not work

D4str, 2020-03-22:
My Magic 1.1 board (RAMPS 14 EFB) repeatedly crashes. After successful upload, the board will restart and everything is as before the upload. The board communicates seamlessly with the display and controls. It can also be read via USB. 

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/megaatmega2560.html
PLATFORM: Atmel AVR 2.0.0 > Arduino Mega or Mega 2560 ATmega2560 (Mega 2560)
HARDWARE: ATMEGA2560 16MHz, 8KB RAM, 248KB Flash
PACKAGES:
 - framework-arduino-avr 5.0.0
 - tool-avrdude 1.60300.190424 (6.3.0)
 - toolchain-atmelavr 1.50400.190710 (5.4.0)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 16 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <LiquidCrystal> 1.5.0
|   |-- <Wire> 1.0
|-- <TMCStepper> 0.6.2
|   |-- <SoftwareSerial> 1.0
|   |-- <SPI> 1.0
|-- <Adafruit NeoPixel> 1.3.5
|-- <U8glib-HAL> 0.4.1
|   |-- <Wire> 1.0
|   |-- <SPI> 1.0
|-- <Adafruit MAX31865 library> 1.0.3
|   |-- <SPI> 1.0
|-- <LiquidTWI2>
|   |-- <Wire> 1.0
|-- <Arduino-L6470> 0.8.0
|-- <SailfishLCD>
|-- <SailfishRGB_LED>
|   |-- <Wire> 1.0
|-- <SlowSoftI2CMaster>
|-- <TMC26XStepper>
|   |-- <SPI> 1.0
|-- <Wire> 1.0
|-- <SPI> 1.0
Building in release mode
Checking size .pio\build\mega2560\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [===       ]  29.5% (used 2418 bytes from 8192 bytes)
Flash: [==        ]  21.3% (used 53982 bytes from 253952 bytes)
Configuring upload protocol...
AVAILABLE: wiring
CURRENT: upload_protocol = wiring
Looking for upload port...
Auto-detected: COM3
Uploading .pio\build\mega2560\firmware.hex

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e9801 (probably m2560)
avrdude: reading input file ".pio\build\mega2560\firmware.hex"
avrdude: writing flash (53982 bytes):

Writing | ################################################## | 100% 6.54s

avrdude: 53982 bytes of flash written
avrdude: verifying flash memory against .pio\build\mega2560\firmware.hex:
avrdude: load data flash data from input file .pio\build\mega2560\firmware.hex:
avrdude: input file .pio\build\mega2560\firmware.hex contains 53982 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 6.37s

avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x0000
         0x14 != 0x0c
avrdude: verification error; content mismatch

avrdude: safemode: Fuses OK (E:FD, H:D8, L:FF)

avrdude done.  Thank you.

*** [upload] Error 1

ellensp, 2020-03-23:
There is something wrong with your mega, either it has faulty flash, or the bootloader it not working correctly.

D4str, 2020-03-23:
bootloader i can't burn it writes an error. It used to work. Is it possible to bypass or fix this problem?

thinkyhead, 2020-03-24:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-03-24 by thinkyhead