MarlinFirmware/Marlin #17257: [K-factor Calibration Pattern] (Generated G-Code is not functioning on my printer)

Nibroc-Fitz, 2020-03-23:
### Bug Description

When generating G-code and starting the test print the nozzle and bed start heating up at the same time (good) and it should auto home but it doesn't and when the bed gets to temp the print starts even though the nozzle is not at temp yet. When it starts, it doesn't prime the nozzle, it goes straight to printing the horizontal test lines while cold and in mid-air.

### My Configurations

Config files and my auto-generated G-code from Marlin's website:

[Bug Report.zip](https://github.com/MarlinFirmware/Marlin/files/4366917/Bug.Report.zip)

### Steps to Reproduce

1. Generate G-code using https://marlinfw.org/tools/lin_advance/k-factor.html
2. Save to the flash drive
3. Print from the flash drive
4. Watch the print freak out.

**Expected behavior:** 
Heat hot end and bed at the same time up to the correct temperature and then auto home and start test print.

**Actual behavior:** 
Both bed and hot end start heating, the bed gets to temp first, doesn't auto home, doesn't start the test print correctly.

thiet-m, 2020-03-25:
i have the same problem,no sd card , but doesn't match with pronterface or cura and usb connection

kayazuki, 2020-05-27:
I have a similar problem, but a tiny bit different..
(Printer: Modix Big60 V2)

When i print form the SD card, it does every step ok: It heats up, waits to reach the temperature, then moves to the start XY position and THEN it does something REALLY weird...
When it is supposed to execute the command:
  **G1 Z0.2 F1200 ; Move to layer height** 
i only hear some growl coming from the stepper motors and the bed does not move a mm up to bring the print-bed closer to the nozzle.
And then... It starts to print **mid air**....

I already changed the F1200 to F300 as the print-bed can only move at a max feed rate of 300mm/min (or 5mm/s).
Still the same growl and no Z movement!

My printer is hooked up to a Raspberry PI running OctoPrint.
So I open the Terminal and I started feeding every single line of the code via the Terminal..
Guess what: Then the bed moves up without a problem............?!?!

I'm baffled...
Can anyone help me here? How is that possible??
The printer runs Marlin version 1.1.9B

boelle, 2020-06-21:
@Nibroc-Fitz still an issue?

Nibroc-Fitz, 2020-06-21:
Not really. I started using an SD card instead of a flash drive.

closed 2020-06-21 by Nibroc-Fitz