MarlinFirmware/Marlin #17259: [BUG] UBL over correcting

robwlakes, 2020-03-23:
### Bug Description

I can successfully issue a G29 P1 commands and I get a 5x5 point matrix probed. G29 T0 reports a mesh like this one:

> Send: G29 T0
> Recv: 
> Recv: Bed Topography Report:
> Recv: 
> Recv:     ( 10,130)                      (180,130)
> Recv:         0       1       2       3       4
> Recv:  4 |[+0.179] -0.005  -0.142  -0.380  -0.755
> Recv:    |
> Recv:  3 | +0.244  +0.062  -0.072  -0.235  -0.585
> Recv:    |
> Recv:  2 | +0.313  +0.127  -0.003  -0.162  -0.514
> Recv:    |
> Recv:  1 | +0.336  +0.136  -0.019  -0.185  -0.550
> Recv:    |
> Recv:  0 | +0.348  +0.130  -0.032  -0.249  -0.621
> Recv:         0       1       2       3       4
> Recv:     ( 10, 10)                      (180, 10)
> Recv: 
> Recv: ok

Now this appears reasonable, the center probe has been set using a Z-Offset number 0.17mm. The printer is pretty well mechanically set up with the X-axis gantry level, though not perfect.

With UBL disabled it is possible to get reliable prints in the center of the build board. As the prints get bigger UBL should be much more useful. My test prints of 4 separate discs with a test trace around them (as per latest Cura) shows the print head is too high where X is near the maximum and too close when X is near its minimum. In other words the head has a positive gradient along the X axis. 

 However turning on UBL and using the mesh above does not seem to correct for the uneven build board, but over compensates in the along the X axis. I think the Y axis is correct. The printing thickness appears consistent along any Y line (ie constant X), but along any X line (ie a constant Y) the thickness is the opposite to the print without any UBL. In other words now the head appears to be on a negative gradient along the X axis, not a zero gradient as one might expect if UBL was working properly.

### My Configurations

**Required:** Attached. [Marlin_Configs.zip](https://github.com/MarlinFirmware/Marlin/files/4367312/Marlin_Configs.zip)

### Steps to Reproduce

1. Determine UBL grid and save mesh
G29 P1 to establish grid
G29 T0 to check it
G29 P6 C0.003 to correct grid center to Z0
G29 S0

2. Print without UBL
G28; Make sure we are grounded!!
M851 Z0.17; Set the Z-Offset
G29 L0; Load Bed Topology 0
;M420 L0 Z5 S1; Activate the UBL: Mesh=0, S1=True, Fade=5
M420 S0; switch off UBL, if required for testing
;G29 J ;Do a bed relevel with a quick 3x3 sample

3. Print with UBL
Repeat steps above
G28; Make sure we are grounded!!
M851 Z0.17; Set the Z-Offset
G29 L0; Load Bed Topology 0
M420 L0 Z5 S1; Activate the UBL: Mesh=0, S1=True, Fade=5
;M420 S0; switch off UBL, if required for testing
G29 J ;Do a bed relevel with a quick 3x3 sample

**Photo Evidence:**
[image]
This is a photo of the UBL in action (or not?).

**Expected behavior:** 
The print with UBL should have a brim around the 4 cylinders that is a consistent thickness.
Without UBL the brim around the 4 cylinders will vary depending on the topology of the build board.
 
**Actual behavior:** [What actually happens]
Without UBL activated the print has a distinct thicker side in the higher X values and varies down to nothing in the lower X values.
With UBL activated, the print has a normal brim thickness in the lower X vales and varies down to nearly nothing at the higher X values.
The slope seems to  be over compensated, creating a see-saw effect of the thick and thin areas.

#### Additional Information

I am rebuilding a Robox printer that originally had an ARM processor with a Mega2560+RAMPS set up. I am happy that I have the mechanics and electronics sorted, but this problem of the UBL not being able to minimise the topology of the build board is very frustrating. I originally pulled the printer apart so I could use open source electronics and firmware, and rescue a printer that had become too expensive to repair with their proprietary parts. It originally had two print nozzles with needle valves that were swapped in the head with a stepper. I have swapped with an ultra simple single nozzle setup.

It uses the nozzle as the actuator of the Z-stop detector.  The print head swings on a single main rod, and a second thinner rod has a sliding contact on it that stops the head falling onto the build bed. When the nozzle contacts the build bed the contact with the second rail is lost and the circuit opens and the Z-Stop switch is activated. So there is a absolute 1:1 correspondence between the sensing system and head position. There is a plastic leaf spring that presses the head's contact against the second thinner rod. This must be strong enough to hold the head steady when printing, but not so strong that the pressure to activate the Z-Stop switch is too high.

[image]

The build board does not have any adjusting screws for levelling. It slides with two bearings at the low X end along a steel rod.  The other end travels on a single skid at the high end of the X travel, forming a triangle support. The Z-Axis has dual motors, one at either end, and these can be manually used to get the board parallel to the X-Axis steal rod and hence the X travel of the head.  The pringiness of this system is a major concern, but I don't know how to quantify its effect.

I apologise in advance for the length of this report and the unusual set up.  The system worked really well under their electronics and software, so I feel I should be able to get similar results if I tune it up well enough. Hence my obsession with getting UBL to work properly on the system. i wish I knew how to get mono-spaced fonts to work for the mesh table as well.

Many thanks if you have any suggestions or assistance you can provide me.
Rob Ward

tvixen, 2020-03-23:
Hi Rob
Please try the latest version, as this version has been fixed up. Not saying it will solve the whole problem, but some of it might.

robwlakes, 2020-03-26:
Hi Tvixen,
I have experimented with the latest bug -fix version and had some success. I believe I have traced the error to my mechanical set up. The Marlin 2.0.4 may have been Ok, and my set up was throwing out of whack. However I swapped to the bug-fix and it did not make much difference. I then suspected that the weight of the wires to the head could have been placing too much pressure on the switching action and making the head press much harder into the base to switch off/on/off.

I repositioned the cabling and found I got an improvement. I hung the cables from the top of the back of the printer, rather than just let them drag along. This reduced the weight leverage and made the switch more sensitive. I ran the UBL mesh probing again, and the print improved.  Encouraged by this I removed a plastic strip I had made to replicate the leaf-spring the original design used to stiffen the switch action, so I could further increase the switch sensitivity. This made considerable difference, I ran the UBL mesh probing again, and the there was a marked improvement.

Recv: Bed Topography Report:
Recv: 
Recv:     ( 10,130)                      (180,130)
Recv:         0       1       2       3       4
Recv:  4 | -0.537  -0.353  -0.168  -0.059  +0.041
Recv:    |
Recv:  3 | -0.468  -0.275  -0.070  +0.085  +0.240
Recv:    |
Recv:  2 | -0.417  -0.204 [-0.000] +0.178  +0.341
Recv:    |
Recv:  1 | -0.373  -0.195  -0.006  +0.162  +0.304
Recv:    |
Recv:  0 | -0.365  -0.202  -0.027  +0.118  +0.254
Recv:         0       1       2       3       4
Recv:     ( 10, 10)                      (180, 10)

The topology seemed to be more consistent in the way it sloped. It no longer had a saddle type shape at the high X values. The base board has a certain amount of springiness that a heavy action switch would have amplified and cause the probing at the (high X, low Y) and (high X, high Y) areas where there is no direct support under those corners. So a lighter switching action would have been definitely an improvement.  The UBL documentation does not have any warnings about heavy action Z-Stop/Probe switches, and probably more critical, how springiness of the board can influence readings. I guess people, just like me, assume there is not much interaction. However I can't make the action too light or it will be an unreliable contact and the head will move up and down when oozing plastic and not be a constant height.

While these experiments were going on I also got another unusual problem arising. The top end of the print was being chopped off at about X=187, the bed size is defined as X=190 (about 10mm in from the edge for safety), so I guess it is detecting going off the bed.  What I was not confident about was that the head was lowering to the bed and scraping around the right shape instead of doing a safety lift as I thought was defined in this line in UBL definitions. The Z-Axis motor was definitely jumping about. See photos below.

  #define UBL_Z_RAISE_WHEN_OFF_MESH 2.5   //RWA  Dec2019
                                          // When the nozzle is off the mesh, this value is used
                                          // as the Z-Height correction value.
My nozzle was going down, when I expected it to be raised? I have 10mm margin inset for the UBL mesh so its maximum should have been X=180, not 190. Confusing, any ideas?

[image]
and
[image]

One or two or more questions. When does the Z-Offset get applied? Does Marlin use it when probing the topology or are they absolute (with respect to the Z-Axis switch) values reported.  Is Z-Offset applied to G30? If UBL is active is it applied to G30?

Just completeness sake I will add a picture of the side on view of my adapted head spacer (light blue) with the black X-Axis + Bearing.  The thicker lower rail is the main stabilizing bar. The belt attaches to the this. The thinner top rail is the stabilizing guide rail and also the bar that the switch/contact makes contact with.  It is a simple metal "nose" pressing onto the back of the rail. As the nozzle presses on the plate the top of the print-head is pushed back and the circuit is broken. Simple enough, but obviously does pose tuning problems.

[image]

I think you may have help solve my UBL "problem" and in turn it may highlight where some extra documentation could be added. I can see though that if most people are using the very light action BL-Touches then it is probably not that relevant.

Cheers and thanks for the assistance,

Rob Ward

tvixen, 2020-03-26:
Ur welcome

Roxy-3D, 2020-03-26:
You should consider enabling the G26 mesh validation command and print a pattern.
If any mesh point is too high or too low, you can quickly edit it with a G29 P4 command.

Sadly, just probing the bed  does not result in fully trusted measurements.   One example would be a slight screw in how level the carraige is as it travels from one side of the X axis to the other side.   With any Y_OFFSET_FROM_EXTRUDER the Z Height measured will change.

There are a lot of different mechanical imperfections that can factor into different parts of the mesh not being fully accurate.    That is why the G26 mesh validation pattern exists and why the mesh points are easily edited with a G29 P4.

robwlakes, 2020-03-26:
I do have G26 enabled and preset to my normal values PLA@55/200/190 I feel confident enough now to try it. I had vision of it gouging its way across the build bed, but it should be OK now. Close enough that no major accident is likely. I will try it and post my findings.

I marvel at what you guys can do with projects like Marlin, and as a I know enough to appreciate what you do, but not enough to be able to really understand the detail, it is easy to have stars in my eyes and think the software will "fix every quirk in every system". Maybe the documentation should give people a reality check here and remind them that the screwdriver and cutters are also their best friends ;-).

I don't want to be a pest but could you help me understand what the 
  #define UBL_Z_RAISE_WHEN_OFF_MESH 2.5 //RWA Dec2019
does. In my experiments it appeared to work in reverse, ie going down 2.5mm rather than up?

And if I do a G30 to probe a point can I turn UBL off for one reading and turn it back on for a second reading? Or is it just raw reading each time?

Once again thanks you folk's work. Very much appreciated.

Rob Ward

Roxy-3D, 2020-03-27:
> I do have G26 enabled and preset to my normal values PLA@55/200/190 I feel confident enough now to try it. I had vision of it gouging its way across the build bed, but it should be OK now. Close enough that no major accident is likely. I will try it and post my findings.

Both G29 P1 and G26 were written to be very protective of the user's bed (and surface).   Where ever the command is given, it works outward from that point.   That is deliberate.   And people have argued it is (a tiny bit) faster to just scan across the bed.   That idea has been rejected because it adds risk to the user's equipment.   

At any time during the G29 P1 or G26 you can press and hold the encoder wheel and abort the command if the nozzle is getting too close to the bed.    You should have time to stop it before anything bad happens because the 'expanding outward' pattern lets you see that every thing was OK at previous points and the bed isn't going to change that much, that fast.

robwlakes, 2020-03-27:
Yes you are correct Roxy-3D, I have run the test G26 and completed it without any catastrophic consequences as you predicted. The results though are not that good (se photo).  there appears to be some other part I have not configured. It appears it has not been doing any retracts at all? Printing over the top of existing lines and having blobs on the ends.

Very Messy :-) but there are quite few circles that are printed correctly. Where am I up to in getting this totally under control?

[image]

This is the second attempt as I aborted the first one an lowered the nozzle (Z-Offset from 0.17 to 0.12) to help it adhere better.  I did this through the LCD and saved the value back to EEPROM. The plastic did not adhere any better.
[image]
Just to reassure it does print reasonably well, though lower temp or more retraction maybe in order. Still needs a good bit of tuning.

Cheers and thanks,
Rob Ward

Roxy-3D, 2020-03-28:
It looks to me like the Z_PROBE_OFFSET_FROM_EXTRUDER needs to be made a little bit smaller.  Maybe by .2mm but you will have to check that and see.  (And assuming it is a normal probe and already has a negative number...   Increase its magnitude by .2mm and keep it a negative number!)

Once you get a better G26 print where most of the pattern is actually adhering to the bed, please post the picture.    I'll be happy to offer an opinion about what I see and how I would proceed if it was me.

Roxy-3D, 2020-03-28:
That 3rd attempt does look good.   Everything is uniform and has good adhesion everywhere.

robwlakes, 2020-03-30:
Hi Roxy-3D,
I have got a fair bit of the way along the track. I don't have my "pin-up" G26 print, but I have just about got a full print.
[image]

You can see from the image above that the Robox has a little silicon trough on the lower LHS and this is used as a purging/wiping area. I have got that tuned up as I was finding it hard to get the first 2-3 G26 lines printed because the nozzle was not primed right. I have since added a priming strip like in the Linear Advance calibration gcode (end in no retraction). It draws the line ok after the priming, but then stops, has a think and then launches into the G26 with a very weak filament flow, and I miss the first 2-3 lines (lower LHS on image above).  I have become somewhat obsessed with recovering this original Robox feature, somehow Robox could land a print 95% of the time on the board and level.

I have tried to use the O (ooze, 10,5 or 0) and P (prime0,5 or10)) settings on G26 in different combinations, and I think I can make them work, but they all seem to fire off before the G26 has a think (maybe waiting for temperatures, as it does have its own settings eg 55/190 so I set the purge/wipe intro script's temperatures to 56/191 hoping it would speed up the start, but each time it drops a bundle of filament, squashes into it and then does not have enough to begin G26 solid lines.  
[image]
You can see in the image above where it has tried to print the lines but very weakly.

Any suggestions/ideas?

Overall I am very grateful, and on a scale of 1 to 10 this is probably me just nit picking the last few details, as the original UBL is definitely working and the G26 provide good evidence of that.

Cheers and thanks again, Rob.

PS when does M420 read the Z-Offset value? Can the Z-Offset value be updated by M851after the M420 or should it be done before?

robwlakes, 2020-04-09:
Hi Guys,
I have to add another couple of notes to this. And first let me begin by thanking your support, and and for your patience.

I felt I had to somehow find another way of verifying how level my bed was outside of the UBL. So I was running 3 GCODE snippets from OctoPrint that would let me
1. move the head to the center of the bed with a G28, then I used the Jog controls to move the head up until I could do the paper "feeler gauge" trick, the snippet zeroed the Z height.
2. move the head to the RHS and drop the head to Z=0, and turn off the steppers, then I did the paper trick by manually moving the RHS stepper at that point, then
3. repeat dropping the head to Z=0  again at the LHS, then return to 2 for the RHS etc
I tried a large number of times and could not get it to level. The software seemed to be 100% fine so I very reluctantly looked at mechanical problems (my source of spares is very limited). And I found the LHS Z-Axis thread has been seriously stripped and was totally unreliable.

[image]

I think one time when I tried to get the two motors to have separate drivers so I could do this thing automatically, one side took off and the RHS rose way above the LHS and if this did not directly cause all of it, it probably completed the stripping of the thread.

I won't close this thread as "Solved" just yet as I would like to fit the new motor+thread that is on order and held up now over Easter, get it working and then confirm that the UBL is good to go. It has probably been trying to compensate for the bodgey travel over the stripped area, as it was near the Z=0 area (where the most wear would occur anyway, as that end does have the extra weight of the X-Axis motor, so it may not have been completely my negligence that it was worn and damaged. The fact the business I bought it off still had a spare motor+thread in stock after not being dealer for the Robox for 2 years said something to me.) Somehow it still managed to print that lovely Benchy pretty good, amazing really.

I don't think UBL was designed to compensate for this sort of problem haha! But I suspect it did its best.

Cheers, Rob

PS The thread is interesting: pitch 0.05mm and diameter 5mm. Not easy to source.

boelle, 2020-06-20:
@robwlakes is the issue still there?

robwlakes, 2020-06-21:
Hi boelle;

I don't think so, as I can requests a 5x5=25 point probe and completes all probing and I can save the correction matrix and it is used (I can see the Z axis adjusting to different parts of the build plate on a big print). I think the improper scanning initially was caused by my mistakes in configuring the size of the build plate and how the safe area for probing was defined within that area.

I have successfully transplanted the replacement Z-Stepper and now small adjustments to the Z-Offset are predictable (ie small increments either way are now reliable). So that has answered most of my concerns.

One of my original concerns persists though. When the UBL probe ends all 25 points, it returns a 
_""  Unknown Command_
statement. So I feel these is still a weed left in there some where that needs to be be pulled out. It may only be a cosmetic tidy up of the code, but who knows? It is a little rattling when trying to get other things right when it reports an "error" no matter what.

My probe is the tip of the print nozzle. The X-Gantry is really on a single track, a rod, and the head can swivel on it. There is a second, thinner rod, that stabilizes it and there is a sliding contact on this thinner rod that is the lower Z-Limit switch. So while printing, the head is pressed against the upper thinner rod and that keeps the head level. When the nozzle and head is pressed pressed hard enough against the build plate the head swivels and Z-Limit switch opens and it flags the presence of the bed. This appears to be a no nonsense solution, but it has several drawbacks. The most critical one is the balance between having a strong enough pressure to keep the print head steady, (ie not have the pressure of the exuding plastic lift is and influence layer height), and to not have too much pressure that when probing the bed it does not flex the bed too much to get the switching effect. The bed is supported in a triangle so there are two corners that get support only through the stiffness of the rest of the system, which is not always enough.

This long winded question (apologies) raises this point: Does the probing of the bed during UBL work on the point at which Z-LowerLimit is switched on (ie max pushing down) or when the probe reconnects (ie just lifted off the board). I have probed it with M48 and the bed probing system has a high enough accuracy and repeatability, but taking the measurement with the least pressure would be preferable. Unfortunately I don't know how to determine the significance of any hysteresis, I maybe grasping at straws. I have seen printers sold recently with pressure sensors in the head to detect using the nozzle so there must be some answers somewhere.

In short though I have been using standard UBL for about 2 months now and it has been set and forget. The work on it is very much appreciated.
 
Any hints, URL etc would be appreciated.

Cheers, Rob

closed 2020-06-23 by boelle