MarlinFirmware/Marlin #17261: [BUG]Enable the G26 Mesh Validation Pattern tool. Can't change values

tvixen, 2020-03-23:
/**
   * Enable the G26 Mesh Validation Pattern tool.
   */
  #define G26_MESH_VALIDATION
  #if ENABLED(G26_MESH_VALIDATION)
    #define MESH_TEST_NOZZLE_SIZE    0.4  // (mm) Diameter of primary nozzle.
    #define MESH_TEST_LAYER_HEIGHT   0.2  // (mm) Default layer height for the G26 Mesh Validation Tool.
    #define MESH_TEST_HOTEND_TEMP  100    //205 (°C) Default nozzle temperature for the G26 Mesh Validation Tool.
    #define MESH_TEST_BED_TEMP      27    //50 (°C) Default bed temperature for the G26 Mesh Validation Tool.
    #define G26_XY_FEEDRATE         20    // (mm/s) Feedrate for XY Moves for the G26 Mesh Validation Tool.
    #define G26_RETRACT_MULTIPLIER   1.0  // G26 Q (retraction) used by default between mesh test elements.
  #endif

### Bug Description

<Trying to change parameters in "G26 Mesh Validation Pattern tool" -->

### My Configurations

 /**
   * Enable the G26 Mesh Validation Pattern tool.
   */
  #define G26_MESH_VALIDATION
  #if ENABLED(G26_MESH_VALIDATION)
    #define MESH_TEST_NOZZLE_SIZE    0.4  // (mm) Diameter of primary nozzle.
    #define MESH_TEST_LAYER_HEIGHT   0.2  // (mm) Default layer height 
    #define MESH_TEST_HOTEND_TEMP  **100**    //205 (°C) Default nozzle temperature for the G26 
    #define MESH_TEST_BED_TEMP      **27**    //50 (°C) Default bed temperature for the G26 Mesh 
    #define G26_XY_FEEDRATE         20    // (mm/s) Feedrate for XY Moves for the G26 Mesh 
    #define G26_RETRACT_MULTIPLIER   1.0  // G26 Q (retraction) used by default between mesh test 
  #endif

### Steps to Reproduce

<Test of Bed_Level validation Pattern tool in display>

1. With UBL enabled in software do:
2. Run the Unified bedlevel process from the display [Step-By-Step UBL] guide
3. In step 3 "Validate Mesh" the Bed_Temp and HotEND_Temp is set to 50C and 190C regardless the value set under  #Define G26_MESH_VALIDATION (in Configuration.h) 

**Expected behavior:** 
1. With UBL enabled in software do:
2. Run the Unified bedlevel process from the display [Step-By-Step UBL] guide
3. In step 3 "Validate Mesh" the Bed_Temp and HotEND_Temp is set to 27C and 100C as default values set with #Define G26_MESH_VALIDATION (in Configuration.h) 

#### Additional Information
I have not tested the rest of the parameters, but it seems like these values are NOT parsed into G26.cpp

qwewer0, 2020-03-23:
So, you want to print G26 with 100°C nozzle temperature?

MAybe there should be a sanity check for G26 nozzle temp to not be lower than EXTRUDE_MINTEMP.

tvixen, 2020-03-23:
Well this is just a test, actually im not printing. And thats not the real issue here. The issue is that the parameters is NOT transferred to G26.cpp

qwewer0, 2020-03-23:
If MESH_TEST_HOTEND_TEMP is lower than 165°C, then you should get a "Specified nozzle temperature not plausible." message.

tvixen, 2020-03-23:
Are u guessing, or did u do the test ?

No there's no error messages.

qwewer0, 2020-03-23:
Not on the printer but on the serial connection. `SERIAL_ECHOLNPGM("?Specified nozzle temperature not plausible.");`
I guess it sets it back to the default if the temp isn't plausible.

tvixen, 2020-03-23:
The test is not done by USB but via the display. 
And as I wrote, this is only when you "Enable the G26 Mesh Validation Pattern tool" and try to change the parameters.

boelle, 2020-06-20:
@tvixen is the issue gone?

tvixen, 2020-06-20:
I have't tested with the 2.0.5.3. version, as I had to switch to Mesh bed level. But nobody seems to have made any corrections to the code.

qwewer0, 2020-06-21:
Test it with the latest bugfix.

closed 2020-06-21 by tvixen