MarlinFirmware/Marlin #17264: [BUG] Y Layer Shift when Skew Value is Set

midopple, 2020-03-23:
### Bug Description

When printing objects with dual nozzles, the Y-layer shift minimal with every Tool Change
The shift ist only when the skewfactor.xy != 0

### My Configurations
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4370544/Configuration.zip)
Rambo V1.1 Board
Dual X System
2 Toolheads
Bedlevel UBL

### Steps to Reproduce

1. Objekt print with booth Toolheads, Modus M605 S0 Fill Control
    Toolchange/Parking come from Simplify3d
2. Set Skew bigger then 0 --> M852 S0.002
3. Print it

Objekt look like this

[image]

**Expected behavior:** [What you expect to happen]
straight sides

**Actual behavior:** [What actually happens]
Layer shift every Toolchange, the size from the skewfactor dont change the size of Layershift
[image]

#### Additional Information
Ther BUG look like the Issue #14022

thinkyhead, 2020-03-24:
I would not be surprised that skew is not fully considered when switching tools. We do disable and re-enable bed leveling, but the planner's `apply_modifiers` and `unapply_modifiers` should probably also be used at the same points. I'll post a PR that you can test soon.

midopple, 2020-03-26:
I found one part in the Software
unskew() is calles one time in FORCE_INLINE static void unapply_modifiers(xyze_pos_t &pos
and also in the funktion void Planner::unapply_leveling(xyz_pos_t &raw) and this function is also startet from unapply_modifers so unskew() is calles twice

thinkyhead, 2020-03-29:
@midopple — That's it, I'm sure. With the patch now applied, is it any better?

midopple, 2020-03-30:
With this Patch is is better but  a minimal offset can still be seen. I measured it over the component height of 50mm and it looks as if 1 step is lost per layer.
I found a reason in the function void Planner::set_machine_position_mm
when i change 
LROUND(a * settings.axis_steps_per_mm[A_AXIS]) to (long)FLOOR(a * settings.axis_steps_per_mm[A_AXIS]) i had no Layershift

This is not the solution but i think there is an eroor in Rounding a Position set aber Toolchange

midopple, 2020-04-05:
Hi

I found an Roundig error in the funktion Planner::set_machine_position_mm
here to examples with different Y Position
my Skew funktion worK on the Y Axis 

Case 1
tool T0 to T1
Y= 244,876
Y-Offset = 0,3
Ynew = 244,576 (- Yoffset)
X T0 = 1
X T1 = 475
Skew = 0,00254
Step Y T0 = 18782
Step Y T1 = 18666
Diff = -116 Steps

case2 T0 to T1 (other Position)

tool T0 to T1
Y= 278,124
Y-Offset = 0,3
Ynew = 277,824 (- Yoffset)
X T0 = 1
X T1 = 475
Skew = 0,00254
Step Y T0 = 21332
Step Y T1 = 21217
Diff = -115 Steps

So it depend on the Y Position on Toolchange 
sometimes are 116 Steps and Sometimes 115 steps
so i has an minimal Layershift

midopple, 2020-04-05:
Could that be the reason for the offset or am I going the wrong way here

boelle, 2020-06-20:
@midopple is issue gone?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

commit fc11e721 (2020-03-29): Fix extra unskew call

Fixes #17264
closed 2020-06-23 by boelle