MarlinFirmware/Marlin #17265: [BUG] Speed fluctuation if M203 E values ​​are less than printing speed

midopple, 2020-03-23:
### Bug Description
If you limit the maximum extrusion rate using the setting M203 E2, this has a retroactive effect on the printing speed. This means that printing does not take place faster than the material can be extruded. With round objects with many segments, this has the error that they are printed at different speeds and generate a wave pattern

With M203 E80 the surface get smooth

[image]

### My Configurations
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4370681/Configuration.zip)

Rambo V1.1 Board
Dual X System
2 Toolheads
Bedlevel UBL

and Prusa I3
with original Firmware

### Steps to Reproduce

1. Objekt Zylinder with fine Resolution, slice with Simplify3D and Printspeed 80mm/s
2. Set the Value M203 E2 --> Max e Speed 2mm/sec
3. Print it
4 . Objekt has the Waves

I have send Debug Information from Planner for every Move with the calculated Speed
and the Speed change from Segment to segment

I also test the Bufferunderrun, the buffer is always over 12 Moves so
this part did'n reduce the speed

Objekt look like this

[image]

**Expected behavior:** [What you expect to happen]
ssmooth areas

**Actual behavior:** [What actually happens]
Waves syncron with the Segments from G-Code files

#### Additional Information

MoellerDi, 2020-03-23:
I'm currently working on another way to limit the maximum extrusion rate w/o affecting retraction speed. It would be great if you could test PR #17017 with the following settings (in addition to you own) to see if it makes any different for you.

//#define SLOWDOWN (disabled)
#define MIN_STEPS_PER_SEGMENT 1

boelle, 2020-06-20:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-20 by boelle

midopple, 2020-06-21:
Hi, 
at the moment i have no solution for this issue.
Menwhile i did not use the Parameter for MAX Extrusion Speed (M203 E) so i dont have the problem.