MarlinFirmware/Marlin #17268: [BUG] Change Filament: printing does not resume at the right place

Bergerac56, 2020-03-23:
With bugfix 2.0.x (no specific version. At least the last one) Change filament seems not to work as I would expect. After a filament change, I would expect that the printer rresume at the last position in order to continue the print.

But it is not the case. Here is the result:
[image]

After investigation, it seems related to the parameters of NOZZLE_PARK_FEATURE. More specifically: #define NOZZLE_PARK_POINT { (X_MIN_POS + 10), (Y_MIN_POS - 10), 20 }. It seems to influence the calculation of the "return" position after filament change. The above picture is a test using those parameters (default parameters of the configuration.h file ).

Even if it would be logic that those parameters would not influence the repositionning of the noozle after changing filament, it seems it does.

To see if it could be improved, I tried this: #define NOZZLE_PARK_POINT { 0, 0, 20 }  with this result:
[image]

It is not yet perfect (very small displacement). But it is already a lot better.

Here are my config files: 
[config.zip](https://github.com/MarlinFirmware/Marlin/files/4371593/config.zip)

I do not know if this behavior is normal (for me it is not) and if this worked before as I never used this function before. I discover the issue after an unexpected filament runout...

Do I miss something?

MarkMan0, 2020-03-26:
Are you sure Y_MIN_POS - 10 is correct? That would mean overt the endstop.
The stepper may time out and get disabled. Try without stepper timeouts

Bergerac56, 2020-03-26:
@MarkMan0 I would say that I am nearly sure it is not correct (at least for my printer) But  that were the default values. I never modified those parameters until the problem with "change filament". Anyway, now that I discovered their existance, I replaced them by {0,0,20} (as my printer parks the head at 0,0)

If this causes the wrong resume, I am just puzzled why those parameters have an influence on it. My understanding is that the head would go back (provided there were no manual moves of the head) to the exact position it left for the filament change. And that, nevertheless any "home" position. 

And it does not explain why the resume is slightly shifted to a very tiny x+something, y+something. I will experiment with very small values for the park position and see if that has an influence.

Bergerac56, 2020-04-28:
I do not believe that this was solved. 
But for me, and to avoid some friendly reminder from our friend Bo :), I adapted #define FILAMENT_RUNOUT_SCRIPT "M600" by adding the desired home position "M600 X0 Y0 Z10"
Till now it does the job. I will then close the issue.

closed 2020-04-28 by Bergerac56