MarlinFirmware/Marlin #17270: [FR] During G28, go to X0 before homing Y

JoshWelsh, 2020-03-23:
### Description

Using my HEVO with sensorless homing, I have "X_MIN_POS" set to "-11" and I would like the homing of X to complete only after the extruder is moved back to X=0.  Currently when the printer homes X, it sets X=-11 and then homes Y, which causes collision between my part cooling fan ducting and some hard parts for the Z axis linear rods.

Basically, go to X=0 before doing anything else in G28.

### Feature Workflow

1. Issued G28
2. X is homed
3. X at home is then set to X_MIN_POS
4. NEW: X goes to X=0
5. G28 carries on and does homing for Y, and so on

#### Additional Information

Hopefully this all makes sense and is simple, but I can provide pics if needed.

thinkyhead, 2020-03-23:
See if this helps:
```gcode
#define HOMING_BACKOFF_MM { 11, 2, 2 }
```

rado79, 2020-03-23:
You should try to set HOMING_BACKOFF_MM to 11 for X axis

rado79, 2020-03-23:
@JoshWelsh For best results getting help with configuration and troubleshooting, please use the following resources:

MarlinFW.org Marlin Documentation
RepRap.org Marlin Forum
Tom's 3D Forums
Facebook Group "Marlin Firmware"
Facebook Group "Marlin Firmware for 3D Printers"
Marlin Configuration on YouTube
Marlin Discord server. Join link: https://discord.gg/n5NJ59y
After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a bug report.

thinkyhead, 2020-03-23:
@rado79 — The proper canned reply is in Markdown and contains hyperlinks. Since the issue is resolved there's no need to refer the user to other forums for further help.

JoshWelsh, 2020-03-24:
Has anyone reported that `HOMING_BACKOFF_MM` is bugged yet?  I ask because if I enable as discussed, it skips the second axis and goes right to Z.  What I mean by this is normally it'll home X, bump back, pause, then go home Z apparently based on the current position being Y0.  If I `#define HOME_Y_BEFORE_X` then it'll home Y, then assume current X is home and proceed to home Z.  This is pretty dangerous as it can crash the machine by shoving the carriage out of bounds.  If you want this to be a support ticket I'm happy to go down that path, but if this hasn't been reported as a bug and addressed yet, I'd like to explore that.  FWIW This is the line I added:
`#define HOMING_BACKOFF_MM { 11, 10, 2 }`

JoshWelsh, 2020-03-24:
Also, G29 doesn't respect X_MIN_POS either.

thinkyhead, 2020-03-24:
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information so that we're not just taking stabs in the dark. Here is the boilerplate:

- Download Marlin `bugfix-2.0.x` to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Do some of the moves that revealed problems before. Take notes.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

rado79, 2020-03-24:
@JoshWelsh try with latest bugfix-2.0.x. Last night changes have been made for homing backoff with core kinematics.

Geitde, 2020-03-24:
@JoshWelsh 

I think it is not skipping homing the second axis, but performing a false positive on the TMC sensor.

Please check this for a possible solution:

https://github.com/MarlinFirmware/Marlin/issues/17213

JoshWelsh, 2020-03-26:
Apologies for the delay, and honestly I very much appreciate the effort everyone puts in to this product, so thank you!  if this turns out to be a config issue on my side, I'm happy to be the guy that stands up and says "my bad, I'm an idiot."  That said, the travel moves are clearly going to off-limits region in `bugfix-2.0.x` as reported, you can see in the below log (I made the first handful of instances bold for easy reference)

Recv: <<< do_blocking_move_to
Recv:   current_position= X119.00 Y122.67 Z8.73 : Probe::set_deployed
Recv: deploy: 1
Recv:   current_position= X119.00 Y122.67 Z8.73 : >>> Probe::run_z_probe
Recv:   current_position= X119.00 Y122.67 Z8.73 : >>> Probe::probe_down_to_z
Recv: BLTouch DEPLOY requested
Recv: BLTouch Command :10
Recv: bltouch.deploy_proc() end
Recv: >>> do_blocking_move_to X119.00 Y122.67 Z1.55
Recv: <<< do_blocking_move_to
Recv: echo:busy: processing
Recv:  T:199.87 /200.00 B:56.88 /57.00 @:79 B@:127
Recv: BLTouch STOW requested
Recv: BLTouch Command :90
Recv: bltouch.stow_proc() end
Recv:   current_position= X119.00 Y122.67 Z3.99 : sync_plan_position
Recv:   current_position= X119.00 Y122.67 Z3.99 : <<< Probe::probe_down_to_z
Recv:   current_position= X119.00 Y122.67 Z3.99 : <<< run_z_probe
Recv: >>> do_blocking_move_to X119.00 Y122.67 Z8.99
Recv: <<< do_blocking_move_to
Recv:  T:199.87 /200.00 B:56.93 /57.00 @:69 B@:127
Recv: echo:busy: processing
Recv: <<< Probe::probe_at_point
Recv: >>> Probe::probe_at_point(242.00, 84.67, raise, 0, probe_relative)
Recv:   current_position= X119.00 Y122.67 Z8.99 :
Recv: >>> do_blocking_move_to X184.00 Y122.67 Z8.99
Recv: <<< do_blocking_move_to
Recv:   current_position= X184.00 Y122.67 Z8.99 : Probe::set_deployed
Recv: deploy: 1
Recv:   current_position= X184.00 Y122.67 Z8.99 : >>> Probe::run_z_probe
Recv:   current_position= X184.00 Y122.67 Z8.99 : >>> Probe::probe_down_to_z
Recv: BLTouch DEPLOY requested
Recv: BLTouch Command :10
Recv: bltouch.deploy_proc() end
Recv: >>> do_blocking_move_to X184.00 Y122.67 Z1.55
Recv: <<< do_blocking_move_to
Recv: echo:busy: processing
Recv:  T:199.87 /200.00 B:57.14 /57.00 @:81 B@:127
Recv: BLTouch STOW requested
Recv: BLTouch Command :90
Recv: bltouch.stow_proc() end
Recv:   current_position= X184.00 Y122.67 Z4.26 : sync_plan_position
Recv:   current_position= X184.00 Y122.67 Z4.26 : <<< Probe::probe_down_to_z
Recv:   current_position= X184.00 Y122.67 Z4.26 : <<< run_z_probe
Recv: >>> do_blocking_move_to X184.00 Y122.67 Z9.26
Recv: <<< do_blocking_move_to
Recv:  T:200.00 /200.00 B:57.34 /57.00 @:58 B@:127
Recv: echo:busy: processing
Recv: <<< Probe::probe_at_point
Recv: >>> Probe::probe_at_point(242.00, 157.33, raise, 0, probe_relative)
Recv:   current_position= X184.00 Y122.67 Z9.26 :
Recv: >>> do_blocking_move_to X184.00 Y195.33 Z9.26
Recv: <<< do_blocking_move_to
Recv:  T:200.00 /200.00 B:57.60 /57.00 @:58 B@:0
Recv: echo:busy: processing
Recv:   current_position= X184.00 Y195.33 Z9.26 : Probe::set_deployed
Recv: deploy: 1
Recv:   current_position= X184.00 Y195.33 Z9.26 : >>> Probe::run_z_probe
Recv:   current_position= X184.00 Y195.33 Z9.26 : >>> Probe::probe_down_to_z
Recv: BLTouch DEPLOY requested
Recv: BLTouch Command :10
Recv: bltouch.deploy_proc() end
Recv: >>> do_blocking_move_to X184.00 Y195.33 Z1.55
Recv: <<< do_blocking_move_to
Recv:  T:200.18 /200.00 B:57.86 /57.00 @:49 B@:0
Recv: echo:busy: processing
Recv: BLTouch STOW requested
Recv: BLTouch Command :90
[... 207 lines cut ...]
Recv:  2 -0.428 -0.178 +0.150 +0.473
Recv:  3 -0.730 -0.410 -0.098 +0.233
Recv: 
Recv: G29 uncorrected Z:13.55
Recv:  corrected Z:13.51
Recv:   current_position= X184.00 Y268.00 Z13.51 : sync_plan_position
Recv: X:184.00 Y:268.00 Z:13.51 E:0.00 Count A:36160 B:-6720 Z:5420
Recv: <<< G29
Recv: ok
Recv:  T:200.00 /200.00 B:57.45 /57.00 @:51 B@:0
Recv:  T:200.00 /200.00 B:57.34 /57.00 @:54 B@:0
Recv:  T:200.00 /200.00 B:57.29 /57.00 @:55 B@:0
Recv:  T:200.00 /200.00 B:57.19 /57.00 @:53 B@:0
Recv:  T:199.87 /200.00 B:57.14 /57.00 @:89 B@:0
Recv:  T:199.87 /200.00 B:57.03 /57.00 @:88 B@:0
Recv:  T:200.00 /200.00 B:56.98 /57.00 @:57 B@:0
Recv:  T:199.87 /200.00 B:56.93 /57.00 @:84 B@:127
Recv:  T:199.87 /200.00 B:56.93 /57.00 @:66 B@:127
Recv:  T:200.00 /200.00 B:56.98 /57.00 @:55 B@:127
Recv:  T:200.18 /200.00 B:57.14 /57.00 @:7 B@:127
Recv:  T:200.00 /200.00 B:57.34 /57.00 @:57 B@:127
Recv:  T:200.18 /200.00 B:57.60 /57.00 @:19 B@:0
Recv:  T:200.00 /200.00 B:57.81 /57.00 @:90 B@:0
Recv:  T:200.00 /200.00 B:58.02 /57.00 @:70 B@:0
Recv:  T:200.00 /200.00 B:58.07 /57.00 @:91 B@:0

JoshWelsh, 2020-03-27:
@thinkyhead curious if you had thoughts on this yet?  Would very much like if G29 stayed in bounds..

JoshWelsh, 2020-04-15:
Is this thing on?

AnHardt, 2020-04-15:
If you need to home x before y - how about 
```
G28 X
G0 X0
G28 Y
```

JoshWelsh, 2020-04-15:
> If you need to home x before y - how about
> 
> ```
> G28 X
> G0 X0
> G28 Y
> ```

Hi, how does this address let alone solve the issue of G29 not respecting minimum positions?

thinkyhead, 2020-07-03:
I've been under a pile of issues and trying to steal time to fix some things and work on the release. Did you figure out your issue yet?

From a casual reading of your log, it looks like you configured the mesh to start at X = -11 and maybe not with a usable mesh inset within the bed.

Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

gec100, 2020-08-30:
I can confirm that with the latest version 2.0.6.1, G29 still does not respect head min positions.

log is attached

it is visible from the log that in my case first G29 point is at X25.00 Y-20.00. my head min positions are as follows
```
// Travel limits (mm) after homing, corresponding to endstop positions.
#define X_MIN_POS -1
#define Y_MIN_POS -20
#define Z_MIN_POS 0
#define X_MAX_POS X_BED_SIZE
#define Y_MAX_POS Y_BED_SIZE
#define Z_MAX_POS 300
```
my NOZZLE_TO_PROBE_OFFSET is -15, 64, -5.15

my PROBING_MARGIN is set to 10

so, if I understand correctly how this should work, my first G29 point should be ABS(X_MIN_POS) + PROBING_MARGIN, ABS(Y_MIN_POS) + PROBING_MARGIN. In my case that would be X11 Y30 on the head 
or X26 Y94 on the probe

thx for the hard work to all!
[log.txt](https://github.com/MarlinFirmware/Marlin/files/5146489/log.txt)

gec100, 2020-08-30:
setting PROBING_MARGIN_FRONT in advanced configuration can fix this but I still think G29 should respect min positions

sjasonsmith, 2020-08-31:
@gec100 it should already be respected. Post your configuration files and maybe that can help spot a problem.

gec100, 2020-09-01:
@sjasonsmith here they are :)

[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/5154640/Configuration.h.txt)
[Configuration_adv.h.txt](https://github.com/MarlinFirmware/Marlin/files/5154641/Configuration_adv.h.txt)