MarlinFirmware/Marlin #17275: [BUG] UBL Incorrect Z compensation

GibsDev, 2020-03-24:
### Bug Description

During (at least the) first layer of print, UBL compensation varies enough to produce an usable first layer.

### My Configurations (and other files)

[Files-and-config.zip](https://github.com/MarlinFirmware/Marlin/files/4373096/Files-and-config.zip)

### Steps to Reproduce

1. Pull latest bugfix-2.0.x edd6f6c5f293c510069d4f60587e6f9b3ecac939
2. Insert configs and enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL`
2. Slice a single layer rectangle with ~5mm brim enabled (Prusa Slicer)
3. Launch pronterface from console and log output to [pronterface-log.log](https://github.com/MarlinFirmware/Marlin/files/4372935/pronterface-log.log)
4. Print gcode file after entering the following GCODEs (I just added them to the beginning of my GCODE file):
; Debugging
M111 S247
; EEPROM defaults
M502
M500
M501
; Heat and probe bed
M190 R60
; Home
G28
; Probe
G29 P1
; UBL settings
G29 F10
G29 A
; Probe offset
M851 Z-0.93
; Save
M500

The Z probe offset is intentionally a little bit higher than it should be to compensate for when the nozzle gets too close to the bed. Otherwise the nozzle pushes into the bed. It is just low enough for the beginning of the print to stick.

**Expected behavior:** [What you expect to happen]

Prints the first layer at a relatively equal height across the whole layer. Any variations of height within close proximity should have gradual transitions.

**Actual behavior:** [What actually happens]

The brim (which is printed first) seems to print at equal height to the bed. As the print progresses the nozzle moves ever so slightly lower and lower into the bed. Later on in the print during the infill, the infill shows a sharp contrast of z height compared to the surrounding brim.

[image]

It also seems that z may even start to rise after bottoming out at a certain threshold, as the end of the infill gets slightly better towards the end. (The infill printed from left to right)

[image]

#### Additional Information

I have been banging my head against the wall these past few days trying to figure out whats wrong. I'm using a Einsy Rambo, which means TMC2130's, so I've tried disabling interpolation. Stealthchop is completely off. I have tried increasing motor current to 1000mah for Z. I've aldo tried re crimping motor cables.

I have repeated this same print several times with the same result. I am using a BLTouch for probing.

[image]

I hope that this is not a bug and just something I can fix myself. But after the past 72 hours of trying to fix this, I feel like I've exhausted all mechanical/electrical/configuration debugging options. I've noticed there are a lot of MBL issues that have come up recently, so perhaps this is worth looking into.

MarcelCountry, 2020-03-24:
I have the same Marlin 2.0.5.1 problem
[image]

AnHardt, 2020-03-24:
Is there any special reason you assume its UBLs fault and not Z-speed/acc/jerk?

When lifting the nozzle looses steps, the nozzle is to deep after returning to the same hight. If that adds up enough the nozzle is so deep, a downward motion skips steps - ending a bit higher.

MarcelCountry, 2020-03-24:
how could I solve? Thank you

Roxy-3D, 2020-03-24:
First...   I would put the nozzle .2mm from the bed.   And with UBL active, I would move the nozzle around to different places on the bed.    Is the nozzle faithfully tracking the contour of the bed?

If it is....  it is probably some other problem we are fighting like AnHardt is suggesting.

MarcelCountry, 2020-03-24:
I have tried all versions from bugfix 2.0 to 2.0.5.1 the result does not change

[image]

MarcelCountry, 2020-03-24:
Thanks for your clarifications I'm freaking out I hope that you will solve soon you cannot print in this way thanks a lot

GibsDev, 2020-03-24:
I just repeated the same test again after setting Z DEFAULT_MAX_ACCELERATION to 20 (1/5th of what it was) and lowering jerk from 5 to 3. It produced an Identical result:

[image]

I'm not sure how moving the head around the print bed manually would be different than printing a first layer? The benefit of doing it while printing is that I can visually see a history of how close it is to the bed.

[pronterface-log-low-accel.log](https://github.com/MarlinFirmware/Marlin/files/4376398/pronterface-log-low-accel.log)

I also added a G29 T1 to the start of the gcode for more info.

AnHardt, 2020-03-24:
> I'm not sure how moving the head around the print bed manually would be different than printing a first layer?

The differences are the warranted absence of 'z-hops' while just moving, no upward pressure from the extruded material, no other z-activity besides the leveling, no possible influence of LA, ... .

GibsDev, 2020-03-24:
Gotcha. I do have LA disabled, and I have verified there are no Z moves within the window where the z height is changing.

[image]

GibsDev, 2020-03-25:
I flashed 1.9.1.1 with the same result. I'm going to try some more hardware troubleshooting.

closed 2020-03-25 by GibsDev

GibsDev, 2020-03-25:
I thought I would post an update for anyone who was hoping for answers from this issue. I have finally got a usable first layer with UBL enabled.

As I've read many times UBL is a last step for tuning the bed of your printer, but I have seen little documentation on what is considered to be a reasonable range to expect UBL to work on. When I submitted this issue, I had a variance of ~1mm Z height across my bed. Reducing this range was the only thing that I changed that produced a result. I am now within 0.4mm, and will likely try to go even further.

My theory is that there is some lost z height in the calculations of UBL which is bottlenecked by compounding errors in 8-bit calculation. I still don't think it was skipped Z steps, because I tried debugging everything from my TMC2130's, and never got any errors about skipped steps. Although I don't know enough about TMCs and steppers to know for sure if I can say that for sure.

I would really like to try having a larger variance (1mm) with something like klipper, so all of the motion control logic could be calculated on a 32-bit processor to see if that would make a difference. But after the headache of troubleshooting this, I'm just glad I can finally print.

TL;DR; Make sure your bed is already as level as possible (<0.4mm variance)

Roxy-3D, 2020-03-28:
> My theory is that there is some lost z height in the calculations of UBL which is bottlenecked by compounding errors in 8-bit calculation.

No.   That is not the case.   The design was careful to make sure the calculations done do not have any accumulative error.    The Z-Correction for a given location is based on the bi-linear intersection of the (X,Y) lines within a given mesh cell.

It is kind of non-nonsensical...   But you can left one side of the bed 10 mm and build a mesh.   UBL will track the very tilted bed and print just fine.