MarlinFirmware/Marlin #17277: [Question] USE_REAL_EEPROM even when no SPI or I2C

GMagician, 2020-03-24:
Adafruit Grand Central has a 8MB QSPI on board and may be used instead of an external eeprom or the integrated Smart EEprom (eeprom emulation using SoC flash).

I'm going to integrate such EEPROM but there is a side effect on this. Caching eeprom page, needed when erasing is required, 4KB of ram.
What I've done now is, when `FLASH_EEPROM_EMULATION` is enabled, to fallback on QSPI if smart eeprom size is 0 (disabled). This solution may works but, as state above, with lot of resources (2% of RAM)..I don't like it.

This QSPI flash is not an emulated device, it's a real device so `USE_REAL_EEPROM` should be set, not `FLASH_EEPROM_EMULATION`.

Using `USE_REAL_EEPROM` will let me to use smart eeprom when `FLASH_EEPROM_EMULATION` and qspi when no `SPI_EEPROM` or `I2C_EEPROM` is selected. Please note that `I2C_EEPROM` is used on wild by ramps 1.4.4 and not sure if in future and spi eeprom will be used by another shiled (I think no one will use AGCM4 now that better all-in-one boards are sold, but who may know).
Maybe a new "define" may be added, ONBOARD_SPI_EEPROM?, to be set side by side with `I2C_EEPROM` and `SPI_EEPROM.`

May be this a viable solution? Any other hint?
Thanks

thinkyhead, 2020-03-24:
You're on the right track, I think. Too bad the 4K cache is required, but I guess the board has the RAM to spare…. I wouldn't worry on that front until lots of users are running out of RAM.

thinkyhead, 2020-03-24:
We can continue discussion over on Discord, if there are other concerns.

closed 2020-03-24 by thinkyhead

GMagician, 2020-03-25:
I don't feel confortable with Discord....too distracting, but ok