MarlinFirmware/Marlin #17278: [BUG] Feedrate = 0% does not work correctly in 2.0.5.0

Protos58, 2020-03-24:
Hello!
I have CNC with Marlin 2.0.5.0
In my work I need to lower the feedrate to 0.
In this case, the following occurs:

If I run a program with zero feed, the CNC does not stand still.
Instead, the program works with the highest possible pitch.
If the feed changes during operation, nothing happens until the program moves to the next line.
It is not possible to stop the feed by reducing it to zero.
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4375301/Configuration.zip)

AnHardt, 2020-03-24:
Sorry.
That can't work. In g-code a 'move' is completely determined before the 'move' is started. Changing the 'feedrate' after the 'move' has been started only affects following 'moves'.
A 'feedrate' of Zero is impossible. 
This would require infinit large timer registers.
The 'move' would last forever. The machine would hang because it can't reach its target.

If a fault at all, it's a conceptual problem of 'g-code', not Marlins.

ellensp, 2020-03-24:
Was thinking on this and I think they mean the feedrate multiplier (via lcd)

AnHardt, 2020-03-24:
Even the ' feedrate multiplier' can't have effect on the currently running move.
Even here a 0% is impossible for the above reasons.

AnHardt, 2020-03-24:
No motion slower than [this](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/Configuration_adv.h#L768-L771) is possible.

Protos58, 2020-03-24:
I do not agree.
Look at the picture of an "adult" cnc, for example in google. Feed regulators have a control range of 0-150% (maybe wider).
This is a common tool for cnc operator.

Protos58, 2020-03-24:
MINIMUM_PLANNER_SPEED = 0. No problem.

GMagician, 2020-03-24:
> If a fault at all, it's a conceptual problem of 'g-code', not Marlins.

That's not a faultor a conceptual problem, in real industrial machines it's used, more than you think

AnHardt, 2020-03-24:
A possible workaround is to split one move into many sub-moves (internally or externally) and to stop sending sub-moves if the feedrate is Zero.

For  'servo driven' machines (without step pulses) there is no problem.

>MINIMUM_PLANNER_SPEED = 0. No problem.

The AVRs do catch that here:
https://github.com/MarlinFirmware/Marlin/blob/8dbbcfd382356eb5d16b8bfa75c713bbc1de1631/Marlin/src/module/stepper.h#L524-L531
Either 32bitters could have a division by 0 problem here, or the AVRs could omit the check.

thinkyhead, 2020-03-24:
Best we can do for you is to have the machine pause and stop accepting moves when it sees a feedrate of 0.

Protos58, 2020-03-25:
> timer = uint32_t(STEPPER_TIMER_RATE) / step_rate;

I talked about this in the topic "...the program works with the highest possible pitch."

Protos58, 2020-03-25:
Here is a similar topic. https://github.com/MarlinFirmware/Marlin/issues/14545
Maybe join forces?
By the way, when the "EMERGENCY_PARSER" is turned on, the "M0" command does not work immediately, but waits for the completion of the buffer position.

GMagician, 2020-03-25:
> ... "M0" command does not work immediately, but waits for the completion of the buffer position.

AFAIK this may be correct, even on industrial machines M0 flush all pending movements before stopping

Protos58, 2020-03-25:
> AFAIK this may be correct

Maybe you know how to make Marlin respond to commands immediately? At least stop immediately.

Protos58, 2020-03-25:
Stop, but then continue the execution of the program on command.

GMagician, 2020-03-25:
> Maybe you know how to make Marlin respond to commands immediately? At least stop immediately.

My was just a comment since I work daily with CNC systems and I know how they behave. To realize what you ask Marlin should ramps down movements (interpolating) and then recalculate all elements on queue before restarting. I think it may be do latter part without problems because I think this is already present into code, but former...

GMagician, 2020-03-25:
> Maybe you know how to make Marlin respond to commands immediately? At least stop immediately.

That's not what M0 should do, in my optinion.

Protos58, 2020-03-25:
Let's say. 
Maybe there is an internal move through planner.cpp or motion.cpp?

GMagician, 2020-03-25:
AFAIK once movement is inserted in to queue al ramps up/down computations are done. Computation purpose is to let execute last movement till a stopped state. When next movement is inserted, all computation are redone (moving back to queue) to let this last to reach 0 speed at end. When a move is executing its speed can't be changed anymore. Doing what you are asking means finding movement in execution (or next one) and change its ramp speed to let it reach 0 speed. Then recompute all other things to start from 0 speed. Not sure enough power on small Soc to to that, but there are here people that know Marlin better than me (I just see the surface of it)

Protos58, 2020-03-25:
Thanks for the theory. Something explained. I think about it.

Protos58, 2020-03-25:
> Best we can do for you is to have the machine pause...

If possible, do it. As a first step, that would be good for me.

Protos58, 2020-03-25:
> Best we can do for you...

Thank you in advance for your help

boelle, 2020-06-20:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-20 by boelle