MarlinFirmware/Marlin #17282: [FEATURE] (No Configurations Available for Secondary Serial Port on MKS_SGEN_L Board)

borland1, 2020-03-24:
### Bug Description
With USB port as primary serial port ( Serial_Port ) , unable to configure a secondary serial port ( Serial_Port2 ) for board MKS_SGEN_L, in configuration.h file.  File pins_MKS_SGEN_L.h only shows  AUX-1 pins which are commented out. 

### My Configurations
[ConfigurationFiles.zip](https://github.com/MarlinFirmware/Marlin/files/4377621/ConfigurationFiles.zip)

#### Additional Information
1. From "Configuring Marlin" documentation, it is not clear how to configure LPC1768's dedicated USB port as primary, or AUX-1 as secondary serial ports.  Documentation refers 'index', which I believe is referring to table in Arduino web site under Reference section, 'Serial',  see link... [https://www.arduino.cc/reference/en/language/functions/communication/serial/](url) 
This table clearly shows pin assignment for various Arduino development boards.

2. For Board:  MKS SGEN L (LPC1768), in addition to the USB port, the four UARTs are available as follows: 				

LPC1768 UART # | Board   Connector/Silkscreen Label (Processor Pin Assign)
-- | --
  |   |   |   |   |  
USB (Bootloader) | USB Type B (Printer) |   |   |  
D+ (P0.29), D- (P0.30) | D+ (P0.29), D- (P0.30) |   |   |  
  |   |   |   |   |  
UART0 | AUX-1 |   |   |   |  
TXD0 (P0.2), RXD0 (P0.3) | TX (P0.2), RX (P0.3) |   |   |  
  |   |   |   |   |  
UART1 | EXP-1 |   |   |   |  
TXD1 (P0.15), RXD1 (P0.16) | LCD D4 (P0.15), LCD RS (P0.16) |   |  
  |   |   |   |   |  
UART2 | E0/E1 Stepper Drivers |   |   |  
TXD2 (P0.10), RXD2 (P0.11) | E1 EN(P0.10), E0 DIR(P0.11) |   |  
  |   |   |   |   |  
UART3 | E1 Stepper Driver |   |   |  
TXD3 (P0.0), RXD3 (P0.1) | E1 DIR (P0.0), E1 STEP (P0.1) |   |  

3. From code comments in Configuration.h file, I assume USB can be configured as primary with line...
#define SERIAL_PORT 0     // USB

and secondary on AUX-1 with line ...
#define SERIAL_PORT2  1  // AUX-1

but it is not clear which UART (USB, UART0, UART1, UART2, UART3) is actually assigned if any.

ellensp, 2020-03-25:
USB is always -1 on controllers that the USB lines are directly connected to the Processor.
Hardware Serial ports are not on pins files.they are in the framework files. 
See https://github.com/p3p/pio-framework-arduino-lpc176x/blob/master/README.md

borland1, 2020-03-25:
Thanks for the clarification,  So, for USB as primary serial port, and AUX-1 as secondary serial port, I just need to configure the defines like this:

#define Serial_Port -1  // USB
#define Serial_Port_2 0  // AUX-1

?
Or if using UART3 as secondary serial port:

#define Serial_Port -1  // USB
#define Serial_Port_2 0  // UART3 - TX(P0.0), RX(P0.1) 

Is that correct?

ellensp, 2020-03-25:
Yes that looks correct

ellensp, 2020-03-25:
Please close if this has solved your issue.

closed 2020-03-25 by borland1