MarlinFirmware/Marlin #17286: [BUG] SKR 1.4 ERR: MAXTEMP: BED

Atlas-39, 2020-03-25:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4378999/Configuration.zip)

### Bug Description

Right now it says:
**ERR: MAXTEMP: BED
PRINTER HALTED
Please Reset**
Every time I boot it up.
Sometimes it says **MAXTEMP: E1** depending on how I fiddle with the firmware (I tried multiple things to fix this issue, all of them were stupid solutions and I'm to embarassed to post them here so I will not).
Curious thing though, I'm using E0 Not E1 (Maybe something to look into later).
I'm relatively new to the firmware side of 3D printing so please excuse me if I leave anything out or say anything stupid

### My Configurations

[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/4379001/Configuration_adv.zip)
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4379005/Configuration.zip)

### Steps to Reproduce

1. Follow Teaching Tech's "[SKR V1.3 Comprehensive guide](https://www.youtube.com/watch?v=GNGN2iSQ5j4)"
2. Instead of LPC1768 (SKR1.3), use LPC1769 for the environment (SKR 1.4 Tubo)
3. Flash the firmware on to your SKR 1.4 Turbo
4. Cry because you've spent 12+ hours trying to fix this issue

**Expected behavior:** [Printer runs normally]

**Actual behavior:** [The display beeps intermittently and displays "ERR: MAXTEMP: BED, PRINTER HALTED, Please Reset]

rado79, 2020-03-25:
> "ERR: MAXTEMP: BED, PRINTER HALTED, Please Reset"

Same error appears, if bed thermistor has short circuit.
If you have not tried, unplug bed thermistor and reset again.

Atlas-39, 2020-03-25:
> > "ERR: MAXTEMP: BED, PRINTER HALTED, Please Reset"
> 
> Same error appears, if bed thermistor has short circuit.
> If you have not tried, unplug bed thermistor and reset again.

The problem is that I've unplugged everything and I'm running the board's power via my PC. Nothing's plugged in apart from the LCD.

thisiskeithb, 2020-03-25:
> The problem is that I've unplugged everything and I'm running the board's power via my PC. Nothing's plugged in apart from the LCD.

With no thermistors plugged in & the board immediately hitting max temp, something is likely damaged on the board.

Atlas-39, 2020-03-25:
> > The problem is that I've unplugged everything and I'm running the board's power via my PC. Nothing's plugged in apart from the LCD.
> 
> With no thermistors plugged in & the board immediately hitting max temp, something is likely damaged on the board.

That's weird, everything was working perfectly before I swapped the LCD connectors

nogarth, 2020-03-30:
I fet same error when i heat my bed up for a print falls over from bed. But if i preheat bed and leave it for a bit it prints ok. Few more trsts tomorrow as i have print running now so gonna test from a cold start tomorrow

RFBomb, 2020-04-21:
Did you ever resolve this? I am having similar issue.... https://github.com/MarlinFirmware/Marlin/issues/17633

closed 2020-04-27 by thinkyhead

mdellen, 2020-05-26:
I also get the max temp E1 (SKR 1.4 turbo) after upgrading to the latest bug fix release. Will try to recompile to see if it is persistent.

mdellen, 2020-05-29:
When compiling the exact same source code today I get the E1 max temp error. When I load the .bin from a few months ago (same source) everything is fine. So it seems something has changed in the libraries and not Marlin itself.

I need to change something in the config but am not able to compile new firmware without getting the error. Tried BTT 2.0.1 branch, 2.0.1, 2.0.4.4 and 2.0.5.2