MarlinFirmware/Marlin #17288: NEO Type RGB not match with function ... RGBW

fred2088, 2020-03-25:
defined neo type not matched with function : 

Marlin-bugfix-2.0.x

  #define NEOPIXEL_TYPE   NEO_RGB

from leds.cpp add this statement 
const uint32_t neocolor = neo.Color(255, 0, 0);

got compile error
sketch\src\feature\leds\neopixel.h:115:26: note: candidate: static uint32_t Marlin_NeoPixel::Color(uint8_t, uint8_t, uint8_t, uint8_t)
   static inline uint32_t Color(uint8_t r, uint8_t g, uint8_t b, uint8_t w) {
                          ^~~~~
sketch\src\feature\leds\neopixel.h:115:26: note:   candidate expects 4 arguments, 3 provided
from 2.0.x beta is using 
pixel.color(255,0,0) for NEO_RGB type

need to changed to net.Color(255,0,0,0) to get compiled...

Thanks

ellensp, 2020-03-25:
Cannot replicate in current marlin bubfix
also in my leds,cpp is
`const uint32_t neocolor = LEDColorWhite() == incol
                            ? neo.Color(NEO_WHITE)
                            : neo.Color(incol.r, incol.g, incol.b, incol.w);`

fred2088, 2020-03-26:
const uint32_t neocolor = LEDColorRed() == incol ? neo.Color(NEO_RED) : neo.Color(incol.r, incol.g, incol.b, incol.w);

that mean need the above cmd to set in RED ?
how about set with custom color code ?

Thanks

closed 2020-04-27 by thinkyhead