MarlinFirmware/Marlin #17291: [BUG] TMC Hybrid Threshold settings go wonky

pmjdebruijn, 2020-03-25:
### Bug Description

The TMC Hybrid Treshhold settings aren't properly reproduced between power cycles.

### My Configurations

[marlin-configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4382402/marlin-configuration.zip)

### Steps to Reproduce

1. Flash firmware (reproducable with 2.0.5.2)
2. Reset to defaults
3. Check TMC Hybrid Threshold settings: X:160 Y:160 Z:20 E1:20
4. Save settings
4. Power machine off
5. Power machine on again
6. Check TMC Hybrid Threshold settings: X:162 Y:162 Z:20 E1:4

**Expected behavior:** [What you expect to happen]

I would expect the saved settings to remain completely unchanged between power cycles.

**Actual behavior:** [What actually happens]

The X and Y threshold seem to be affected by what would almost seems like a rounding errors or something.

And the E0/E1 threshold is forgotten entirely. Presumably this is because E0 and E1 are being confused?

#### Additional Information

I have the issue on a SKR Mini E3

pmjdebruijn, 2020-03-27:
So that mystery is solved:
https://github.com/MarlinFirmware/Marlin/blob/7124b2164d5c863ed1013df57a424739420003fc/Marlin/src/core/language.h#L351

Still the value wobble is still there...

qwewer0, 2020-03-30:
SKR Mini E3 v1.2
Flashed the latest bugfix and the Hybrid Threshold values changed, from 100 / 100 / 3 / 30 to 100 / 48 / 6

boelle, 2020-06-20:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

unilogicbv, 2020-06-24:
This issue is still reproducable with the latest stable 2.0.5.3

pmjdebruijn, 2020-06-24:
This issue is still reproducable on the latest stable firmware 2.0.5.3

pmjdebruijn, 2020-06-27:
I think I realized why this is happening...

If I understand this correctly, Marlin doesn't actually save my real setting to EEPROM, it saves the derived value that been sent to the Trinamic chip, which is read out again.

That derived value, depends on the respective steps/mm setting:
https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/src/feature/tmc_util.h#L44
`return 12650000UL * msteps / (256 * thrs * spmm);`

And I'm seeing this big difference on my E0 threshold, because I'm actually changing my E0 steps/mm on initial configuration, as my Ender 3 has a BMG extruder, where my E steps go from 93 to 400 (both which are slightly underextruding).

And this holds up with the actual numbers I'm seeing 20 * 93 / 400 = 4 (integer rounded)

As for the 160 -> 162 presumably that is just some integer rounding that happening after the math from and to the chip, which seems to hold up in my numbers as well:
12650000 * 32 / 256 / 160 / 160 = 61 (integer rounded)
12650000 * 32 / 256 / 160 / 61 = 162 (integer rounded)

The good news, this issue only occurs in this defined condition, which usually only happens for people who are modifying machines.

The bad news, there will be no trivial one line bugfix, this is basically a debatable design choice, where it probably wasn't per-se the best idea, to store derived values, as opposed to a reference ground truth in EEPROM.

Though Marlin's UI ideally should represent the step granularity:
with 160 steps/mm: 152 154 156 159 162 164 167 170 173 176 179
with 80 steps/mm: 76 77 78 79 81 82 83 85 86 88 89

Please re-open this bug, as it's root cause has been clearly identified.

pmjdebruijn, 2020-06-28:
https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/src/lcd/menu/menu_advanced.cpp#L476

A mitigation would be possible by keeping the old steps/mm in a temp var after changing it, and reading out the hybrid value, multiplying it by the old steps/mm value and dividing it by the new steps/mm value and then saving it back. But pretty it would be not...

thinkyhead, 2020-07-17:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

pmjdebruijn, 2020-07-19:
This is still the case with current bugfix-2.0.x

thisiskeithb, 2020-07-19:
I do believe this is intentional: https://github.com/MarlinFirmware/Marlin/issues/17630#issuecomment-619519254

qwewer0, 2020-07-19:
> I do believe this is intentional: [#17630 (comment)](https://github.com/MarlinFirmware/Marlin/issues/17630#issuecomment-619519254)

@thisiskeithb That doesn't explains why did my HT values go from 100 / 100 / 3 / 30 to 100 / 111 / 48 / 6.

thisiskeithb, 2020-07-19:
> That doesn't explains why did my HT values go from 100 / 100 / 3 / 30 to 100 / 111 / 48 / 6.

It may be another issue outside of how values are stored/retrieved for TMC drivers (@teemuatlut can correct me if I'm wrong).

Make sure you send `M502`, followed by `M500` after updating your firmware to pull the values you have set in your config/EEPROM.

qwewer0, 2020-07-19:
@thisiskeithb It didn't happen right after a firmware flashing, and I always do `M502`+`M500` after I flash a new firmware.

@pmjdebruijn Do you have a solid way to reproduce the issue? It happens really rarely on my board.

closed 2020-08-25 by github-actions[bot]

pmjdebruijn, 2020-08-25:
@qwewer0 It's only noticable when you the E-Step are changed significantly. Updating firmware itself isn't relevant.

To reproduce:

- Reset to to defaults. So I have E93 and TMC-E-Thres of 20.
- Change E steps to 400.
- Save settings
- Turn machine off
- Turn machine on
- Check E-step, which is still as expected 400, then check TMC-E-Thres of 4.

I've already added a pretty much full analysis:
https://github.com/MarlinFirmware/Marlin/issues/17291#issuecomment-650596464

Please re-open. Also, can a tag be added so it's not auto-closed, given this is a confirmed bug with analysis included.