MarlinFirmware/Marlin #17294: [BUG] Auto bed levelling does nothing

VanKurt, 2020-03-25:
### Bug Description

I select "Auto bed level" from the LCD. The printer then homes all axes.
After that it stops and nothing more happens.

### My Configurations

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4382855/Marlin.zip)

### Steps to Reproduce

1. Select "Auto bed level" from the LCD

**Expectedbehavior:** I would have expected the printer to probe several points o the bed.

**Actual behavior:** The printer just stops/aborts the levelling after homing the axes.

shitcreek, 2020-03-25:
Did your Bltouch work before and this is a new issue?

VanKurt, 2020-03-26:
Before I was using Klipper. Just switched to Marlin a few days ago.
To be clear: the BLTouch is working properly. It's just the auto levelling in Marlin that gets stuck after homing...

mpieroway, 2020-03-26:
I am seeing the same problem with my inductive probe. On G29 or G29 P1 command it probes the same location that the Z homed to and does not proceed to the next point. It is not building the mesh in order to be able to calibrate further. It does however probe the specified mesh if using the G29 J command.
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/4389620/Configuration.txt)
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/4389621/Configuration_adv.txt)

grauerfuchs, 2020-03-26:
Confirmed; I'm having the same issue with an inductive probe using auto bilinear leveling. Upon initiation from LCD, it homes, moves to the first leveling point, ignores the triggered probe and instead moves to the full Z depth allowed in settings, and then returns a leveling failure message to the serial port. Manual initiation via G29 results in the same.

grauerfuchs, 2020-03-26:
I'm starting to suspect a library issue. I just did a roll-back through each named/tagged revision all the way to 2.0.1 and the machine is exhibiting the same behavior on every single one. I know I had a working probe with this configuration within the last couple months.

mpieroway, 2020-03-27:
> ### Bug Description
> I select "Auto bed level" from the LCD. The printer then homes all axes.
> After that it stops and nothing more happens.
>
> ### My Configurations [...]

Have you tried using the bugfix? I also found with my LCD I had to change the firmware coding to send the proper G code to the printer.

tphan26363, 2020-03-27:
Actually, I have been able to adjust the BL touch sensor into a certain range and it start working.  if it is detected that the probe is out (-1mm. +1mm) range of the zero position, it failed to probe. How ever it can only run once( I have option BI-LINEAR).  Try the 2nd time and the probe is totally disable.  I have to recycle power so I can start the G29 again

VanKurt, 2020-03-28:
Here's a little update:
I disabled the following defines in my config_adv.h:

#if PROBE_SELECTED && !IS_KINEMATIC
  //#define MIN_PROBE_EDGE_LEFT 60
  //#define MIN_PROBE_EDGE_RIGHT 235
  //#define MIN_PROBE_EDGE_FRONT 35
  //#define MIN_PROBE_EDGE_BACK 225
#endif

Now the bed levelling actually works! But sadly it now probes very close to the edges of my bed where the holding clamps sit. So the probing results are pretty much useless.

Is there any other way to limit the area in which the probing happens?

nik012003, 2020-04-02:
I'm having the same exact problem. Solved by doing:
#if PROBE_SELECTED && !IS_KINEMATIC
//#define MIN_PROBE_EDGE_LEFT 60
//#define MIN_PROBE_EDGE_RIGHT 235
//#define MIN_PROBE_EDGE_FRONT 35
//#define MIN_PROBE_EDGE_BACK 225
#endif

sjasonsmith, 2020-04-02:
In both of the recent posts the commented out probe edges, if enabled, would have created overlapping edges with a probable region negative in size.

These setting are not needed by most people. Just set MIN_PROBE_EDGE and be done. If you need these, each value should be a distance from the edge of the bed, not from the origin point.

There should probably be sanity checks to verify the sum of two edges is not more than the total size of the bed (or perhaps half the bed). When doing so we’ll have to remember there is an edge case for NOZZLE_AS_PROBE where these values are allowed to be negative.

nik012003, 2020-04-02:
Thank you, now it's working perfectly :)

grauerfuchs, 2020-04-02:
Seems there might be more than one issue discussed here, as the issue I'm experiencing appears more related to the use of the z-min endstop as the probe pin (still not working for me) vs. the min probe edges, which appears to be solved. Glad to hear the probe edges are solved. I'll move off, as these issues have diverged.

thinkyhead, 2020-04-02:
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information so that we're not just taking stabs in the dark. Here is the boilerplate:

- Download Marlin `bugfix-2.0.x` to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

sjasonsmith, 2020-04-02:
I believe my response addresses the problem from the original reporter. And @nik012003.  Adding a sanity check is probably all that is needed.

If @grauerfuchs has different issues they should probably not be tacked onto this issue.

grauerfuchs, 2020-04-02:
Agreed. Would you prefer me to delete my posts from this issue so it won't confuse others?

boelle, 2020-06-20:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-20 by boelle