MarlinFirmware/Marlin #17296: DELTA PRINTER NOT DOING UBL CORRECTLY MARLIN 2.0.x

3ddev-j, 2020-03-25:
Hi everyone, recently uploaded marlin fw to my BTT SKRv1.3 and i've calibrated everything but ubl is not working properly.

It does not move to probe points, and only performs the hole calibration on a single point when i click at lcd " build cold mesh".

Also attached a short video that shows the problem.

Could you please help me?

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4383477/Configuration.zip)

boelle, 2020-06-20:
@3ddev-j is issue gone?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle