MarlinFirmware/Marlin #17297: [BUG] BTT GTR V1.0 I2C EEPROM can not wtite MARLIN 2.0.5.2

DJDOUBLEG, 2020-03-25:
[config.zip](https://github.com/MarlinFirmware/Marlin/files/4383553/config.zip)
[image]
[image]
[image]

Onboard 24c64.

I tried this setting too:
[image]
Same issue.

yavinmaster, 2020-03-26:
Have you tried doing a M502 then M500?  Different versions sometimes change the stored data so you need to do this to correctly format the EEPROM.

DJDOUBLEG, 2020-03-28:
> Have you tried doing a M502 then M500? Different versions sometimes change the stored data so you need to do this to correctly format the EEPROM.

Yes i tried it.
M500 instant reboot.

NAPCAL, 2020-04-03:
Confirmed reboot with 2.0.5.3 as well. Note per the schematics, the 24C64 is on PH4/PH5, and the expansion I2C (J65) is on PH7/PH8. Unclear, if the code is using these pins for an I2C EEPROM. 

I tried an external EEPROM on J65 but no change. Every time a write command happens (EEPROM INT, or Save settings), the code stalls and then causes a board reboot.

NAPCAL, 2020-04-03:
Sending a M500
SENT: M500
READ: echo:Error writing to EEPROM!
Error writing to EEPROM!
READ: echo:Error writing to EEPROM!
Error writing to EEPROM!
READ: echo:Error writing to EEPROM!
Error writing to EEPROM!
READ: echo:Error writing to EEPROM!
Error writing to EEPROM!
READ: echo:Error writing to EEPROM!
Error writing to EEPROM!
READ: echo:Error writing to EEPROM!
Error writing to EEPROM!
READ: echo:Error writing to EEPROM!
Error writing to EEPROM!
READ: echo:Error writing to EEPROM!
Error writing to EEPROM!
READ: echo:Error writing to EEPROM!
Error writing to EEPROM!
READ: echo:Error writing to EEPROM!
Error writing to EEPROM!
READ: echo:Error writing to EEPROM!
Error writing to EEPROM!
READ: echo:Error writing to EEPROM!
Error writing to EEPROM!
READ: echo:Error writing to EEPROM!
Error writing to EEPROM!
READ: echo:Error writing to EEPROM!
Error writing to EEPROM!
READ: echo:Error writing to EEPROM!
Error writing to EEPROM!
READ: echo:Error writing to EEPROM!
Error writing to EEPROM!
READ: echo:Error writing to EEPROM!
Error writing to EEPROM!
READ: echo:Error writing to EEPROM!
Error writing to EEPROM!
READ: echo:Error writing to EEPROM!
Error writing to EEPROM!
READ: echo:Error writing to EEPROM!
Error writing to EEPROM!
READ: echo:Error writing to EEPROM!
Error writing to EEPROM!
READ: echo:Error writing to EEPROM!
Error writing to EEPROM!
READ: echo:Error writing to EEPROM!
Error writing to EEPROM!
READ: echo:Error writing to EEPROM!
Error writing to EEPROM!
READ: echo:Error writing to EEPROM!
Error writing to EEPROM!
READ: echo:Error writing to EEPROM!
Error writing to EEPROM!
READ: echo:Error writing to EEPROM!
Error writing to EEPROM!
READ: echo:Error writing to EEPROM!
Error writing to EEPROM!
WARNING: Device unplugged while connected to port Disconnected.

Sending M501
SENT: M501
READ: echo:EEPROM version mismatch (EEPROM=? Marlin=V76)
EEPROM version mismatch (EEPROM=? Marlin=V76)
READ: echo:Hardcoded Default Settings Loaded
Hardcoded Default Settings Loaded

Sending M502
SENT: M502
READ: echo:Hardcoded Default Settings Loaded
Hardcoded Default Settings Loaded

When sending an M500 only errors until it reboots,

NAPCAL, 2020-04-04:
Found the correction, the assigned pins for the on-board I2C EEPROM are not correct. When checking the Demo Marlin provided by BTT the \buildroot\share\PlatformIO\variants\BIGTREE_SKR_PRO_1v1\variant.h​ file has the correct pin assignments
// I2C Definitions
#define PIN_WIRE_SDA            PH5
#define PIN_WIRE_SCL            PH4

and the others do not. 
 // I2C Definitions
#define PIN_WIRE_SDA            PB7
#define PIN_WIRE_SCL            PB6

As a test on my Marlin 2.0.5.3, I made these changes in the \buildroot\share\PlatformIO\variants\BIGTREE_SKR_PRO_1v1\variant.h
// I2C Definitions
#if STM32F4X_PIN_NUM >= 176
	#define PIN_WIRE_SDA            PH5
	#define PIN_WIRE_SCL            PH4
#else
	#define PIN_WIRE_SDA            PB7
	#define PIN_WIRE_SCL            PB6
#endif

Works with M500-M504 commands, Initialize EEPROM, Store Settings, Load Settings, Restore Defaults from the display.

DJDOUBLEG, 2020-04-04:
Thank you! I try this.

DJDOUBLEG, 2020-04-04:
It's Confirmed.
This settings is working.
Thx.

closed 2020-04-04 by DJDOUBLEG

sjasonsmith, 2020-04-04:
I suggest leaving this open until the code is updated. This will be important to fix for other GTR users also.

thinkyhead, 2020-04-27:
Found it anyway…

commit 5d1498f2 (2020-04-27): Fix SKR Pro / GTR i2c pins

Fixes #17297