MarlinFirmware/Marlin #17301: E0_AUTO_FAN_PIN cannot be FAN_PIN

PlasmaHH, 2020-03-26:
This is more a kind of a question rather than a bugreport, but maybe its a bugreport, depending on the outcome here. I am using the bugfix-2.0.x branch at the moment.

I have a BTT SKR Pro Board and wired the extruder fan to FAN0 (parts fan to 1, chassis fan to 2) which is pin `PC8` and then in Configuration_adv tried to set it to `#define E0_AUTO_FAN_PIN PC8`.

However the Build fails with SanityCheck.h:1484 saying that I can not set it, the comment says
>Make sure auto fan pins don't conflict with the fan pin
but it doesn't say why. 

In theory I could switch the parts and extruder fans pins, but why is this not allowed? A quick skim through some code doesn't reveal any reason to me. If it is for whatever reason required for some other boards, maybe this is then a feature/bug request to allow it for boards where it shouldn't be a problem.

(And yes, I am aware of that I have to then set the fan index in cura etc. for the parts fan, I don't mind)

Grogyan, 2020-03-30:
I am having the same issue with using the release version of Marlin 2.0.5.2 on the SKR1.3
This was working on Marlin 2.0.1

jonathanlundstrom, 2020-03-30:
Same issue here :(

randellhodges, 2020-03-30:
pins_BTT_SKR_PRO_V1_1.h defines `#define FAN_PIN ` as `PC8` and `FAN_PIN` is what is used for the part cooling fan (I believe it is assumed that FAN_PIN is always used for part cooling). When you `#define E0_AUTO_FAN_PIN PC8` inside the Configuration_adv.h, you now have 2 fans using the same pin.

I would use the defines from the pins file. Either move them physically around, or try this:

Configuration_adv.h
```
#define E0_AUTO_FAN_PIN FAN1_PIN
#define CONTROLLER_FAN_PIN FAN2_PIN
```
I assume when you say chassis fan, you mean the controller fan?

then maybe at the bottom (or top) of the file do
```
#define FAN_PIN                             FE5   // was PC8
#define FAN1_PIN                            PC8   // was PE5
```

Not sure if you need to undefine before the define for compiler warnings.

Just thinking out loud. No idea if it'll work as expected. Or you could change them right in the pins file, but I don't recommend that.

PlasmaHH, 2020-03-30:
> pins_BTT_SKR_PRO_V1_1.h defines `#define FAN_PIN ` as `PC8` and `FAN_PIN` is what is used for the part cooling fan (I believe it is assumed that FAN_PIN is always used for part cooling). When you `#define E0_AUTO_FAN_PIN PC8` inside the Configuration_adv.h, you now have 2 fans using the same pin.

This is just one way. The board has just three FAN terminals and you can wire things however you want, and it so happened that I wired FAN0 as the extruder cooling fan, FAN1 as the part cooling fan and FAN2 as the chassis fan. 

Sure, I can rewire them, but the question is why does marlin force me to do it?

> 
> I assume when you say chassis fan, you mean the controller fan?
Probably, the thing that cools the electronics

> #define FAN_PIN                             PE5   // was PC8
> #define FAN1_PIN                            PC8   // was PE5

Yes, you could swap them all around in the pin definitions, but that would be somewhat confusing with the silkscreen on the actual board.

I would just personally prefer to be as free as possible with my pin assignments, its not a showstopper. Its just easier to do it in the configuration header than to have to swap things around physically again...

randellhodges, 2020-03-30:
FAN_PIN is the part cooling fan for the first extruder. I am not aware of any way to tell Marlin to use something else.

If you want to be free with pin assignments, then you'd need to submit a pull request to change the logic.

To get the level of freedom you'are looking for, you'd need to create a define for the part cooling fan, like `E0_AUTO_FAN_PIN` maybe `E0_COOLING_FAN_PIN` then change everywhere to use these defines. Then a person can map as they like. To help out legacy users, provide a fallback to `E0_COOLING_FAN_PIN` = `FAN_PIN` if it isn't defined.

Take a look at [temperature.cpp](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/module/temperature.cpp) for more information on the places `FAN_PIN` and `FANX_PIN` are used. (That is why I suggested changing their pin assignments if changing the hardware wasn't an option).

I would agree that having an explicit define for each would be cleaner, but it might be a bit of work to get it in place.

PlasmaHH, 2020-03-30:
> FAN_PIN is the part cooling fan for the first extruder. I am not aware of any way to tell Marlin to use something else.

There is the point I am wondering, how is marlin using this knowledge in any way? In the below mentioned .cpp it doesn't seem to have a special role, other than being a bit inconsistent with a name FAN0_PIN probably being more appropriate. Yes, M106 without any option uses index 0 as default, and slicers use that as a default, but they seem to all have the ability to give an index in the machine settings. 

Let me ask this in another way: are there any places in marlin where FAN_PIN has a special role and is enabled as a side effect of anything else than a gcode command that (optionally) takes a fan index parameter (and just defaults to 0) ?

 
> If you want to be free with pin assignments, then you'd need to submit a pull request to change the logic.
>
> To get the level of freedom you'are looking for, you'd need to create a define for the [...]

I can't find a place where a fan has the explicit meaning of "parts cooling fan" only "fan to be enabled by a gcode command like M106 which defaults to that pin when no explicit one is given".

Another way to view it: If I now wire it to FAN1_PIN then "M106 P1" would collide with the auto fan feature, but there seems to be no warning for it? Or will the fan not be available for the M106 command when it is set as the auto fan? I can't find any place in the code where this is the place, but thats why I am asking, I don't  know it like the back of my hand, people here might.

> 
> Take a look at [temperature.cpp](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/module/temperature.cpp) for more information on the places `FAN_PIN` and `FANX_PIN` are used. (That is why I suggested changing their pin assignments if changing the hardware wasn't an option).

For me changing the hardware is an option, but as I said initially I am just wondering why. When I will have time (and with covid there sure will be some) I will disable the sanity check just to see if something breaks. I am already using FAN1 as the parts cooling fan just fine.

> I would agree that having an explicit define for each would be cleaner, but it might be a bit of work to get it in place.
Yeah, there are surely a lot of things that are a higher priority, like the revamp of the SPI stuff to get SD card to work properly on that board...

randellhodges, 2020-03-30:
I think it is implicitly coupled by `thermalManager.fan_speed[active_extruder]`. I didn't dig into the thermalManager class.

I'm guessing, originally, the Marlin had support for E0, E1, E2 and then the idea was FAN(1)_PIN, FAN2_PIN, FAN3_PIN was defined for part cooling and hotend cooling fans were just assumed to be always on.

Later, auto_fan for hotend cooling was added and that gave the ability to map to whatever fan you wanted, generally under the assumption that you'd use one of the other fan pins (since it has a mosfet).

Then things like chamber fan and controller fan was added. They gave the ability to map but there was nothing added to give that mapping ability for the part cooling fans themselves.

This is all guesses on my part.

PlasmaHH, 2020-03-30:
> I think it is implicitly coupled by `thermalManager.fan_speed[active_extruder]`. I didn't dig into the thermalManager class.

Its late and I will soon go to sleep, which is maybe why I don't quite understand the code. But so far it looks like it assumes that `thermalManager.fan_speed[active_extruder]` is the speed of the extruder cooling fan, so for extruder 0 it would map to FAN[0]_PIN in the end, just as I have connected it. The code seems to be for an additional PID term that tries to compensate power loss for extruder cooling fan as accurately as possible by taking its speed into account. So not about the parts cooling fan there. I will later dig deeper into it as it kinda feels like that PID functionality could be buggy then...

> This is all guesses on my part.
They sound sensible.

minosg, 2020-04-14:
Question.

In the [temperature.cpp](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/module/temperature.cpp) linked above there is the following hash define

`  #define INIT_E_AUTO_FAN_PIN(P) do{ if (P == FAN1_PIN || P == FAN2_PIN) { SET_PWM(P); SET_FAST_PWM_FREQ(FAST_PWM_FAN_FREQUENCY); } else SET_OUTPUT(P); }while(0)`

While in the [Conditionals_post.h ](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/inc/Conditionals_post.h)the following code exists

`#define _HAS_FAN(P) (PIN_EXISTS(FAN##P) && CONTROLLER_FAN_PIN != FAN##P##_PIN && E0_AUTO_FAN_PIN != FAN##P##_PIN && E1_AUTO_FAN_PIN != FAN##P##_PIN && E2_AUTO_FAN_PIN != FAN##P##_PIN && E3_AUTO_FAN_PIN != FAN##P##_PIN && E4_AUTO_FAN_PIN != FAN##P##_PIN && E5_AUTO_FAN_PIN != FAN##P##_PIN && E6_AUTO_FAN_PIN != FAN##P##_PIN && E7_AUTO_FAN_PIN != FAN##P##_PIN)`

Which is a bit contradicting the logic. 

Are AUTO_FAN Pins meant to be treated as fan pins or not?

The code in those files indicates that:
1. The AUTO_FAN is not initialized in a similar manner as a fan and will ignore a SOFT_PWM configuration entry
2. AUTO_FAN is not treated as a fan and does not increment FAN_COUNT ( probably not addressable by gcode ?)
3. Is it assumed that the pin provided by the user contains a hardware timer and is running  ( I cannot find any initialization logic for the AUTO_FAN pins)

I am wondering as well what would the logic be of the Sanity check. It could all make sense somehow that I fail to see..

boelle, 2020-06-20:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-20 by boelle

kishkaru, 2020-06-30:
BTT SKR PRO v1.1 / v1.2 boards have 3 fully programmable fan headers:
---
[image]
---
[image]

See official [pinout diagram](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/blob/master/SKR-PRO-V1.2/manual/SKR-PRO-V1.2PIN.PDF) and [schema diagram](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/blob/master/SKR-PRO-V1.2/Schematic/SKR-PRO-V1.2.PDF).

---

In `Marlin/src/pins/stm32f4/pins_BTT_SKR_PRO_common.h`, these are currently defined as:

```
#define FAN_PIN                             PC8   // Fan0 (parts cooling fan)
#define FAN1_PIN                            PE5   // Fan1 (configurable: e.g. extruder cooling fan)
#define FAN2_PIN                            PE6   // Fan2 (configurable: e.g. controller fan)
```

Unfortunately, Marlin is hard-coded to use `FAN_PIN` (`Fan0` on the board) as a parts-cooling fan and is reserved/cannot be used for anything else in code. That's why we get `You cannot set E0_AUTO_FAN_PIN equal to FAN_PIN.` and/or `You cannot set CONTROLLER_FAN_PIN equal to FAN_PIN.` when compiling.
See [this post](https://github.com/MarlinFirmware/Marlin/issues/10316#issuecomment-379525737) for clarification from @thinkyhead:
> Since the common FAN_PIN (part cooling fan) has a setting for all boards and is always defined, if you need to use that pin, you would have to comment it out in the relevant pins_MYBOARD.h file. Currently the pin isn't overridable from configurations (at least, without throwing a #warning).

---

A workaround I did was to redefine `PC8/Fan0` PIN as another fan (e.g. `FAN3`):
```
#define FAN3_PIN                            PC8   // Fan0 (e.g. chamber cooling fan)
#define FAN1_PIN                            PE5   // Fan1 (e.g. extruder cooling fans)
#define FAN2_PIN                            PE6   // Fan2 (e.g. controller fan)
```

You can then use `FAN3_PIN` in the Marlin code as an extra fan to do anything you want (e.g. use for `CHAMBER_AUTO_FAN_PIN`). Yes, someone reading your Marlin config will wonder what `FAN3_PIN` is since exists none physically on the board, but hopefully they will check the `pins_BTT_SKR_PRO_common.h` file for clarification.
AFAIK, I don't see any negative consequences of no `FAN_PIN` defined in Marlin, and you get an extra fan to play with!

**Note:** This means that you'll need to wire in a separate parts cooling fan to +12V somewhere on the board! Easy location, bottom right of board:
[image]

---
I agree with others here. Ideally, we should change Marlin itself not to hard-code `FAN_PIN` as a reserved, non-programmable fan header. The problem is that 99% of other boards out there does not have a programmable `FAN_PIN`, so it would break existing boards?

tablatronix, 2020-07-03:
Not sure if this is the same or related, but E0_AUTO_FAN_PIN also does not allow for M106 fan control, as it is not a "fan", which is annoying for both testing and dynamic gcode.