MarlinFirmware/Marlin #17303: [FR]Support Sensorless Endstops for Extruders

tmartin007, 2020-03-26:
Context (example):

MB:                   BigtreeTech SKR Pro V 1.1
Motor-Drivers: TMC2209

This concerns a CNC6040 being converted into a PNP robot with an automatic adhesive dispenser. - The dispenser consists of a syringe - the plunger is moved by a spindle driven by a stepper motor. When the adhesive is completely pushed out, the plunger is at the bottom of the syringe and cannot be moved "downwards" any further. When the syringe is filled (automatically), the "top" part of the piston will eventually (e.g.) bounce into the "bottom" part of the shaft-coupler that connects the spindle to the motor and cannot be moved "upwards" any further.

It would be great, if sensorless endstops could be configured for both ends.

Thank you very much in advance for considering.

Best regards!

rjacobs1969, 2020-04-20:
I can see this being useful for 3D printers as well, Using stallguard it would be possible load filament automatically even when the hotend is cold

marcossfernandez, 2020-10-20:
I'm testing with sensorless Extruder for syringe as well with no success at this moment. I been able to do so with phisical endstops which it isn't ideal.
I'm stuck in interpreting the DIAG1 pin signal for E_Min_Pin (1.26) to trigger the sensorless endstop for E.

DerAndere1, 2020-12-22:
You can use my 6-axis-Marlin. It has support for conventional Endstops/ Software Endstops and auto-homing on 6 axes (X,X,Z,A,B,C). Here is the documentation with more info: https://github.com/DerAndere1/Marlin/wiki

marcossfernandez, 2020-12-22:
Thank you. Already had sucessfully done with physical endstops several months ago. But for a syringe extruder doing sensorless make it simple detect when empty and trigger a pause.
The main limitation i think is the speed needed to detect motor stall for e-endstop  because It is way too low. I contacted Trinamic for advice but they doubt this approach will work.
Maybe the main developers of Marlin or Temualt can give a solution in the future.

sjasonsmith, 2020-12-22:
The general consensus is that switches are a lot less trouble than sensorless.

DerAndere1, 2020-12-22:
Especially if you have lead screws on the axis, I fear that sensitivity of the stallguard is not good enough, so it will trigger late

marcossfernandez, 2020-12-22:
Yes, we have ballscrews for syringe extruders. Speed is the most limitation part, not sensitivity or repeatibility.
It's needed some speed to detect the stall

stylesuxx, 2021-03-16:
@marcossfernandez do you mind sharing your code for the version with physical end-stops. I am having a similar use-case and it would save me quite some time if you could share your solution. Thank you.

tmartin007, 2021-03-16:
Hi Chris,

I’d love to share my code, however, I had paused my project about a year ago 
and - in the meantime - I have forgotten a lot. - I am planning to resume this  
work in April and will let you know, if that is not too late for you.

Cheers,

-Thomas

marcossfernandez, 2021-03-16:
> @marcossfernandez do you mind sharing your code for the version with physical end-stops. I am having a similar use-case and it would save me quite some time if you could share your solution. Thank you.

I can share the 1.1.9 version with functional e-endstop. With newer 2.x versions hadn't get lucky to do that, and left aside.
For this project we switched to Duet and RRF, which it's easy to do it.

stylesuxx, 2021-03-16:
@tmartin007 Yes - please let me know.

@marcossfernandez Would be happy to see your 1.x solution. What were the issues to get it implemented on 2.x?

marcossfernandez, 2021-03-16:
> @marcossfernandez Would be happy to see your 1.x solution. What were the issues to get it implemented on 2.x?

I'll share it as soon I get some free time.

marcossfernandez, 2021-03-16:
I can't find right now where are the files, it was a old project that we did, but i get the notes we did for e-endstop modifications (marlin 1.1.9 custom+ramps 1.4 hw). This is the code we do to get the e-endstop work (trigger a pause when the syringe is empty and allow retract and purge for a new one). 

**Modifications in Configuration.h:**

```
#define USE_EMIN_PLUG

#define E_MIN_ENDSTOP_INVERTING true // set to true to invert the logic of the endstop.

#define E_MIN_POS 0
```

**Modifications in MarlinMain.cpp**	(we created a custom gcode to use diverse tools, one with a syringe which use the e-endstop when empty):

```
   if (selectedHead==1){
      endstops.e_axis_interruption();
   }   
  endstops.event_handler();
  idle();
```
  
**Modifications in Pins_ramps.h:**

`#define E_MIN_PIN          15`

**Modifications in Endstops.cpp**

```
// public:
void GcodePausa();
bool Endstops::enabled, Endstops::enabled_globally; // Initialized by settings.load()
volatile uint8_t Endstops::hit_state;

Endstops::esbits_t Endstops::live_state = 0;

  #if HAS_E_MIN
    #if ENABLED(ENDSTOPPULLUP_EMIN)
      SET_INPUT_PULLUP(E_MIN_PIN);
    #else
      SET_INPUT(E_MIN_PIN);
    #endif
  #endif
  
  #define ENDSTOP_HIT_TEST_E() _ENDSTOP_HIT_TEST(E,'E')
     ENDSTOP_HIT_TEST_E();
	 
  #if HAS_E_MIN
    ES_REPORT(E_MIN);
  #endif
  
    #if HAS_E_MIN
      UPDATE_ENDSTOP_BIT(E, MIN); 
  #endif
  
    void Endstops::e_axis_interruption() {
    
     UPDATE_ENDSTOP_BIT(E, MIN); 
      if (stepper.axis_is_moving(E_AXIS)) {//
       if (!stepper.motor_direction(E_AXIS)) { // +direction
        #if HAS_E_MIN
           if (READ(E_MIN_PIN) != E_MIN_ENDSTOP_INVERTING){
       
                _ENDSTOP_HIT(E, MIN);
                GcodePausa();
                planner.endstop_triggered(E_AXIS);
            }
         #endif
      }    
     }    
  }//e_axis_interruption 
```
  
  **Modifications in Stepper.cpp:**

```
  if (!!current_block->steps[E_AXIS]) SBI(axis_bits, E_AXIS);
      //if (!!current_block->steps[A_AXIS]) SBI(axis_bits, X_HEAD);
      //if (!!current_block->steps[B_AXIS]) SBI(axis_bits, Y_HEAD);
      //if (!!current_block->steps[C_AXIS]) SBI(axis_bits, Z_HEAD);
      axis_did_move = axis_bits;
```

**Modifications in Endstops.h:**

#define VALIDATE_HOMING_ENDSTOPS

```
enum EndstopEnum : char {
  X_MIN,
  Y_MIN,
  Z_MIN,
  Z_MIN_PROBE,
  X_MAX,
  Y_MAX,
  Z_MAX,
  X2_MIN,
  X2_MAX,
  Y2_MIN,
  Y2_MAX,
  Z2_MIN,
  Z2_MAX,
  E_MIN, //3dlimitLess
  E_MAX //3dlimitLess
};
```

And also we did a lot of modifications in Configuration_store.cpp but not related to e-endstop. 

With 2.x version we did a similar approach but we encountered some problems related to pause and buffers (the pause wasn't fast enough) that we didn't fight. After that, we switch to RRF and Duet, because Marlin was limited when heading to build non standard machines.

I hope could been help you a bit.

stylesuxx, 2021-03-16:
@marcossfernandez awesome, thank you - that is pretty much what I am attempting right now on 2.x

DerAndere1, 2021-03-16:
@stylesuxx: for most purposes, 6 axis Marlin works well. It supports physical endstops on up to 6 axes (axis names XYZABC or XYZUVW).  The most up-to-date code (https://github.com/DerAndere1/Marlin/tree/6axis_PR1 ) is based on current Marlin bugfix-2.0.x. In case you want to be able to auto-home the E axis with G28, marcossfernandez changes are not enough. Source code for Marlin 1.9.x with Endstops on the E axis and auto-homing (supports G-code Syntax G28 E) can be found in another branch of my repository, here: https://github.com/DerAndere1/Marlin/blob/Marlin2ForPipetBot_v1 .

marcossfernandez, 2021-03-16:
Interesting to know that your 6axis version does it well, specially with e-endstops. We pretended to get it sensorless but it's impossible. I would take as start point your code if back to Marlin Fw for this kind of printer. Great job!

Thank you a lot

stylesuxx, 2021-03-16:
@DerAndere1 Very interesting, I will check it out. I need some features from the Marlin bugfix-2.0.x branch - so that is great news that you support that. From what I can see, you are planning on pushing your changes back upstream - right? Thanks for the heads up.

DerAndere1, 2021-03-16:
Yes I opened a pull request to get https://github.com/DerAndere1/Marlin/tree/6axis_PR1 merged into upstream FirmwareMarlin/Marlin. 6 axis Marlin does NOT support endstops on E axis. But with 6 axis Marlin, you can a 4th axis (default axis name: 'A') with physical endstops for driving additional tools like syringes. Physical endstops on the E axis are only supported by the modified Marlin 1.9.x (https://github.com/DerAndere1/Marlin/tree/Marlin2ForPipetBot_v1). By the way: you may be interested in my syringe pump robot control software: pyotronext (https://gitlab.com/RobotsByDerAndere/pyotronext) for scripting using opentrons apiv2 protocol style Python scripts) and the GUI GGCGen (https://gitlab.com/RobotsByDerAndere/CartesianRobotsByDerAndere/GGCGen).

marcossfernandez, 2021-03-17:
In our attempt with the 2.x version that we made, we were able to pause when pressing the e-endstop, but we tried to replicate partially the operation of the M600 functions without success. All without SD or LCD, only using host software as interface (repetier server). It was a mess.

Our intention was to be able to leave alone printing with a syringe and that it was emptied, pause printing, park it in a certain area and the operator be able to replace it with another loaded one, purge and clean and then continue printing.

We couldn't get the entire chain to work properly.

It is good to know that there is at least one other point of view and to be able to try it and give Marlin another chance.

Thanks @DerAndere1

tmartin007, 2021-05-11:
Dear DerAndere,

thank you very much for your message.

My project had been paused about a year ago, and I have only just resumed the work, yesterday.

> You can use my 6-axis-Marlin. It has support for conventional Endstops/ Software Endstops and auto-homing on 6 axes (X,X,Z,A,B,C)

I very much appreciate the hint and offer! - This seems to be exactly, what I was looking for.
I have already updated my project form Marlin 2.0.4.4 to 2.0.8. Hence, it should be straight forward to “migrate” to your 6-axis branch.

I have read the info at: https://github.com/DerAndere1/Marlin/wiki <https://github.com/DerAndere1/Marlin/wiki> - but I am not completely certain yet 
what needs to be done and would very much appreciate a little guidance … - thank you very much in advance for you efforts!

I am using a BIGTREE SKR Pro V1.1 (which can support up to 6 motor drivers) with TMC 2209 on X, Y, Z and E0, where E0 is intended to
drive the plunger of a syringe to micro-dispense some adhesive. I would like to “sensorless home” the plunger of the syringe, where 
the home position of the plunger would be: all-the-waydown (i.e. syringe is empty). 
I was hoping that this can be accomplished with your Multi-axis-Marlin firmware.

I do understand step 1)  :-) change  #define LINEAR_AXES 3 to #define LINEAR_AXES 4
Step 1) seems to implicitly establish “I” as a new valid prefix for a “homable" axis.

However, I am  not sure, wether I correctly  understand step 2) 
i.e. the definition of:

 [I, [J, [K]]]_STEP_PIN,
 [I, [J, [K]]]_ENABLE_PIN,
 [I, [J, [K]]]_DIR_PIN,
etc

Could you please confirm or correct me:

Looking for only 1 additional axis to be “sensorless homable”, only “I” would be relevant.
Hence, I have to define the following symbols to suit the dispenser setup (i.e. stepper, syringe size, etc):

 #define I_STEP_PIN   E0_STEP_PIN,
 #define  I_ENABLE_PIN E0_ENABLE_PIN,
 #define  I_DIR_PIN    E0_DIR_PIN,
etc

Will the above create “I” as some kind of ‘alias' for “E0”  (making “I” homable with G28I)
while the motor is connected to the driver currently controlling “E0” ?

Also, the following c-macros all seem to expect 4 arguments (i.e. for X, Y, Z, E0).
How shall I provide values for “I” (if more homeable axis (i.e. J,K) were needed, 
how would their values be assigned)

DEFAULT_AXIS_STEPS_PER_UNIT, 
DEFAULT_MAX_FEEDRATE, 
DEFAULT_MAX_ACCELERATION, 
HOMING_FEEDRATE_MM_M, 
AXIS_RELATIVE_MODES, 
MICROSTEP_MODES, 
MANUAL_FEEDRATE,

HOMING_BUMP_DIVISOR,
HOMING_BACKOFF_POST_MM, 
BACKLASH_DISTANCE_MM. 

It would be great, if you could refer to some sample configuration utilizing e.g. 4 homable axis.

Again, for you efforts, thank you very much in advance!

Best regards,

Thomas

thisiskeithb, 2021-05-16:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/14497

closed 2021-05-16 by thisiskeithb

tmartin007, 2021-05-16:
Hi Keith,

> Duplicate of #14497 <https://github.com/MarlinFirmware/Marlin/issues/14497>sorry to disagree - 14497 <https://github.com/MarlinFirmware/Marlin/issues/14497> is only remotely related. - This, however, is not about "filament run-out",
but sensorless “homing” of other axis than “x”,”y”,”z" (e.g. “e”) instead.  DerAndere1’s solution
points into the right direction (likely leading to a solution), but there are still open questions ...

marcossfernandez, 2021-07-15:
Agree. It's not the same as filament sensor M600, so duplicate not aply here. Concept is similar to M600 but not their behaviour.

We demand a inmediate stop or pause on endstop/sensorless for A EXTRUDER axis. Mainly would be dispensers or syringes that has physical limits. 
Also it would need to stop the command buffer inmediatelly and split the part not extruded in the continue command buffer after continue. Maybe doing a segmentation as RRF3.3 does with their **M669 Sxxx** to monitor the endstop more times a second:
https://duet3d.dozuki.com/Wiki/Gcode#Section_M669_Set_kinematics_type_and_kinematics_parameters

thisiskeithb, 2021-07-15:
> Agree. It's not the same as filament sensor M600, so duplicate not aply here. Concept is similar to M600 but not their behaviour.

Marlin can execute whatever command you'd like when a filament runout event is detected:

https://github.com/MarlinFirmware/Marlin/blob/3be35a6bd6845a160c416629ea98392129cd6eb3/Marlin/Configuration.h#L1469-L1472

The discussion should occur in the linked FR.

marcossfernandez, 2021-07-15:
How do I trigger an **immediate** shot for the M600 command, without LCD or SD card? Pause, for example, **in the middle** of a large G1 extrude command? How can I split the buffer command and continue? All this done with a printer only interconnected with a host (repeater host or server / pronterface / octoprint)

We didn't get it to work with the M600 this way, without hacking the Marlin code in an extreme way. And our conclusion was that it was impossible with sensorless Trinamic and very difficult with the physical limit switch (@ 2.0.7.2 version at least). We gave up.

DerAndere1, 2021-07-21:
Support for additional axes is now merged into MarlinFirmware/Marlin bugfix-2.0.x. For your usecase it makes sense to writa a dedicated host software. You can base your host software on printrun which is open source.  Design the host software so that you can specify a volume as input. The host could split big movements into tiny segments corresponding to 1µL and make printrun send multiple small G1 moves that in total correspond to the desired volume. Then you can use `FILAMENT_RUNOUT_SCRIPT` and `M600` in Marlin.  Regarding setup: I would suggest to configure Marlin with `EXTRUDERS 0` and LINEAR_AXES 4 and `AXIS4_NAME 'A'`. I opened a discussion here: https://github.com/DerAndere1/Marlin/issues/51 .