MarlinFirmware/Marlin #17304: 2.0.5.2 Compiler Warning with "USBComposite_stm32f1@==0.91"

Lord-Quake, 2020-03-26:
This concerns commit https://github.com/MarlinFirmware/Marlin/commit/1674df00b14009e7c3c2daa5c8585cabe3fc7fc4

The effected line in platformio.ini should read:

`USBComposite for STM32F1@==0.91`

Taomyn, 2020-03-28:
I made the same change and the warning disappears, thanks

thisiskeithb, 2020-03-28:
I verified the fix on an SKR Mini E3 1.2 build, so I put in a PR: https://github.com/MarlinFirmware/Marlin/pull/17321.

damvcoool, 2020-03-28:
I just build simply using USBComposite for STM32F1, which uses v0.97
Is there a specific reason why i should not use a newer Version of USBComposite for STM32F1

it's currently working fine for me (i am just wondering)

thisiskeithb, 2020-03-28:
> Is there a specific reason why i should not use a newer Version of USBComposite for STM32F1

I’m not sure what prompted the specific version requirement.

ellensp, 2020-03-29:
see #17217 for details

closed 2020-03-29 by thinkyhead