MarlinFirmware/Marlin #17306: [BUG] Extruder stops extruding during prints

Mosher23, 2020-03-26:
### Bug Description
So my extruder stops extruding during prints with no failure message.
On different prints it stops in different places, but in Linear Advance calibration code every time on 2nd line.

### My Configurations
Marlin 2.0.5
SKR 1.4 Turbo TMC5160
E3D Hemera
[Configuration Files.zip](https://github.com/MarlinFirmware/Marlin/files/4389696/Configuration.Files.zip)
[LINADVkfactor.gcode.zip](https://github.com/MarlinFirmware/Marlin/files/4389699/LINADVkfactor.gcode.zip)
### Steps to Reproduce
1. use g-code i have provided
2. Print
3. it Fails on 2nd line

**Expected behavior:** 
Print with no extrusion stop

**Actual behavior:** 
Extrusion stops with no warning or feedback whatsoever

#### Additional Information
This bug is related to this issue i think [#16076](https://github.com/MarlinFirmware/Marlin/issues/16076)

Exalik, 2020-03-28:
Hello,

I have the same issue on my Geeetech A10M (board GT2560 v3.1) with TMC2208

Mosher23, 2020-03-28:
So i did one more Test print now and it failed on the same spot. 
This time i made debug logging of my TMC drivers to see if they cause it.

```
; start the Test pattern
;
G4 P2000 ; Pause (dwell) for 2 seconds
G1 X71 Y72.5 F7200 ; move to start
M900 K0 ; set K-factor
M117 K0 ; 
G1 E0.5 F2700 ; un-retract
G1 X91 Y72.5 E0.7982 F1200 ; print line
G1 X131 Y72.5 E1.5965 F6000 ; print line
G1 X151 Y72.5 E0.7982 F1200 ; print line
G1 E-0.5 F2700 ; retract
G1 X71 Y77.5 F7200 ; move to start
M900 K0.2 ; set K-factor
M117 K0.2 ; 
G1 E0.5 F2700 ; un-retract
G1 X91 Y77.5 E0.7982 F1200 ; print line
**G1 X131 Y77.5 E1.5965 F6000 ; print line** <- print stops extruding here
G1 X151 Y77.5 E0.7982 F1200 ; print line
G1 E-0.5 F2700 ; retract
G1 X71 Y82.5 F7200 ; move to start
 
```

[Failed Lin Advance Print log.txt](https://github.com/MarlinFirmware/Marlin/files/4396907/Failed.Lin.Advance.Print.log.txt)

[image]

Mosher23, 2020-03-29:
So i have found a problem, its Linear Advance.
After disabling it extrusion don‘t stop anymore.
So now i have a question if Linear advance is going to be fixed anytime soon?

ellensp, 2020-03-29:
Today I watched a interesting youtube ( https://www.youtube.com/watch?v=VJ_TGoqVoM8 if interested) on how some people, despite the world wide pandemic, still expect their shit fixed yesterday if not sooner. When most of the planet is in lock down, and upside down...  I now see what he meant.   
Just turn off LA and be grateful your not dead.   
It will get fixed when it gets fixed.

Mosher23, 2020-03-29:
> Today I watched a interesting youtube ( https://www.youtube.com/watch?v=VJ_TGoqVoM8 if interested) on how some people, despite the world wide pandemic, still expect their shit fixed yesterday if not [...]

I don’t need your opinion if its not on subject.
GTFO

sjasonsmith, 2020-03-29:
The reality is that this is unlikely to be fixed any time soon. It has been like this for a long time, and nobody has stepped up to fix it. It is waiting for the right person to come along and care enough to learn about the problem and implement a solution. I gave it about a week of my time before deciding it was not a priority for me. Maybe that person is you?

Mosher23, 2020-03-29:
> The reality is that this is unlikely to be fixed any time soon. It has been like this for a long time, and nobody has stepped up to fix it. It is waiting for the right person to come along and care [...]
I would really like to help with that problem, but i am not a programmer.

petrce, 2020-04-19:
is M900 K0 at the beginning of the gcode enough to work around this?

randellhodges, 2020-04-19:
I don't know about the 5160, but with the 2208 you could not have `#define STEALTHCHOP_E`. Try commenting that out and let E run in spreadcycle.

If that doesn't do it, your only option is to disable LA completely, that is, comment it out and compile.

ipartola, 2020-06-06:
@sjasonsmith Can you elaborate on what the actual problem is and what you discovered? I don't have experience with the Marlin codebase in particular but I am interested in at least understanding what the problem is and whether it's just how this board is wired or if it's a software issue, if not taking a whack at it.

sjasonsmith, 2020-06-06:
> @sjasonsmith Can you elaborate on what the actual problem is and what you discovered? I don't have experience with the Marlin codebase in particular but I am interested in at least understanding [...]

The steps that are generated by linear advance are not coordinated with the normal extrusion steps. This results in inconsistent pulse timing. I suspect this looks like extremely fast acceleration to TMC drivers in some situations, resulting in them shutting down.

ipartola, 2020-06-06:
> The steps that are generated by linear advance are not coordinated with the normal extrusion steps. This results in inconsistent pulse timing. I suspect this looks like extremely fast acceleration to TMC drivers in some situations, resulting in them shutting down.

Got it. So theoretically this is a strictly Marlin problem that can be solved by syncing the LA steps with the regular extruder timing? I guess I should start reading the documentation about where this code is and how to test it.

boelle, 2020-06-21:
@Mosher23 still an issue?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle

mianos, 2020-07-13:
This is still an issue.

randellhodges, 2020-07-13:
@mianos Did you try it with stealthchop disabled on E?

https://github.com/MarlinFirmware/Marlin/issues/17306#issuecomment-616016200

mianos, 2020-07-13:
My apologies! Yes, with the with STEALCHOP for the extruder commented out it keeps moving the extruder. 
No closed due to lack of activity. Close due to a solution.
Thanks.

cab404, 2020-07-31:
https://gist.github.com/cab404/09ae2db0d52df49ead473d5ebeae3bc4

Got stealthchop and s curve disabled, still fails halfway through a benchy. 

Interestingly, it looks like it starts failing at about the same physical height, no matter the size of a benchy. (N=3, deviation ±1mm)

Going to enable TMC debug, and see what's happening.

cab404, 2020-07-31:
I noticed that while compiling with TMC2208_STANDALONE, TMCStepper dependency doesn't get used, and I can't enable TMC_DEBUG. Probably I can't control StealthChop as well.

cab404, 2020-07-31:
@randellhodges ?

cab404, 2020-07-31:
[I read a bit more about standalone mode](https://html.alldatasheet.com/html-pdf/958414/TRINAMIC/TMC2225-SA/1959/5/TMC2225-SA.html): there are three variants of TMC2225-SA (one I have, compatible with 2208) configuration
1. standalone, not configuring at all
2. standalone, configure once
3. FULL POWER

Currently, I see support for mode 1 and 3. Is there a way to support mode 2 with available pins?

cab404, 2020-08-02:
So, after a lot of tinkering with my drivers, I finally succeeded in flashing EN_SPREADCYCLE=1 and TOFF=2 into OTP, and now it works flawlessly on my X5SA. All standalone driver users should either disable lin_pressure, or use that. 

I will make a pull request to documentation soon.