MarlinFirmware/Marlin #17308: [BUG] M48 Fail

inventabuild, 2020-03-27:
### Bug Description: Sending M48 from Pronterface I hear clicking that sounds like it's coming from the Z-axis motors, but the X-axis doesn't move down at all, it just hangs there.

### My Configurations
Running Marlin 1.1.9.34 which is Lulzbot's latest release for the TAZ 6.  Attached are my config files. 

[Conditionals_LulzBot.h.txt](https://github.com/MarlinFirmware/Marlin/files/4390690/Conditionals_LulzBot.h.txt)
[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/4390691/Configuration.h.txt)
[Configuration_adv.h.txt](https://github.com/MarlinFirmware/Marlin/files/4390692/Configuration_adv.h.txt)

thisiskeithb, 2020-03-27:
Since you’re using a fork of Marlin, you should post in their [issue queue](https://gitlab.com/groups/lulzbot3d/-/issues).

Roxy-3D, 2020-03-27:
And that means M48 failed?   

Is the Z axis jammed and unable to move?

inventabuild, 2020-03-27:
> And that means M48 failed?
> 
> Is the Z axis jammed and unable to move?

Yes, M48 failed to execute properly as the Hot End on the X-axis never went down to tap the bed, it just stood still about 50 mm off the bed where it was before I sent M48.

No the Z-axis is not jammed because I can home w/ G28 and Auto Bed Level w/ G29 with the Hot End going down and tapping the bed like it's supposed to, but w/ G48 the Hot End never goes down.

Please let me know if you have any idea where the hang up is.

Roxy-3D, 2020-03-28:
Well...   I kind of doubt it is M48 just because nobody else is complaining about it, and it hasn't changed in a long time.    It is more likely some regression has crept into the probing code.

You might want to see if you can probe the bed and build a mesh for bed leveling.   It is very possible you will see the same issue.

inventabuild, 2020-04-12:
If M48 was working properly in Marlin 1.1.9.34 then Lulzbot's programmers broke M48 with their modifications to Marlin for their TAZ printers.  Bilinear Bed Leveling was also broken, but Linear Bed Leveling was working fine and that's what the TAZ printers used with their 4 point corner taps for auto bed leveling.

I upgraded to Marlin 2.0.5.3 and all is good now with M48 and Bilinear Bed Leveling.

closed 2020-04-12 by inventabuild

Roxy-3D, 2020-04-13:
Well, the good news is it is all 'Open Source'.   If there is something really good in their code, we can move it to the main branch.