MarlinFirmware/Marlin #17310: Firmware rebooting after print is done

byemi, 2020-03-27:
### Bug Description

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

The issue happens every time and after a print is finished. There are no steps needed in order to reproduce the issue. 

**Expected behavior:** [What you expect to happen]
Of course the expectation would be not to have this issue.

**Actual behavior:** [What actually happens]

The issue is visible on the LCD and happens even on the entire system. More exactly, the firmware reboots after the print is finished and the axis are going into home position. The hardware on which the 3d printer works is a simple ramps 1.4 + arduino with Mega2560 + 12864 LCD. 
The firmware version running is 2.0.5

#### Additional Information
There are no additional information that can be added for the moment. 
[Config_files.zip](https://github.com/MarlinFirmware/Marlin/files/4392282/Config_files.zip)

Exalik, 2020-03-27:
Hello,

I have a Geeetech A10M with TLC2208 and BLTouch.

I have the same issue with Marlin 2.0.4.4

With Marlin 2.0.5, the print does not start, it stops just after the bed level.

boelle, 2020-06-20:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-20 by boelle

byemi, 2020-06-21:
Hello Bo,
Thank you for closing the topic. As you can also see, there were no solutions for the problem I still have and looking forward to solve the issue. Keep the topic closed, as I see that there is no chance of someone to give me an answer. Sorry for not closing the topic long ago.