MarlinFirmware/Marlin #17311: [BUG] Print freeze when use heated bed

lucableve, 2020-03-27:
Hi, I have a problem with the printer on skr 1.4, when it starts to print the self-leveling is successful but when the extrusion starts octoprint gives me an error.
The strangest thing is that the print after 10 cm clicks and then always hangs in the same point.
If I launch the print without a heated bed, the print is successful.

I tried to :
Remove 5v usb octoprint
Changed power supply with one from 30a
Made the hotend and bed pid
To exclude octoprint I tried to print from sd, always the same error.

What can it be ?

I attach the configuration files and logs
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4392958/Marlin.zip)
[serial.log](https://github.com/MarlinFirmware/Marlin/files/4392991/serial.log)
[octoprint.log](https://github.com/MarlinFirmware/Marlin/files/4392992/octoprint.log)

studiodyne, 2020-04-01:
Need to know if it's hardware or software , then , disable thermal protection for all the machine a compile a new firmware , try to print with the bed ,if it's successfull ,  you know that you have a software issue or settings about thermal protection

LaFouy, 2020-04-05:
I have a similar concern but with SKR 1.3
- temperature setpoints are ok (bed heats up and nozzle too)
- once the temperatures are reached I launch the print
- G28 probing in the center of the bed .. ok
- G29 the probe is positioned at the bottom left for the 1st probe. Immediately a MINTEMP error message and the bed / nozzle switch off.
All my print launches stop there with these MINTEMP messages.
if I print without heating instructions for the bed, it works (2 successful prints..1h30 and 50min approximately).

and with thermal protection disabled, printing still fails.. hadware issue.. defective bed temperature sensor I guess ?

lucableve, 2020-04-05:
Hi, In the end, due to exhaustion, I changed all sensors of temps and now working , but I noticed that in version 2.0.5 the bltouch tolerances are very tight. After arranging the bltouch magnet and perfectly adjusting the plate and bltouch height, the g29 is successful and prints.

closed 2020-04-05 by lucableve

thinkyhead, 2020-04-05:
The `PROBING_HEATERS_OFF` and `PROBING_FANS_OFF` options are recommended if electrical issues with the board might cause interference with the probe. This is specifically for noise, not low current, but it helps with both.

LaFouy, 2020-04-06:
Hi, juste to let you know that i changed the bed sensor of temperature and everything is fine now for me too.