MarlinFirmware/Marlin #17313: Can't compile for printrboard/Teensy and UBL -ubl_motion.cpp:34:17: error: redefinition of 'void set_current_from_destination()'

sredoje1973, 2020-03-27:
Trying to compile using marlin-config and using config files below.  

Can compile "stock" config files with ATmega board no problem.  I might have to double check.  When Teensy2p selected,  I get following compile error. No other errors in compile (a few warnings though)
:

---------------------------------------------------------------
marlin\ubl_motion.cpp: In function 'void set_current_from_destination()':
Marlin\ubl_motion.cpp:34:17: error: redefinition of 'void set_current_from_destination()'
     inline void set_current_from_destination() { COPY(current_position, destination); }
                 ^
In file included from Marlin\ubl_motion.cpp:26:0:
Marlin\Marlin.h:549:13: note: 'void set_current_from_destination()' previously defined here
 inline void set_current_from_destination() { COPY(current_position, destination); }
             ^

------------------------------------------------------------------

//file Configuration.h Release:tags/1.1.9
//section info
#define STRING_CONFIG_H_AUTHOR "(Dr.Phil3D)"
//#define SHOW_BOOTSCREEN 
//section machine
#define MOTHERBOARD BOARD_PRINTRBOARD_REVF
#define CUSTOM_MACHINE_NAME "Printrbot Simple Metal"
#define MACHINE_UUID "40ce61d1-2ed0-4530-8304-55e3eeaaebff"
#define INVERT_X_DIR true
#define INVERT_Z_DIR true
#define X_BED_SIZE 152
#define Y_BED_SIZE 152
#define Z_MAX_POS 152
//section extruder
#define DEFAULT_NOMINAL_FILAMENT_DIA 1.75
#define HOTEND_OFFSET_Z.0 { 0.0, 1.3 }
//section temperature
#define TEMP_SENSOR_CHAMBER 7
#define TEMP_BED_HYSTERESIS 5
#define HEATER_0_MINTEMP 3
#define BED_MINTEMP 3
#define DEFAULT_Kp.0 40.15
#define DEFAULT_Ki.0 7.10
#define DEFAULT_Kd.0 56.73
#define PIDTEMPBED 
#define DEFAULT_bedKp.0 70.61
#define DEFAULT_bedKi.0 7.18
#define DEFAULT_bedKd.0 173.53
#define PREHEAT_1_TEMP_HOTEND 200
#define PREHEAT_2_TEMP_HOTEND 230
//section homing
//#define USE_YMIN_PLUG 
#define USE_YMAX_PLUG 
#define Z_MIN_ENDSTOP_INVERTING true
#define Z_MIN_PROBE_ENDSTOP_INVERTING true
#define X_DRIVER_TYPE A4988
#define Y_DRIVER_TYPE A4988
#define Z_DRIVER_TYPE A4988
#define E0_DRIVER_TYPE A4988
#define Z_HOMING_HEIGHT 3
#define Y_HOME_DIR 1
#define Z_SAFE_HOMING 
//section motion
#define DEFAULT_AXIS_STEPS_PER_UNIT { 80, 80, 2020, 94.5 }
#define DEFAULT_MAX_FEEDRATE { 150, 150, 7, 40 }
#define DEFAULT_MAX_ACCELERATION { 2000, 2000, 100, 10000 }
#define DEFAULT_ACCELERATION 1500
#define DEFAULT_TRAVEL_ACCELERATION 1500
#define DEFAULT_EJERK 4
#define S_CURVE_ACCELERATION 
//section probes
#define FIX_MOUNTED_PROBE 
#define X_PROBE_OFFSET_FROM_EXTRUDER 20
#define Y_PROBE_OFFSET_FROM_EXTRUDER 0
#define XY_PROBE_SPEED 7000
#define MULTIPLE_PROBING 2
#define Z_CLEARANCE_DEPLOY_PROBE 4
#define Z_CLEARANCE_BETWEEN_PROBES 3
#define Z_CLEARANCE_MULTI_PROBE 3
#define Z_PROBE_OFFSET_RANGE_MIN -4
//section calibrate
#define AUTO_BED_LEVELING_UBL 
#define RESTORE_LEVELING_AFTER_G28 
#define G26_MESH_VALIDATION 
#define MESH_TEST_BED_TEMP 60
#define PROBE_Y_FIRST 
#define EXTRAPOLATE_BEYOND_GRID 
#define MESH_INSET.0 10
//#define Z_PROBE_END_SCRIPT "G1 Z10 F7000\nG1 X0 Y0 F7000"
//section extras
#define EEPROM_SETTINGS 
#define FAN_SOFT_PWM 
#define NEOPIXEL_STARTUP_TEST 
//section lcd
#define SDSUPPORT 
#define NO_LCD_MENUS 
#define REPRAP_DISCOUNT_SMART_CONTROLLER 

//file Configuration_adv.h Release:tags/1.1.9
//section temperature
#define PID_EXTRUSION_SCALING 
//section extruder
#define E0_AUTO_FAN_PIN 44
#define LIN_ADVANCE 
#define LIN_ADVANCE_K 0.08
//section extras
#define DAC_MOTOR_CURRENT_DEFAULT {70,70,65,70}
//section lcd
#define BABYSTEPPING 
#define BABYSTEP_ZPROBE_OFFSET 
#define DOUBLECLICK_FOR_Z_BABYSTEPPING

sjasonsmith, 2020-04-13:
Can you attach configuration files as required by the issue template? Nobody wants to apply all those settings manually to reproduce your issue.

sredoje1973, 2020-04-17:
Hi.  configuration.h and configuration_adv.h attached as requested. 
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4493377/Configuration.zip)

thinkyhead, 2020-04-17:
You mixed old and new files together. Get a fresh download and just copy over your configs.

closed 2020-04-17 by thinkyhead

sredoje1973, 2020-04-17:
Maybe I wasn't clear.  If I take the original files from source, and change settings to enable UBL, it compiles fine for ATmega..  If I try to compile for Teensy2 it fails as above.  Attached is configuration.h file that I used to go through those exact steps. 
[Configuration.h.zip](https://github.com/MarlinFirmware/Marlin/files/4494430/Configuration.h.zip)

thinkyhead, 2020-04-18:
But there is no Marlin\Marlin.h file any more. Your source is out of date.