MarlinFirmware/Marlin #17316: Faulty SD card reader or need to update Marlin

toddy333, 2020-03-28:
I recently installed SKR 1.4 turbo and TFT 35 v3 (Ender version). I used the firmware from https://www.thingiverse.com/thing:4192395 and it was a good starting place except I disabled BL Touch and some other features that I wasn't using. It worked but I had issues with reading from my mainboard's sd card (sometimes i could read print from sd card but sometimes i could not). Now the situation got worse. I have to now keep an SD card in the mainboard or it won't work properly (screen won't stay on, usb to computer won't be detected, red and green light near sd card reader on mainboard cuts off). Once I put an SD card in there, everything works again (red and green light cut on again) except computer and mainboard doesn't detect sd card. This is the real funny thing: When I push in the sd card harder into the mainboard and restart the printer, all of the sudden I can read my sd card again from my computer and also my mainboard can upgrade firmware. But I have to keep the sd card pushed into the mainboard with some force. 

Can this be a Marlin config issue or is this a faulty sd card reader? I tried various sd cards, and same issue. Actually the mini sd card to standard size sd card adapter works best for my mainboard 's messed up sd card reader.

If I cannot fix the sd card reader, is there a simple way I can bypass my mainboard's sd card reader to upgrade Marlin firmware?

thisiskeithb, 2020-03-28:
You can try the latest `bugfix-2.0.x` from this repo, but I suspect your SD card slot/board is faulty.

toddy333, 2020-03-28:
Thanks for a quick reply. I will try out the latest firmware and also see what I can do with the seller. It hasn't been that long since I got the mainboard.

closed 2020-03-28 by toddy333