MarlinFirmware/Marlin #17317: BLTouch G29 Failure: CoreXY w/ Dual Z Axis

LabBoxPrinter, 2020-03-28:
### Bug Description

When G28 is used the machine homes correctly to X_min, Y_min, and Z_max. However, when G29 is used after, the machine moves the BLTouch to the correct position of the first probing point, touches the sensor pin to the bed once, then stops. Debugging over serial returns a "Error: Probing failed Count A4000 B -1600 Z4000"

### My Configurations

### Steps to Reproduce

1. Dual Z Axis CoreXY using  MKS SGEN L and TMC2209 drivers 
2. G28 
3. G29

**Expected behavior:** 

Autoleveling with G29 

**Actual behavior:** 

Probing of 1/8 point on the bed and then no more movement. Error of serial: "Error: Probing failed Count A4000 B -1600 Z4000"

In addition, G29 does not move the probe to the center of the bed first, as instructed by Z_SAFE_HOMING. But instead moves to the first probing position. 

#### Additional Information

The BLTouch is hooked up correctly to Z_min and SERVO_0 on my board. I tried using one of those Y adapters for the dual Z axis to see if that was what is causing the bug. However, after disabling dual z axis, and doing G29, Z_SAFE_HOMING worked but the Z_MIN endstop would not respond in software, and the hotend crashed into the bed. 

Z auto alignment seems to work with the probe 
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4396656/Marlin.zip)

TimoBre, 2020-05-10:
Hallo!
I had a similar behavior.
My dual Z axes are limited with optical limit switches. I also have a BLTouch 3.1.
My bed homes to all hardware entstops well. X Y and Z1 and Z2.
If I used the Z-axis command Z STEPPER AUTO ALIGN, both of them worked without problems, too.
The G28 command then did not work.
Tt moved to the first position, do the 1st probe and stopped. The homing order after it worked again. Just no more G28. Allways stopped after the 1st probing.

I changed in my version in Conditionals.lcd.h the line 598 (checked with Marlin 2.x from 2020 May 10th) 
from
#if Z_HOME_DIR < 0 && (!HAS_CUSTOM_PROBE_PIN || ENABLED(USE_PROBE_FOR_Z_HOMING))_
to
#if Z_HOME_DIR < 0 && (HAS_CUSTOM_PROBE_PIN || ENABLED(USE_PROBE_FOR_Z_HOMING))_

**Without the !** (!= do it negative)
My setup works fine
 
Here are my files. Changes are marked with  // geändert    
(German language)
I use notepad++ with the COMPARE plug in to compare it fast (maybe a hint).

My setup is a HEVO/24V/TMC2209/2208 both UART/double Z/Bltouch 3.1/ SKR 1.1 Pro,  CoreXY-System
I worked only in the files platformio.ini, Configuration.h,  Configuration_adv.h and  Conditionals_LCD.h

May it helps you

[05 Hevo Software.zip](https://github.com/MarlinFirmware/Marlin/files/4606050/05.Hevo.Software.zip)

LabBoxPrinter, 2020-05-10:
Make sure that your z offset is not 0

TimoBre, 2020-05-12:
@LabBoxPrinter
Why! A the moment i stand on the tube....
Sure more than 0 Negative the nozzle will pressed into the bed.

boelle, 2020-06-20:
@LabBoxPrinter is this still an issue?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle