MarlinFirmware/Marlin #17318: [BUG] twice [env:LPC1768] entry in platformio.ini

mcopta, 2020-03-28:
### Bug Description

#
# ATLPC1768
#
[env:LPC1768]

# NXP LPC176x ARM Cortex-M3
#
[env:LPC1768]

################################

VSC breaks with:
Error: Invalid 'C:\Users\User\Desktop\Marlin-2.0.x\Marlin-2.0.x\platformio.ini' (project configuration file): 'While reading from 'C:\\Users\\User\\Desktop\\Marlin-2.0.x\\Marlin-2.0.x\\platformio.ini' [line 64]: option 'lib_deps' in section 'env' already exists'

thisiskeithb, 2020-03-28:
Can you zip up your `platformio.ini` file and attach it?

I just compiled an `LPC1768` (SKR 1.3) build based on the latest `bugfix-2.0.x` without errors.

MarkusNiewoehner, 2020-03-28:
i also have no Problems with bugfix-2.0.x but with 2.0.5.2

thisiskeithb, 2020-03-28:
> My platformio.ini is just the same as in the 5.2 repo

The files are slightly different between the two, so make sure you're not recycling old files. Try downloading a fresh copy of `2.0.5.2` and see if the issue persists. I just compiled a `2.0.5.2`/`LPC1768` config without an error message.

Just for good measure, can you zip up and attach your configs (and `platformio.ini`) for review?

MarkusNiewoehner, 2020-03-28:
[platformio.zip](https://github.com/MarlinFirmware/Marlin/files/4397806/platformio.zip)
Don't know why this file is different and buggy(downloaded yesterday), the file here on github is ok now.

thisiskeithb, 2020-03-28:
Your `platformio.ini`:
```ini
#
# ATLPC1768
#
[env:LPC1768]
platform          = atmelavr
board             = megaatLPC1768
board_build.f_cpu = 16000000L
lib_deps          = ${common.lib_deps}
  TMC26XStepper=https://github.com/trinamic/TMC26XStepper/archive/master.zip
src_filter        = ${common.default_src_filter} +<src/HAL/AVR>
```

It should be:
https://github.com/MarlinFirmware/Marlin/blob/7124b2164d5c863ed1013df57a424739420003fc/platformio.ini#L57-L66

Start with a clean download of `2.0.5.2` and change _just_ the `default_envs` to `LPC1768` and nothing else in that file.

boelle, 2020-06-20:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-20 by boelle