MarlinFirmware/Marlin #17325: [BUG] G76 not working correctly

mavk0, 2020-03-29:
### Bug Description

(Hey, this is my first bug report and English is not my first language. Sorry about possible mistakes in reporting this.)

Pinda v2 probe temperature offset calibration process doesn't work correctly for me. I'm unsure if this is a bug or configuration problem. 

After executing G76 the calibration process starts fine: axis get homed, probe moves (up) to the cooldown point, bed heats up to 60c, probe centers itself on xy-coodinate-axis **BUT** it doesn't move down on the z-axis. So it stays high above the bed waiting to reach target temp of 30c.

If I enclose my printer then the probe can eventually reach that temperature and it goes down to probe the bed, but this will lead to problems since now the probe is unable to cooldown sufficiently. The calibration slows down and eventually timeouts.

If I remember correctly I think this feature was working for me earlier on marlin-bugfix-2.0.x version that was downloaded @ 18.01.2020. Since then I've downloaded two bugfix-releases: 30.1.2020 and latest two days ago on 27.3.2020. Neither one works for me.

My printer is custom p3steel/prusa type with skr v1.3+tmc2209 that I control via octoprint. My pinda v2 and mk52 are clones, but both seem to work (temperatures get reported).

### My Configurations

[configs.zip](https://github.com/MarlinFirmware/Marlin/files/4398746/configs.zip) 
note: I've flagged all my modifications to the default configs with <| to make finding changes easier for myself. 

### Steps to Reproduce

1. Execute G76 command from octoprint terminal
2. Probe moves up to the cooldown position
3. Bed to heats up to 60c

**Expected behavior:** [What you expect to happen]

4. Probe moves down to the "heat up" position (1mm above bed)

**Actual behavior:** [What actually happens]

4. Probe moves to the center of the bed but doesn't move down on z-axis (stays at z=120 in my case)

thisiskeithb, 2020-03-29:
The hotend is supposed to move to the cooldown position while the bed heats up. See Marlin's [Probe Temperature Compensation](https://marlinfw.org/docs/features/probe_temp_compensation.html) documentation for more info.

mavk0, 2020-03-29:
Yes, I agree. But after the bed has heated up that document says ”Move the probe to the probing point (1mm above heatbed)”.That doesn’t happen. It moves but only on xy-axis to the center but not down to 1mm. Maybe I’m failing to explain my problem in a clear manner, but this happens on my printer: >G761) homes xyz2) printhead moves up to cooldown position (upper right side of the frame for me @ z = 120)3) bed heats up to 60c---4) when 60c is reached: printhead moves to center _but_ doesn’t move down as described in that document: ”Move the probe to the probing point (1mm above heatbed)”.---5) wait until the probe heats up to the target (30°C) The problem is that completing that next phase is very difficult when probe is so high above the bed @ z = 120. Lähettäjä: thisiskeithbLähetetty: sunnuntai 29. maaliskuuta 2020 16.25Vastaanottaja: MarlinFirmware/MarlinKopio: mavk0; AuthorAihe: Re: [MarlinFirmware/Marlin] [BUG] G76 not working correctly (#17325) The hotend is supposed to move to the cooldown position while the bed heats up. See Marlin's Probe Temperature Compensation documentation for more info.

mavk0, 2020-04-04:
Closing this as [#17394](https://github.com/MarlinFirmware/Marlin/pull/17392) should fix my problem.

closed 2020-04-04 by mavk0