MarlinFirmware/Marlin #17326: [FR] G30 - 4 decimal places

qwewer0, 2020-03-29:
### Description

Is there a way to make G30 - Single Z-Probe - to output in 4 decimal places, instead of 2?

ellensp, 2020-03-30:
You realize that is 0.1 micron?  no way your machine has that much accuracy or repeatably.

ellensp, 2020-03-30:
I don't see anything obvious in the code that limits the DP, what is probably the issue is that you would need > 10000 steps/mm  to resolve the Z position to this level.

qwewer0, 2020-03-30:
So, it could be that the given G30 output is rounded up by the software, and that is why I only see 2 decimal places?
1/16 micro stepp is 0.0025mm, 2.5 micron.

qwewer0, 2020-03-30:
Yes, it is a software limit. G30 gives value to 4 decimal places with Python.

ellensp, 2020-04-02:
What are your Steps/mm for Z ?

qwewer0, 2020-04-02:
400, but why does it matter?

ellensp, 2020-04-02:
So I can replicate your setting, to try and see what is going on.

qwewer0, 2020-04-02:
Ok, but I don't think it has anything to do with it, as G29 and G30 gives 2 decimal places, but with G29 S-1 It lists all the Z value to 4 decimal places. But I want the G30 values and not the UBL values at the moment.

qwewer0, 2020-07-12:
https://github.com/MarlinFirmware/Marlin/pull/18367

closed 2020-07-12 by qwewer0