MarlinFirmware/Marlin #17328: Compile / linking error SKR Pro

MikeEitel, 2020-03-29:
# NO SUPPORT REQUESTS PLEASE

Support Requests posted here will be automatically closed!

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. See https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question.

For best results getting help with configuration and troubleshooting, please use the following resources:

- RepRap.org Marlin Forum http://forums.reprap.org/list.php?415
- Tom's 3D Forums https://discuss.toms3d.org/
- Facebook Group "Marlin Firmware" https://www.facebook.com/groups/1049718498464482/
- Facebook Group "Marlin Firmware for 3D Printers" https://www.facebook.com/groups/3Dtechtalk/
- Marlin Configuration https://www.youtube.com/results?search_query=marlin+configuration on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a Bug Report at https://github.com/MarlinFirmware/Marlin/issues/new/choose).

..................
My marlin 2.0.4  configuration compiles like a charm
Now with version 2.0.5.2  same config seems to compile ok but ( I guess linking) throws error in the last phase of building .elf file.   No red errors in the phase above
..........................
.pio/build/BIGTREE_SKR_PRO/src/src/lcd/extensible_ui/lib/ftdi_eve_touch_ui/language/language.cpp.o:H:\___3DPrinting\MeTriumDuo_SKR_V2.0.5/Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\language/language.cpp:27: first defined here
**collect2.exe: error: ld returned 1 exit status
*** [.pio\build\BIGTREE_SKR_PRO\firmware.elf] Error 1**
................................
Is this a new bug or how can I find what is wrong
THX for hints in advance

[platformio.zip](https://github.com/MarlinFirmware/Marlin/files/4399216/platformio.zip)

chgi, 2020-03-30:
Try cleaning up your source. In my clean download of Marlin 2.0.5.2, there is no `\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\language\language.cpp`.
There is however a `\src\lcd\extui\lib\ftdi_eve_touch_ui\language\language.cpp`.

MikeEitel, 2020-03-30:
Big THX chgi

That did it.
I have to change my way of working! Until now I only saved my config files to a save place and unzipped the new marlin version over my project. Than replace by the (compared and upgraded) old config files. But this does not work with the changed directory structure of the reorganised STM32xx processors....

Now I can compile succesfull.

closed 2020-03-30 by MikeEitel