MarlinFirmware/Marlin #17330: [BUG] Default  motherboard configuration and baudrate changed

ebraiman, 2020-03-30:
### Bug Description

### My Configurations N/A

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
[Marlin-2.0.x.zip](https://github.com/MarlinFirmware/Marlin/files/4400355/Marlin-2.0.x.zip)

### Steps to Reproduce

1. Download current version of Marlin firmware 2.0.5.2
2. Extract
3. Search configuration.h for, "motherboard"

**Expected behavior:** [What you expect to happen]
#define BAUDRATE 250000

// Enable the Bluetooth serial interface on AT90USB devices
//#define BLUETOOTH

// Choose the name from boards.h that matches your setup
#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_RAMPS_14_EFB
#endif

**Actual behavior:** [What actually happens]

#define BAUDRATE 500000

// Enable the Bluetooth serial interface on AT90USB devices
//#define BLUETOOTH

// Choose the name from boards.h that matches your setup
#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_MALYAN_M200
#endif

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

Grogyan, 2020-03-30:
I found this too, very annoying.

thisiskeithb, 2020-03-30:
I don't think https://github.com/MarlinFirmware/Marlin/commit/81860e9deb7bdc66d87384abcf050d57b8b14588 should have been applied to the default config in the `2.0.x` branch, but I tagged @thinkyhead.

thisiskeithb, 2020-03-30:
I created a PR to revert the config changes: https://github.com/MarlinFirmware/Marlin/pull/17332

waynoedm, 2020-03-31:
Agreed. Default settings have been changed in the current downloadable release.

sjasonsmith, 2020-04-13:
@ebraiman it seems this has been fixed. Would you mind closing this issue?

ebraiman, 2020-04-14:
Verified.

closed 2020-04-14 by ebraiman