MarlinFirmware/Marlin #17334: SKR PRO+TMC5160 BTT v1.2 =noise

Yury777, 2020-03-30:
Hello everyone,
I’am working on a self-made CoreXY printer:
SKR PRO
24 Volt 
TMC5160 v1.2 from BIGTECHTREE (removed the CLK pin)
Nema 17, 0.9’’ 42BYGHM809
TFT 3.5 v.2
Mechanical Endstops

the printer prints normally, but makes a lot of noise.
How can it be quieter? 
Here are my settings & log from M122:
[config.zip](https://github.com/MarlinFirmware/Marlin/files/4403645/config.zip)

M122
		X	Y	Z	E
Enabled		true	true	true	false
Set current	850	850	850	850
RMS current	836	836	839	839
MAX current	1179	1179	1183	1183
Run current	13/31	13/31	16/31	16/31
Hold current	7/31	7/31	9/31	9/31
Global scaler	129/256	129/256	133/256	133/256
CS actual	7/31	7/31	9/31	9/31
PWM scale	15	14	19	9
stealthChop	true	true	true	true
msteps		16	16	16	16
tstep		max	max	max	max
PWM thresh.	4	4	98	32
[mm/s]		1235	1235	20	30
OT prewarn	false	false	false	false
triggered
 OTP		false	false	false	false
off time	3	3	3	3
blank time	24	24	24	24
hysteresis
 -end		-1	-1	-1	-1
 -start		1	1	1	1
Stallguard thrs	0	0	0	0
DRVSTATUS	X	Y	Z	E
sg_result	0	0	0	0
stallguard				
fsactive				
stst					
olb					
ola					
s2gb					
s2ga					
otpw					
ot					
Driver registers:
		X	0x80:07:40:00
		Y	0x80:07:40:00
		Z	0x80:09:40:00
		E	0x80:09:40:00

Testing X connection... OK
Testing Y connection... OK
Testing Z connection... OK
Testing E connection... OK

DroneMang, 2020-05-09:
Here are my configs.

Basically the same setup.  I have sensorless homing working, and I did not cut the pins.  I searched the bigtree facebook page on this topic and arrived at that conclusion.

Do you have stealthchop enabled with hybrid threshold?  Try tuning down your current too.
[Marlin-2.0.x-9 5160.zip](https://github.com/MarlinFirmware/Marlin/files/4604371/Marlin-2.0.x-9.5160.zip)

boelle, 2020-06-20:
@Yury777 is this still an issue?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle