MarlinFirmware/Marlin #17339: [BUG] (short description)

Skyllaer, 2020-03-30:
Bug Description

after ending a print, the LCD keeps cycling between active printed time / remaining print time
[config.zip](https://github.com/MarlinFirmware/Marlin/files/4404853/config.zip)

### My Configurations

uploaded, using 2.0.5.2

### Steps to Reproduce

slice a hollow cube with 1 wall, no bottom and top layers.
In cura post proces add show remaining time
Print the cube

**Expected behavior:** 

Finishing the cube, the printer would go back in the ready state.

**Actual behavior:** 

Timings between the cura / printers remaining timer are way off. Cura's timer is correct.
After the print is finished, the printer get's stuck in the "active / remaining time" and doesn't get into the ready state again.

#### Additional Information
[image]

thinkyhead, 2020-05-01:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

boelle, 2020-06-20:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-20 by boelle