MarlinFirmware/Marlin #17343: [BUG] Bed leveling fade effectively disabled, the default 10mm is overwritten with 0mm

CarlosGS, 2020-03-30:
The fade height is [initialized as 10mm](https://github.com/MarlinFirmware/Marlin/blob/659316f797e086d84ef94c5aa407f618d6e81600/Marlin/src/module/configuration_store.cpp#L624), which is a reasonable value.
**But it is then [overwritten with 0mm](https://github.com/MarlinFirmware/Marlin/blob/659316f797e086d84ef94c5aa407f618d6e81600/Marlin/src/module/configuration_store.cpp#L2474).**

So starting with an empty EEPROM, the fading is initially disabled. I don't think it's the desired behavior since it continuously updates Z during the whole print.

Possibly related:  #12410

(Tried to keep the bug report concise, thanks so much for this software, you are awesome!!)

CarlosGS, 2020-05-07:
This is still an issue in the latest version.

boelle, 2020-06-20:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

CarlosGS, 2020-06-21:
Hi, apart from ENABLE_LEVELING_FADE_HEIGHT, this bug is 100% configuration agnostic.

My bug report links to the problematic code lines (I've verified the two conflicting definitions are still present in the latest version).

The only thing left is to decide whether to keep 10mm or 0mm as the default. I'd suggest 10mm since 0mm keeps it enabled for the whole prints. Cheers! :)

boelle, 2020-06-21:
so its more a config issue than a bug?

CarlosGS, 2020-06-21:
I'd consider it an "initialization consistency" bug, since when configuring a new machine you can't enable level fade and expect an initialization of 10mm. At this moment there are two choices for users:
- On every fresh install (or eeprom clearance), manually setting a level height value, through the LCD or gcode commands, to overwrite the wrong 0.0mm.
- Manually modify configuration_store.cpp to guarantee a default initialization >0.0mm in any firmware updates.

**Otherwise, some users get 10mm default fading, and some 0mm -> whole print fading (!!)**. Sorry if I couldn't summarize it properly before: Changing `new_z_fade_height` 0.0 to 10.0 ([current line](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/src/module/configuration_store.cpp#L2513)) would solve the issue.

CarlosGS, 2020-08-15:
Hi, as far as I can tell, the bug is still present:
- Fade defined as 10.0 in the configuration: https://github.com/MarlinFirmware/Marlin/blob/42fbd527f3bcc317ae6851aca5a45bdbf979de55/Marlin/src/module/configuration_store.cpp#L637
- Initialized as 0.0 when configuration is not available (i.e. first run): https://github.com/MarlinFirmware/Marlin/blob/42fbd527f3bcc317ae6851aca5a45bdbf979de55/Marlin/src/module/configuration_store.cpp#L2480
Cheers!

qwewer0, 2020-08-15:
Maybe a setting in Configuration.h would be useful?

```
#define ENABLE_LEVELING_FADE_HEIGHT
#if ENABLED(ENABLE_LEVELING_FADE_HEIGHT)
  #define LEVELING_FADE_HEIGHT 10
#endif
```

qwewer0, 2020-08-15:
After an `M502` my fade height too changes to **0**.

qwewer0, 2020-09-10:
Is there a way to get an editable fade height in Configuration.h or Configuration_adv.h? So that value is kept after a default restore?

CarlosGS, 2020-09-10:
+1 to that solution, it would be ideal.

(Latest version renamed the file, issue still present, I've updated the above references)

qwewer0, 2020-10-11:
Still a problem.

CarlosGS, 2020-11-14:
This is still present in the latest code.

To summarize: The only thing left is to decide whether to keep 10mm or 0mm as the default. I'd suggest 10mm since 0mm keeps it enabled for the whole prints.

sjasonsmith, 2020-11-29:
@CarlosGS please review the pull request references above, and provide feedback. I believe this will satisfy your request if it is accepted.

CarlosGS, 2020-11-29:
Yep! Brilliant! Thanks!! :D

sjasonsmith, 2020-11-29:
This was merged, so I'll close this out.

PR #20316 (2020-11-29): Add default fade height configuration
### Description

Fade height is currently unintuitive for users. It is stored in the EEPROM, yet there is no way to configure a default value to use when re-initializing the EEPROM.

This change adds a configuration option, which is uncommented by default and set to 10.0mm. This is a slight behavior change, but probably matches what people intuitively expect to be default behavior. It could easily be commented out by default if that is a concern, in which case it defaults to 0.0.

### Benefits

Allows fade height setting to persist across EEPROM reset, by  [... cut]
closed 2020-11-29 by sjasonsmith