MarlinFirmware/Marlin #17344: [BUG] fadding in Z with UBL

pemtyum, 2020-03-31:
**Hello!
I started using UBL to level the bed, build a DIY system very similar to BlTouch for my printer. Yesterday I finished setting everything in marlin and started printing at around 18:00.
Since then I am printing. I take out a model and send another impression right there.
Today, at about 4:00 pm, I notice a problem in the last print.
What you see in the image, the first impression is the one on the right, and in order as it deformed as it printed, up to the model on the left where you first notice the problem.**

Hola!
Empece a usar UBL para nivelar la cama, arme un sistema DIY muy similar al BlTouch para mi impresora. ayer termine de setear todo en marlin y empece a imprimir como a las 18:00.
Desde entonces estoy imprimiendo. saco un modelo y ahi mismo mando nuevamente otra impresion.
Hoy, como a las 16:00, me doy cuenta de un problema en la ultima impresion.
Lo que se ve en la imagen, la primera impresion es la de la derecha, y en orden como se fue deformando a medida que imprimia, hasta el modelo de la izquierda donde por primera vez note el problema.

[image]

**I have a FADE HIGH = 10mm configured for the UBL.
In the G-code file I'm printing, at first I have (among other things):
G29 L1
G29 J
(i.e. I don't have a G28)
and at the end I have a code M84
thinking, I came to the conclusion, that by repeating this many times, the height Z is deviated, and when executing only G29 J, a correction of the height Z is not made!
and then during the 10mm correction, the layer height is corrected to compensate for the difference that occurs between the theoretical position and the deviation that occurs between turning the motors on and off.
Remember that in the first impression, the G29 J command made a G28 homing first, but not in the rest of the impressions.**

**So to fix provisionally, before sending a new impression, I ordered a G28, and then the file again. After this it is again printing perfect.**

**Anyway, I think the 3-point test performed by G29 J is not correcting Z's home plate.
Whether or not the engines are turned off, the correction should only be for the bed tilt, leading to a new height Z and not the actual one as it currently is.**
**I do not know if marlin is able to work with a vector normal to the plane and make corrections while ascending in Z, but I assumed that it would be so (in smotthieboard it is so)
I hope it is an easy problem to solve. Cheers!**

Tengo configurado un FADE HIGH = 10mm para el UBL.
En el archivo de codigo G que estoy imprimiendo, al principio tengo (entre otras cosas):
G29 L1
G29 J
(es decir, no tengo un G28)
y al final tengo un codigo M84
pensando, llegue a la conclusion, de que al repetir muchas veces esto, se va desviando la altura Z, y al ejecutar solo G29 J, no se hace una correccion de la altura Z!
y entonces durante los 10mm de correccion, se corrige la altura de capa para compensar la diferencia que se produce entre la posicion teorica y la desviacion que se produce entre apagar y encender los motores.
Recorde que en la primera impresion, el comando G29 J, hizo primero un homing G28, pero en el resto de impresiones no.

Por tanto para solucionar de forma provisoria, antes de mandar una nueva impresion, le ordene un G28, y luego el archivo nuevamente. Luego de esto nuevamente esta imprimiendo perfecto.

En fin, creo que el test de 3 puntos ejecutado por G29 J no esta corregiendo el home de Z
Ya sea que se apague o no los motores, la correccion solo deberia ser de la inclinacion de cama, llevando a una nueva altura Z y no a la real como es actualmente.
Desconozco si marlin es capaz de trabajar con un vector normal al plano y hacer correcciones mientras asciende en Z, pero suponia que seria asi (en smotthieboard es asi)
Espero que sea un inconveniente facil de resolver. Saludos!

**PS: I find it hard to express myself in English, so to be as precise as possible choose to use google translate. I leave versions in both languages**

PD: me cuesta expresarme en ingles, asi que para ser lo mas preciso opte por usar google translate. Dejo versiones en ambos idiomas

Moyster, 2020-03-31:
as far as i know, 1.1.x is deprecated, but you mention you use the stable release, so... before someone else asks you if you tried marlin 2.x / bugfix 2.x, did you try 1.1.x-bugfix ? ( https://github.com/MarlinFirmware/Marlin/archive/bugfix-1.1.x.zip )

thinkyhead, 2020-04-27:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

closed 2020-06-20 by boelle